zephyrproject-rtos/zephyr #40614: poll: the code judgment condition is always true

Zhaoningx, 2021-11-24:
In the source code of poll.c,
[image]

The num_events is a signed variable and 0U is the unsigned variable,  this condition compares these two variables, as we all know, the num_events will change the type to an unsigned variable, and I'm designing a negative test case, and I try to set the num_events as -1, an error "num_events too large" will return. Otherwise, the condition is always true. So I think this branch can not be covered, whatever the value of num_events.

And this changs is showed in commit `bbbc38ba8f023bb2f1451e55568a7f3f4a273586`.

The test case runs in zephyrproject/zephyr, the branch is main.

The log:

when the num_events is **-1** and the second part is **0U**:
**E: syscall z_vrfy_k_poll failed check: num_events too large**
(the branch is not tested, the error is due to other conditions)

when the num_events is **-1** and the second part is **0**:
**E: syscall z_vrfy_k_poll failed check: num_events >= 0**
(the branch is tested)

dcpleung, 2021-11-24:
Would your test work after removing `U`?

Zhaoningx, 2021-11-25:
> Would your test work after removing `U`?

Hi @dcpleung, yes, I have tested this, and there is a existing test case named `test_k_poll_user_num_err` to test this point, and I find that the output is not expected, so I dig into it. As the bug description said:
The output of this test case is:
===================================================================
START - test_k_poll_user_num_err
E: syscall z_vrfy_k_poll failed check: num_events too large
 PASS - test_k_poll_user_num_err in 0.2 seconds
===================================================================

and after removing `U`, the log:
===================================================================
START - test_k_poll_user_num_err
E: syscall z_vrfy_k_poll failed check: num_events >= 0
 PASS - test_k_poll_user_num_err in 0.2 seconds
===================================================================

So need I to remove `U `in poll.c?

PR #40660 (2021-11-25): poll: modify the function z_vrfy_k_poll
Removing the `U `to avoid the type of num_events changed.
And make sure it is meaningful Z_SYSCALL_VERIFY micro.

Fixed #40614
commit 5287ce8f (2021-11-25): poll: modify the function z_vrfy_k_poll

Removing the 'U' to avoid the type of num_events changed.
And make sure it is meaningful Z_SYSCALL_VERIFY micro.

Fixed #40614
commit 22fc9853 (2022-01-25): poll: modify the function z_vrfy_k_poll

Removing the 'U' to avoid the type of num_events changed.
And make sure it is meaningful Z_SYSCALL_VERIFY micro.

Fixed #40614
closed 2021-11-25 by nashif