zephyrproject-rtos/zephyr #40620: zephyr with cadence xtensa core dsp LX7 ，helloworld program  cannot be entered after the program is executed

bawanqin1987, 2021-11-24:
**Describe the bug**
when i test the zephyr sample ,helloworld or sof,and when i running the program end,the dsp can not enter the idle state,and enter the arch halt

**To Reproduce**
Steps to reproduce the behavior:
1. linux cmd enter,with zephyr/sample/helloworld or sof,build -b ls_adsp_csk6
2. produce the zephyr elf,then i load the elf with cadence IDE tool Xlporer (8.0.18,hifi4 software version 2020.4.target hardware version LX7.1.4)
3. and run step and return, now i check the call trace,and is not running in IDLE state ,dsp doesn't enter waiti asm.

**Expected behavior**
expected the dsp enter idle state

**Impact**
The program did not end properly

**Logs and console output**
see below my ide call trace:

[image]

**Environment (please complete the following information):**
 - OS: (Linux ubuntu/redhat)
 - Toolchain (Zephyr SDK, hifi4 xt-xcc)
 - zephyr sdk version:2.6.8,commit id:7cf0a5c4a1883767ed6d29408a14872c3d718fc7

**Additional context**
Add any other context about the problem here.

andyross, 2022-01-20:
Where can I find the soc/board definition for ls_adsp_csk6?  This seems likely to be a problem in the platform layer.  Who did the integration?

bawanqin1987, 2022-01-21:
The existing zephyr only supports LX6 platform, I wonder if LX7 support?  Because lX7 and LX6 hiFI DSP start the assembly instructions are not quite the same ,I don't know if it matters!
ls_adsp_csk6 is my new add arch.

hongshui3000, 2022-01-21:
@bawanqin1987 
Please don't turn on your high priority interrupt routine, it will cause weird errors. (If your architecture has high-priority interrupts, such as the timer timer inside the architecture, it is a high-priority interrupt, and the xtensa initialization code will not block these interrupts)

andyross, 2022-02-01:
Any motion on this?  Are you guys able to share the soc/board definitions or otherwise help us isolate a problem with the open source project?  If not, we might want to close this and look for another support channel; the github bug database is for bugs with the open source components.

bawanqin1987, 2022-02-07:
ok, i will share my soc/board
[ls_adsp.zip](https://github.com/zephyrproject-rtos/zephyr/files/8013881/ls_adsp.zip)

we use hifi4 lx7 arch,and the core setup code :
[xea2.zip](https://github.com/zephyrproject-rtos/zephyr/files/8013965/xea2.zip)

andyross, 2022-02-08:
Um... the ls_adsp file is nothing but a directory tree, and xea2.zip contains only assembly code and doesn't seem to have any Zephyr integration.  Can't you submit this as a pull request or push a branch somewhere I can get and try to pull?  I won't be able to build it if it's targetted to the Cadence toolchain, but I can at least read through it and make suggestions.

bawanqin1987, 2022-02-08:
ok,i will pull

andyross, 2022-02-23:
Any update, @bawanqin1987?  We're going to need to close this soon.  If this still needs attention and you have the updates requested, you can always reopen it or submit a new issue.

andyross, 2022-03-01:
Closing this per process.  Again, that's not a refusal; just reopen/resubmit if you have code.

closed 2022-03-01 by andyross