zephyrproject-rtos/zephyr #40624: twister: coverage: Using --coverage flag for on-target test make tests last until time limit

PerMac, 2021-11-24:
**Describe the bug**
Calling twister with coverage flag and device testing with:
`./scripts/twister --coverage --device-testing --device-serial /dev/ttyACM0 -T tests/kernel/ -p nrf52840dk_nrf52840 -v -v`
makes all tests lasting ~60s instead ~5s. On the console one can see that even though `DEVICE: PROJECT EXECUTION SUCCESSFUL` is printed way earlier, the test is only ended after ~65s which is suspiciously close to the timeout limit. Nevertheless, tests are marked as passed. I assume there is some logic error with the workflow for `--coverage` but I haven't looked in the code yet. 

**To Reproduce**
Steps to reproduce the behavior:
1. have e.g. nrf52840dk connected
2. run `./scripts/twister --coverage --device-testing --device-serial /dev/ttyACM0 -T tests/kernel/ -p nrf52840dk_nrf52840 -v -v`
3. observe very long execution times and twister waiting even though a test printed its end.

**Expected behavior**
The execution time is not as log.

**Impact**
Annoyence. Can be a show stoper for larger tests suites

**Logs and console output**
e.g. 
```
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    -  1/96 nrf52840dk_nrf52840       tests/kernel/mp/kernel.multiprocessing             SKIPPED (filter)
INFO    -  2/96 nrf52840dk_nrf52840       tests/kernel/fifo/fifo_usage/kernel.fifo.usage     PASSED (device 65.413s)
INFO    -  3/96 nrf52840dk_nrf52840       tests/kernel/context/kernel.common                 PASSED (device 64.545s)
INFO    -  4/96 nrf52840dk_nrf52840       tests/kernel/fifo/fifo_api/kernel.fifo             PASSED (device 64.575s)
```
All lasted ~65s

**Environment (please complete the following information):**
 - OS: Linux 18.04
 - Toolchain Ubuntu 18.04
 - Commit SHA or Version used: 67a265b0df

cfriedt, 2021-11-30:
@PerMac - would it alleviate the problem if there were a `--timeout` flag for twister that overrides YAML / the default?

PerMac, 2021-12-01:
@cfriedt The tests pass, it is not an issue with a too short timeout. The issue is that all of them take suspiciously long (equal to timeout value) to finish. I think there is some simple logic issue in the code (missing exit condition?), and twister doesn't end the test on time but waits for the maximal allowed time (timeout). But I haven't got time to check the code ye.

nashif, 2021-12-14:
The issue here is that we are waiting for coverage data that is not coming, because the platform does not support coverage, ie.. it does not enable `HAS_COVERAGE_SUPPORT`.

PerMac, 2021-12-15:
Do we need an extra filtration then when `--coverage` is used?

nashif, 2022-02-10:
> Do we need an extra filtration then when `--coverage` is used, or a warning?

Hmm if you know of any easy way to do that, please go ahead. I do not see this as a bug  however.

closed 2022-02-10 by nashif