zephyrproject-rtos/zephyr #40638: netusb ECM driver locking mutex from ISR

ottojo, 2021-11-24:
**Describe the bug**
When using USB networking, the driver accesses networking functions which lock mutexes from ISR context. This leads to failed assertion `mutexes cannot be used inside ISRs.`.

* `ecm_status_cb` is registered as USB status callback [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/usb/class/netusb/function_ecm.c#L435) (apparently called via ISR)
* `ecm_status_cb` calls `ecm_status_interface` [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/usb/class/netusb/function_ecm.c#L374)
* `ecm_status_interface` calls `netusb_enable` [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/usb/class/netusb/function_ecm.c#L356)
* `netusb_enable` calls `net_if_up` [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/usb/class/netusb/netusb.c#L102)
* `net_if_up` locks mutex [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/net_if.c#L3978)

* `netusb_enable` calls `net_if_up` [here](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/usb/class/netusb/netusb.c#L102)

**To Reproduce**
(Working on providing a minimal example...)

**Expected behavior**
Assertion should not fail, the interface should not be enabled from ISR context.

**Impact**
The network interface seems to work, but this makes it impossible to me to use `__ASSERT`, since the program can't run with `CONFIG_ASSERT` enabled.

**Logs and console output**
(WIP)

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.13.1
 - Zephyr: 70b0734b5e53d53b49e1fd95feef5cd2ff233117 (reasonably recent main branch)
 - Board: `teensy41`

jfischer-no, 2021-11-24:
@ottojo There is no workaround, USB_DeviceNotificationTrigger needs to be offloaded to a thread/workqueue.

danieldegrasse, 2022-01-20:
@ottojo I can reproduce the error on a RT1064 EVK building the "echo_server" sample with the following command:
`west build -p always -b mimxrt1064_evk samples/net/sockets/echo_server -DOVERLAY_CONFIG=overlay-netusb.conf -DCONFIG_ETH_MCUX=n -DCONFIG_ASSERT=y`. On my board, #42019 fixes the issue (per @jfischer-no's suggestion of offloading the ISR to a thread) can you verify that branch works for you?

PR #42019 (2022-01-28): drivers: usb_dc_mcux: Move mcux callback handler out of ISR context
MCUX USB Device controller ISR was making usb callbacks directly, which caused assertion
failures when a callback attempted to lock a mutex. Move USB callback
handler to separate thread, and make ISR notify thread via message
queue. This change is based off the `usb_dc_kinetis` driver, which uses the same approach to offload usb interrupts

Fixes #40638
commit 2ac1b429 (2022-01-28): drivers: usb_dc_mcux: Move mcux callback handler out of ISR context

MCUX usb ISR was making usb callbacks directly, which caused assertion
failures when a callback attempted to lock a mutex. Move USB callback
handler to separate thread, and make ISR notify thread via message
queue.

Fixes #40638
closed 2022-01-28 by carlescufi