zephyrproject-rtos/zephyr #40640: Segfault when using composite USB device

holgerschurig, 2021-11-24:
**Describe the bug**
I try the zephyr/samples/subsys/usb/cdc_acm with two CDC_ACM ports and get a segfault.

My plan was however to use "usbip" on Linux to test things out faster compared to always flash things into my device.

**To Reproduce**
Steps to reproduce the behavior:

```
$ cd zephyr/samples/subsys/usb/cdc_acm
$ cat native_posix_64.overlay 
&zephyr_udc0 {
    status = "okay";
    cdc_acm_uart0 {
        compatible = "zephyr,cdc-acm-uart";
        label = "CDC_ACM_0";
    };
    cdc_acm_uart1 {
        compatible = "zephyr,cdc-acm-uart";
        label = "CDC_ACM_1";
    };
};
$ rm -rf buiild
$ west build -b native_posix_64
$ build/zephyr/zephyr.elf 
UART_0 connected to pseudotty: /dev/pts/3
Segmentation fault
```

Maybe the backtrace is helpful?

```
$ gdb build/zephyr/zephyr.elf 
Reading symbols from build/zephyr/zephyr.elf...
(gdb) r
Starting program: /home/schurig/d/v73/zephyr/samples/subsys/usb/cdc_acm/build/zephyr/zephyr.elf 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff7c87700 (LWP 93534)]
UART_0 connected to pseudotty: /dev/pts/3
[New Thread 0x7ffff7472700 (LWP 93535)]
[New Thread 0x7ffff6c71700 (LWP 93536)]
[New Thread 0x7ffff6470700 (LWP 93537)]
[New Thread 0x7ffff5c55700 (LWP 93538)]

Thread 3 "zephyr.elf" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff7472700 (LWP 93535)]
0x0000000000404e56 in usb_get_cfg_data (iface=<optimized out>)
    at /home/schurig/d/v73/zephyr/subsys/usb/usb_descriptor.c:292
292			if (__usb_data_start[i].interface_descriptor == iface) {
(gdb) bt
#0  0x0000000000404e56 in usb_get_cfg_data (iface=<optimized out>)
    at /home/schurig/d/v73/zephyr/subsys/usb/usb_descriptor.c:292
#1  usb_fix_descriptor (head=0x4111ab <cdc_acm_cfg_0+8>)
    at /home/schurig/d/v73/zephyr/subsys/usb/usb_descriptor.c:401
#2  0x00000000004050b5 in usb_get_device_descriptor ()
    at /home/schurig/d/v73/zephyr/subsys/usb/usb_descriptor.c:487
#3  0x0000000000404b2e in usb_device_init (dev=<optimized out>)
    at /home/schurig/d/v73/zephyr/subsys/usb/usb_device.c:1645
#4  0x00000000004091b4 in z_sys_init_run_level (level=level@entry=2)
    at /home/schurig/d/v73/zephyr/kernel/device.c:72
#5  0x0000000000409271 in bg_thread_main (unused1=<error reading variable: value has been optimized out>, 
    unused2=<error reading variable: value has been optimized out>, 
    unused3=<error reading variable: value has been optimized out>) at /home/schurig/d/v73/zephyr/kernel/init.c:186
#6  0x00000000004029f0 in z_thread_entry (entry=<optimized out>, p1=<optimized out>, p2=<optimized out>, 
    p3=<optimized out>) at /home/schurig/d/v73/zephyr/lib/os/thread_entry.c:36
#7  0x00000000004061a7 in posix_thread_starter (arg=0x0)
    at /home/schurig/d/v73/zephyr/arch/posix/core/posix_core.c:305
#8  0x00007ffff7e58ea7 in start_thread (arg=<optimized out>) at pthread_create.c:477
#9  0x00007ffff7d88def in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

**Environment (please complete the following information):**
 - OS: Linux
 - Commit SHA or Version used: "git rev-parse HEAD" says 503b7e6f9ff07c2d340cefb6ae970495373e0153

**Additional context**
With only one serial port (e.g. without the overlay file creating CDC_ACM_UART1) I don't get a segfault.

However, with a similar entry in an overlay for a STM32 Nucleo board I don't have a segfault-equivalent, the two USB ports are working fine there.

```
$ tail -n 12 nucleo_f303re.overlay 
zephyr_udc0: &usb {
    pinctrl-0 = <&usb_dm_pa11 &usb_dp_pa12>;
    status = "okay";
    cdc_acm_uart0 {
        compatible = "zephyr,cdc-acm-uart";
        label = "CDC_ACM_0";
    };
    cdc_acm_uart1 {
        compatible = "zephyr,cdc-acm-uart";
        label = "CDC_ACM_1";
    };
};
```

jfischer-no, 2021-11-24:
> My plan was however to use "usbip" on Linux to test things out faster compared to always flash things into my device.

TBH driver _usb_dc_native_posix_, is not suitable for production use, for many reasons, it has nothing to do with posix either. I started to rewrite it, https://github.com/zephyrproject-rtos/zephyr/pull/34069, but current implementation has also many problems and USBIP actually needs HCI interface API, so I postponed it until new USB device controller API or USB HCI driver API are available.
And another thing, using usbip tool is not necessarily faster than flashing, and vhci-hcd can be very nasty.

holgerschurig, 2021-11-29:
I already used usbip. My device implements some JSON protocol on top of CDC_ACM, and I was able to get it running with usbip --- when I did only use one CDC_ACM device.

Now I tried two of them, and I only get a bit of data from my device, then it stops. So I though I can debug this faster with usbip. It's not super fast, but okayish.

BTW, I used this start script:

```
#!/bin/sh

modprobe vhci_hcd

DEV="/dev/serial/by-id/usb-ZEPHYR*"

test -e $DEV || {
	echo "Waiting for USB IP device ..."
	while true; do
		usbip attach -r localhost -b 1-1 2>/dev/null && break
		sleep 0.2
		continue
	done
}

echo "Waiting for serial USB device ..."
while true; do
	N=`ls -l $DEV 2>/dev/null | wc --lines`
	if [ $N = 0 ]; then
		sleep 0.2
		continue
	fi
	if [ $N = 1 ]; then
		break
	fi
	echo "More on than one device found:"
	ls $DEV
	exit
done

bin/testsuite `ls $DEV`
```

jfischer-no, 2021-12-09:
I finally invested time to follow backtrace you left in the bug description, thanks for that. It turns out that it is an issue in the macro USBD_CFG_DATA_DEFINE. `usb_cfg_data` structures in CDC ACM class  are placed with a gap or padding in specified RAM section on native_posix_64, this breaks the possibility to iterate over the structures. It does not happen with the `usb_cfg_data` structures in _tests/subsys/usb/desc_sections_, is a little strange. I have opened #41036 to resolve the issue.

cfriedt, 2021-12-10:
@jfischer-no - can you please also try to reproduce in v2.7.0?

jfischer-no, 2021-12-10:
> @jfischer-no - can you please also try to reproduce in v2.7.0?

yes, zephyr-v2.7.0 is affected.

PR #41036 (2021-12-10): usb: add iterable section in ram for usb_cfg_data 
Add iterable section in ram for struct usb_cfg_data and
new makro USBD_DEFINE_CFG_DATA which should be used to
define usb_cfg_date structures.
Rework USB device support to use macro STRUCT_SECTION_FOREACH.

Fixes: #40640
closed 2021-12-10 by MaureenHelm