zephyrproject-rtos/zephyr #40657: Cannot enable secondary pwm out channels on stm32f3

helo9, 2021-11-25:
**Describe the bug**
I face problems when trying to use the secondary pwm output channels on stm32f3_diso.
Inspecting the Pins shows no output, while primary channel is working.

**To Reproduce**
Steps to reproduce the behavior:
1. create an application using pwm output for stm32f3_disco rev E
2. use an overlay with the following content 
```
&pwm1 {
    pinctrl-0 = <&tim1_ch1_pa8 &tim1_ch2_pa9 &tim1_ch3_pa10 &tim1_ch4_pe14>;
    //pinctrl-0 = <&tim1_ch1n_pe8 &tim1_ch2n_pe10 &tim1_ch3n_pe12 &tim1_ch4_pe14>;
};
```
3. `west build --board=stm32f3_disco@E`
4. Flash and check PWM output
5. comment out first pinctrl line and uncomment second
6. Build and flash
7. Check PWM output on secondary pins -> not working

**Expected behavior**
The secondary pins should work the same way, the primary ones work.

Note: Inspecting the `TIM1_CCER` shows no difference, for both firmwares ( `0x1111` for both configurations,
while I expected `0x1444` for the second one, `x/xw 0x40012C20` in gdb)

**Environment (please complete the following information):**
 - OS: Linux, Manjaro
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA: 503b7e6f9ff07c2d340cefb6ae970495373e0153

helo9, 2021-11-25:
I think like pinctrl already works correctly, but in `pwm_stm32_pin_set` cannot set the correct bits as [pwm_stm33.c#L262](https://github.com/zephyrproject-rtos/zephyr/blob/5287ce8ff003f9ad7428864a9e53a42424b3d135/drivers/pwm/pwm_stm32.c#L262) can only select the ordinary pins.
So I would add another ch2ll like array, store which channels should use the N pin in the configuration and choose the array in .._pin_set from configuration. Did I miss something?

FRASTM, 2021-11-29:
It seems to be the key for this usecase
```
/** Channel to LL mapping. */
static const uint32_t ch2ll[TIMER_MAX_CH] = {
	LL_TIM_CHANNEL_CH1N, LL_TIM_CHANNEL_CH2N,
	LL_TIM_CHANNEL_CH3N, LL_TIM_CHANNEL_CH4,
#if TIMER_HAS_6CH
	LL_TIM_CHANNEL_CH5, LL_TIM_CHANNEL_CH6
#endif
};
```

Now we could think of extending the TIMER_MAX_CH  at least for the F3 serie, considering that "  CH5 and CH6 channels are not available for all F3 devices"

FRASTM, 2022-01-21:
One Proposal is to create a second _dts/bindings/pwm/st,stm32-pwm-n.yaml_  beside the st,stm32-pwm.yaml
Then for timers which have "N" channels, two pwm nodes are created in the DTS on the timer instance. Both pwm and pwm-n have the same model.
Example for Timer 1 of the stm32f303.dtsi:
```
			pwm {
				compatible = "st,stm32-pwm";
				status = "disabled";
				label = "PWM_1";
				#pwm-cells = <3>;
			};

			pwm_n {
				compatible = "st,stm32-pwm-n";
				status = "disabled";
				label = "PWM_1N";
				#pwm-cells = <3>;
			};
```

The boards enables the timer 1 pwm channels 1 
```
&timers1 {
	st,prescaler = <10000>;
	status = "okay";
	pwm1: pwm {
		status = "okay";
		pinctrl-0 = <&tim1_ch1_pa8>;
		pinctrl-names = "default";
	};
	pwm1n: pwm_n {
		status = "okay";
		pinctrl-0 = <&tim1_ch1n_pa7>;
		pinctrl-names = "default";
	};
};
```

FRASTM, 2022-01-24:
Testing the samples/basic/blinky_pwm on the nucleo_l45rzi I can see the pb with red led on pin PB14: 
setting the PB14 pin for timer15 channel 1 -> test ok
setting the PB14 for the timer1 channel 2n -> led off

FRASTM, 2022-01-24:
In case of output N the LL_TIM_OC_StructInit must init the pwm with 
		oc_init.OC**N**State = LL_TIM_OCSTATE_ENABLE;
		oc_init.OCState = LL_TIM_OCSTATE_DISABLE;

zzzh, 2022-02-16:
> Testing the samples/basic/blinky_pwm on the nucleo_l45rzi I can see the pb with red led on pin PB14: setting the PB14 pin for timer15 channel 1 -> test ok
> 
> ```
> 		red_pwm_led: red_pwm_led {
> 			pwms = <&pwm15 1 4 PWM_POLARITY_NORMAL>;
> 		};
> ```
> 
> setting the PB14 for the timer1 channel 2n -> led off
> 
> ```
> 		red_pwm_led: red_pwm_led {
> 			pwms = <&pwm1 2 4 PWM_POLARITY_NORMAL>;
> 		};
> ```

@FRASTM 
In this line: pwms = <&pwm1 2 4 PWM_POLARITY_NORMAL>;
what does 4 mean? is there a document for it?

FRASTM, 2022-03-21:
The ./dts/bindings/pwm/st,stm32-pwm.yaml   defines the _period_ to be 

```
pwm-cells:
  - channel
# period in terms of nanoseconds
  - period
  - flags
```

also see the ./include/devicetree/pwms.h

Mainly used with PWM devices like in the drivers/led/led_pwm.c
```
#define LED_PWM(led_node_id)						\
{									\
	.dev		= DEVICE_DT_GET(DT_PWMS_CTLR(led_node_id)),	\
	.channel	= DT_PWMS_CHANNEL(led_node_id),			\
	.period		= DT_PHA_OR(led_node_id, pwms, period, 100000),	\
	.flags		= DT_PHA_OR(led_node_id, pwms, flags,		\
				    PWM_POLARITY_NORMAL),		\
},
```

closed 2022-03-31 by carlescufi