zephyrproject-rtos/zephyr #40658: shtcx not reporting correct humidity value

joshld, 2021-11-25:
**Describe the bug**
shtcx.c source references temp struct member rather than humidity within shtcx_channel_get, suggest making this change

**Impact**
Values based on the temperature raw values used for humidity. (seemingly around 10-15% off)

Suggested change from:

```
static int shtcx_channel_get(const struct device *dev,
			      enum sensor_channel chan,
			      struct sensor_value *val)
{
	const struct shtcx_data *data = dev->data;

	if (chan == SENSOR_CHAN_AMBIENT_TEMP) {
		shtcx_temperature_from_raw(data->sample.temp, val);
	} else if (chan == SENSOR_CHAN_HUMIDITY) {
		shtcx_humidity_from_raw(data->sample.temp, val);
	} else {
		return -ENOTSUP;
	}

	return 0;
}
```

to:
```
static int shtcx_channel_get(const struct device *dev,
			      enum sensor_channel chan,
			      struct sensor_value *val)
{
	const struct shtcx_data *data = dev->data;

	if (chan == SENSOR_CHAN_AMBIENT_TEMP) {
		shtcx_temperature_from_raw(data->sample.temp, val);
	} else if (chan == SENSOR_CHAN_HUMIDITY) {
		shtcx_humidity_from_raw(data->sample.humidity, val);
	} else {
		return -ENOTSUP;
	}

	return 0;
}
```

In addition to these changes I would also suggest revising the measure_wait_us array as SHTC3 appears to reference SHTC1 values and vice versa. So either change to these definitions:
```
#define CHIP_SHTC1      1
#define CHIP_SHTC3      0
```

or flip the values in the array to this:
```
/* measure_wait_us[shtcx_chip][MEASURE_MODE] */
static const uint16_t measure_wait_us[2][2] = {
	/* shtc1: 14.4ms, 0.94ms */
	{ 14400, 940 }, /* shtc1 */
	/* shtc3: 12.1ms, 0.8ms */
	{ 1210, 800 }, /* shtc3 */
};
```

str4t0m, 2021-11-25:
Thanks for reporting this. PR #40661 should address the issues. Could you have a look at it?

joshld, 2021-11-25:
Reviewed.

PR #40661 (2021-11-29): drivers: sensor: shtcx: fix humidity and timing parameters
This commit fixes humidity value and model dependent timing parameters.

closes #40658
closed 2021-11-29 by nashif