zephyrproject-rtos/zephyr #40668: Issue with twister code coverage tests not working with minimal C library (nRF52840)

lairdjm, 2021-11-25:
**Describe the bug**
We are trying to get twister working for testing code on real devices and to get code coverage reports, for the current test, we are using 2 pinnacle 100's with pyocd-based debuggers attached, FTDI UART chips are used for the UART interface on these board. We have enabled coverage for the board by adding `select HAS_COVERAGE_SUPPORT` to our `Kconfig.board` file, generated a hardware map using `./scripts/twister --generate-hardware-map p100.yml` which looks like this:
```
- connected: true
  id: 046536025020322038523254323031323630303497969903
  platform: pinnacle_100_dvk
  product: DAPLink CMSIS-DAP
  runner: pyocd
  serial: /dev/ttyUSB0
- connected: true
  id: 046536025020322038523254323031323830303597969903
  platform: pinnacle_100_dvk
  product: DAPLink CMSIS-DAP
  runner: pyocd
  serial: /dev/ttyUSB1
```
And we are running the kernel tests using `./scripts/twister --coverage --device-testing --hardware-map p100.yml -p pinnacle_100_dvk -T tests/kernel/` - this has given us a list of failures, all with timeout errors (out of 95 tests, 36 failed):
```
ZEPHYR_BASE unset, using "/tmp/zephyr/zephyr"
Renaming output directory to /tmp/zephyr/zephyr/twister-out.11
INFO    - Zephyr version: v2.7.99-1827-g632628ec634d
INFO    - JOBS: 16
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 164 test scenarios (164 configurations) selected, 69 configurations discarded due to filters.

Device testing on:

| Platform         |                                               ID | Serial device   |
|------------------|--------------------------------------------------|-----------------|
| pinnacle_100_dvk | 046536025020322038523254323031323630303497969903 | /dev/ttyUSB0    |
| pinnacle_100_dvk | 046536025020322038523254323031323830303597969903 | /dev/ttyUSB1    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    9/  95   9%  skipped:   78, failed:    0
ERROR   - pinnacle_100_dvk          tests/kernel/mem_slab/mslab_threadsafe/kernel.memory_slabs.threadsafe FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/pinnacle_100_dvk/tests/kernel/mem_slab/mslab_threadsafe/kernel.memory_slabs.threadsafe/device.log                                                                             
INFO    - Total complete:   10/  95  10%  skipped:   78, failed:    1
ERROR   - pinnacle_100_dvk          tests/kernel/profiling/profiling_api/kernel.common.profiling FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/pinnacle_100_dvk/tests/kernel/profiling/profiling_api/kernel.common.profiling/handler.log                                                                                     
INFO    - Total complete:   14/  95  14%  skipped:   81, failed:    2
ERROR   - pinnacle_100_dvk          tests/kernel/pending/kernel.objects                FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/pinnacle_100_dvk/tests/kernel/pending/kernel.objects/device.log
INFO    - Total complete:   15/  95  15%  skipped:   81, failed:    3
ERROR   - pinnacle_100_dvk          tests/kernel/fifo/fifo_api/kernel.fifo             FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/pinnacle_100_dvk/tests/kernel/fifo/fifo_api/kernel.fifo/device.log                                                                                                            
INFO    - Total complete:   20/  95  21%  skipped:   85, failed:    4
ERROR   - pinnacle_100_dvk          tests/kernel/mem_slab/mslab_api/kernel.memory_slabs.api FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/pinnacle_100_dvk/tests/kernel/mem_slab/mslab_api/kernel.memory_slabs.api/device.log                                                                                           
INFO    - Total complete:   21/  95  22%  skipped:   85, failed:    5
ERROR   - pinnacle_100_dvk          tests/kernel/fatal/exception/kernel.common.stack_sentinel FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/pinnacle_100_dvk/tests/kernel/fatal/exception/kernel.common.stack_sentinel/device.log                                                                                         
INFO    - Total complete:   24/  95  25%  skipped:   87, failed:    6
ERROR   - pinnacle_100_dvk          tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/pinnacle_100_dvk/tests/kernel/mem_protect/stack_random/kernel.memory_protection.stack_random/device.log                                                                       
INFO    - Total complete:   25/  95  26%  skipped:   87, failed:    7
ERROR   - pinnacle_100_dvk          tests/kernel/mem_slab/mslab_concept/kernel.memory_slabs.concept FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/pinnacle_100_dvk/tests/kernel/mem_slab/mslab_concept/kernel.memory_slabs.concept/device.log                                                                                   
INFO    - Total complete:   37/  95  38%  skipped:   98, failed:    8
ERROR   - pinnacle_100_dvk          tests/kernel/mem_slab/mslab/kernel.memory_slabs.api FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/pinnacle_100_dvk/tests/kernel/mem_slab/mslab/kernel.memory_slabs.api/device.log                                                                                               
INFO    - Total complete:   38/  95  40%  skipped:   98, failed:    9
ERROR   - pinnacle_100_dvk          tests/kernel/sched/deadline/kernel.scheduler.deadline FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/pinnacle_100_dvk/tests/kernel/sched/deadline/kernel.scheduler.deadline/device.log                                                                                             
INFO    - Total complete:   43/  95  45%  skipped:  102, failed:   10
ERROR   - pinnacle_100_dvk          tests/kernel/sched/metairq/kernel.scheduler.metairq FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/pinnacle_100_dvk/tests/kernel/sched/metairq/kernel.scheduler.metairq/device.log                                                                                               
INFO    - Total complete:   44/  95  46%  skipped:  102, failed:   11
ERROR   - pinnacle_100_dvk          tests/kernel/sched/preempt/kernel.scheduler.preempt FAILED: Timeout
[... 116 lines cut ...]
#7  0x0000e99a in gcov_static_init () at /tmp/zephyr/zephyr/subsys/testsuite/coverage/coverage.c:273
#8  0x00024f8a in z_cstart () at /tmp/zephyr/zephyr/kernel/init.c:403
#9  0x0000f2c4 in z_arm_prep_c () at /tmp/zephyr/zephyr/arch/arm/core/aarch32/prep_c.c:190
#10 0x0001182a in z_arm_reset () at /tmp/zephyr/zephyr/arch/arm/core/aarch32/cortex_m/reset.S:154
```
At this point, I've gone back to basics with just trying to get code coverage working on an `nrf52840dk_nrf52840` board with the `hello_world` sample application, I find that:
* With the default build configuration and coverage enabled, the module faults at startup
* With the a configuration which has coverage enabled and heap size set to 16KB, the module faults at startup
* With the a configuration which has coverage enabled and uses the newlib C library instead (with default 2KB heap), the module starts and runs, and then outputs the expected gcov data, though I haven't checked/parsed it to see if it's valid or not.

So this seems like code coverage is inherantly not supported when using the default minimal C library? So how would we go about using twister to generate code coverage reports of the code base?

**To Reproduce**
Steps are above, but would need specific hardware to reproduce the setup

**Expected behavior**
The tests to pass

**Impact**
Showstopper. I've been trying to get a test system setup for weeks now and getting quite frustrated

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA or Version used: 632628ec634d9b2a0d7898fe53e28697e278f0a5

lairdjm, 2021-11-26:
Perhaps unrelated to minimal C library as in some other instances it seems impossible to get it to work and the whole application freezes at a pointer function call in `gcov_static_init`, file `subsys/testsuite/coverage/coverage.c` with a kernel test:
```
Breakpoint 2, gcov_static_init () at /tmp/zephyr/zephyr/subsys/testsuite/coverage/coverage.c:266
266             uintptr_t func_pointer_start = (uintptr_t) &__init_array_start;
(gdb) step
267             uintptr_t func_pointer_end = (uintptr_t) &__init_array_end;
(gdb) 
269             while (func_pointer_start < func_pointer_end) {
(gdb) print func_pointer_start
$2 = 233095
(gdb) print func_pointer_end
$3 = 233408
(gdb) step
272                     p = (void (**)(void)) func_pointer_start;
(gdb) stepi
0x0000f182      272                     p = (void (**)(void)) func_pointer_start;
(gdb) 
273                     (*p)();
(gdb) 
0x0000f186      273                     (*p)();
(gdb) 
0x0000f188      273                     (*p)();
(gdb) 
0x000cb900 in ?? ()
```

lairdjm, 2021-11-26:
Why is the gcov linker section resides in a CPP #ifdef? https://github.com/zephyrproject-rtos/zephyr/blob/main/include/linker/common-rom.ld#L85
If I try code coverage with twister and forcing CPP support, the applications no longer seem to lock up in the gcov section, they still don't work because of MPU faults, but that's a bit more progress:
```
INFO    - Total complete:    9/  95   9%  skipped:   78, failed:    0
ERROR   - bl654_dvk                 tests/kernel/profiling/profiling_api/kernel.common.profiling FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/bl654_dvk/tests/kernel/profiling/profiling_api/kernel.common.profiling/handler.log
INFO    - Total complete:   11/  95  11%  skipped:   79, failed:    1
ERROR   - bl654_dvk                 tests/kernel/mem_slab/mslab_concept/kernel.memory_slabs.concept FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/bl654_dvk/tests/kernel/mem_slab/mslab_concept/kernel.memory_slabs.concept/handler.log
INFO    - Total complete:   14/  95  14%  skipped:   81, failed:    2
ERROR   - bl654_dvk                 tests/kernel/sched/preempt/kernel.scheduler.preempt FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/bl654_dvk/tests/kernel/sched/preempt/kernel.scheduler.preempt/handler.log
INFO    - Total complete:   16/  95  16%  skipped:   82, failed:    3
ERROR   - bl654_dvk                 tests/kernel/fifo/fifo_usage/kernel.fifo.usage     FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/bl654_dvk/tests/kernel/fifo/fifo_usage/kernel.fifo.usage/handler.log
INFO    - Total complete:   20/  95  21%  skipped:   85, failed:    4
ERROR   - bl654_dvk                 tests/kernel/pending/kernel.objects                FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/bl654_dvk/tests/kernel/pending/kernel.objects/handler.log
INFO    - Total complete:   23/  95  24%  skipped:   87, failed:    5
ERROR   - bl654_dvk                 tests/kernel/mem_slab/mslab_threadsafe/kernel.memory_slabs.threadsafe FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/bl654_dvk/tests/kernel/mem_slab/mslab_threadsafe/kernel.memory_slabs.threadsafe/handler.log
INFO    - Total complete:   24/  95  25%  skipped:   87, failed:    6
ERROR   - bl654_dvk                 tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/bl654_dvk/tests/kernel/mem_heap/mheap_api_concept/kernel.memory_heap/handler.log
INFO    - Total complete:   35/  95  36%  skipped:   97, failed:    7
ERROR   - bl654_dvk                 tests/kernel/sched/metairq/kernel.scheduler.metairq FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/bl654_dvk/tests/kernel/sched/metairq/kernel.scheduler.metairq/handler.log
INFO    - Total complete:   37/  95  38%  skipped:   98, failed:    8
ERROR   - bl654_dvk                 tests/kernel/mem_heap/k_heap_api/kernel.k_heap_api FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/bl654_dvk/tests/kernel/mem_heap/k_heap_api/kernel.k_heap_api/handler.log
INFO    - Total complete:   40/  95  42%  skipped:  100, failed:    9
ERROR   - bl654_dvk                 tests/kernel/mem_slab/mslab/kernel.memory_slabs.api FAILED: Timeout
ERROR   - see: /tmp/zephyr/zephyr/twister-out/bl654_dvk/tests/kernel/mem_slab/mslab/kernel.memory_slabs.api/handler.log
INFO    - Total complete:   41/  95  43%  skipped:  100, failed:   10
```
And some faults:
```
*** Booting Zephyr OS build v2.7.99-1865-g5287ce8ff003  ***
Running test suite t: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x2002f1a0
E: r0/a1:  0x20204181  r1/a2:  0x3bec07b7  r2/a3:  0x09388233
E: r3/a4:  0x1d66eb4d r12/ip:  0x8cd49a2a r14/lr:  0xc3cc73a7
E:  xpsr:  0xa2d5aa00
E: Faulting instruction address (r15/pc): 0x826b8b84
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x2002d090 (main)
E: Halting system
```
-
```
*** Booting Zephyr OS build v2.7.99-1865-g5287ce8ff003  ***
thread1 take sem
thread2 take sem
Running test suite s: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x2002e3e0
E: r0/a1:  0x27833142  r1/a2:  0x1ad7bfc5  r2/a3:  0x52751949
E: r3/a4:  0xff1a2e01 r12/ip:  0x0b803a0c r14/lr:  0x7bcf7b6b
E:  xpsr:  0xc7e1c000
E: Faulting instruction address (r15/pc): 0xd0079100
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x2002c310 (main)
E: Halting system
```

However, it still seems that .gcda files are not being created so there is still no code coverage

lairdjm, 2021-11-26:
@PerMac Any ideas/suggestions for what to try here?

PerMac, 2021-11-26:
No, I don't know why. You'll have to wait for bug triage so a proper person could be assigned to your issue.
 Some tips and questions. Are the same tests failing every time?  Try to use "--subset 1/3", --subset 2/3", etc. The last part where every single test failed looks as if your board got into a corrupted state and is blocked.  
I created this issue related to coverage recently, but don't know if it is related https://github.com/zephyrproject-rtos/zephyr/issues/40624 Have you checked device.log or tried connecting e.g minicom and checking what the output is?
It is also hard for me to tell is there a single issue or multiple ones in your report. If multiple, each needs a separate task.
I was able to reproduce the issue with "hello_word" on nrf52840dk_nrf52840 and got no issues for native_posix:
```
10:59 $ scripts/twister -T samples/hello_world/ -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 --coverage -v
ZEPHYR_BASE unset, using "/home/maciej/zephyrproject2/zephyr"
Renaming output directory to /home/maciej/zephyrproject2/zephyr/twister-out.49
INFO    - Zephyr version: v2.7.99-1781-g67a265b0df7c
INFO    - JOBS: 4
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 1 test scenarios (1 configurations) selected, 0 configurations discarded due to filters.

Device testing on:

| Platform            | ID   | Serial device   |
|---------------------|------|-----------------|
| nrf52840dk_nrf52840 |      | /dev/ttyACM0    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 1/1 nrf52840dk_nrf52840       samples/hello_world/sample.basic.helloworld        PASSED (device 63.698s)

INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 skipped with 0 warnings in 70.58 seconds
INFO    - In total 1 test cases were executed, 0 skipped on 1 out of total 407 platforms (0.25%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Generating coverage files...
geninfo: WARNING: no .gcda files found in /home/maciej/zephyrproject2/zephyr/twister-out - skipping!
lcov: ERROR: no valid records found in tracefile /home/maciej/zephyrproject2/zephyr/twister-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/maciej/zephyrproject2/zephyr/twister-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/maciej/zephyrproject2/zephyr/twister-out/coverage.info
lcov: ERROR: no valid records found in tracefile /home/maciej/zephyrproject2/zephyr/twister-out/coverage.info
genhtml: ERROR: no valid records found in tracefile /home/maciej/zephyrproject2/zephyr/twister-out/coverage.info
```

lairdjm, 2021-11-26:
With code coverage on and without the CPP fix, it is always the same tests that fail yes, pretty much most of them, because they lock up in the gcov setup function and enter a fault. It does seem to be that the null pointer check using DWT is the cause of the bug I reported with `nrfjprog -f NRF52 --reset` not working, if I change it to use null pointer check using MPU then I can successfully reboot modules in this state, the MPU null pointer was switched to DWT 6 months ago so this might explain why I wasn't seeing the issue on nRF connect SDK 1.6 but am seeing it in mainline zephyr.
However, if I run with coverage with the fix, I just see endless MPU faults e.g.:
```
*** Booting Zephyr OS build v2.7.99-1865-g5287ce8ff003  ***
Running test suite m: ***** MPU FAULT *****
E:   Stacking error (context area might be not valid)
E:   Data Access Violation
E:   MMFAR Address: 0x200302c0
E: r0/a1:  0x0000001c  r1/a2:  0x0000000a  r2/a3:  0x00000014
E: r3/a4:  0x00000000 r12/ip:  0x200302e8 r14/lr:  0x0000020c
E:  xpsr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
E: Current thread: 0x2002c5a8 (main)
E: Halting system
```
Two of the other tests failed for a different reason in this configuration:
```
0x00000000 r14/lr:  0x000189f5
E:  xpsr:  0x00031600
E: Faulting instruction address (r15/pc): 0x45000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x2002c290 (idle 00)
Caught system error -- reason 0
E: ***** USAGE FAULT *****
E:   Illegal load of EXC_RETURN into PC
E: r0/a1:  0x00086a66  r1/a2:  0x00000000  r2/a3:  0x00086a66
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x000189f5
E:  xpsr:  0x00031600
E: Faulting instruction address (r15/pc): 0x45000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x2002c290 (idle 00)
Caught system error -- reason 0
E: ***** USAGE FAULT *****
E:   Illegal load of EXC_RETURN into PC
E: r0/a1:  0x00086bf5  r1/a2:  0x00000000  r2/a3:  0x00086bf5
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x000189f5
E:  xpsr:  0x00031600
E: Faulting instruction address (r15/pc): 0x45000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x2002c290 (idle 00)
Caught system error -- reason 0
E: ***** USAGE FAULT *****
E:   Illegal load of EXC_RETURN into PC
E: r0/a1:  0x00086d84  r1/a2:  0x00000000  r2/a3:  0x00086d84
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x000189f5
E:  xpsr:  0x00031600
E: Faulting instruction address (r15/pc): 0x45000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x2002c290 (idle 00)
Caught system error -- reason 0
E: ***** USAGE FAULT *****
E:   Illegal load of EXC_RETURN into PC
E: r0/a1:  0x00086f13  r1/a2:  0x00000000  r2/a3:  0x00086f13
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x000189f5
E:  xpsr:  0x00031600
E: Faulting instruction address (r15/pc): 0x45000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x2002c290 (idle 00)
Caught system error -- reason 0
E: ***** USAGE FAULT *****
E:   Illegal load of EXC_RETURN into PC
E: r0/a1:  0x000870a2  r1/a2:  0x00000000  r2/a3:  0x000870a2
E: r3/a4:  0x00000000 r12/ip:  0x00000000 r14/lr:  0x000189f5
E:  xpsr:  0x00031600
E: Faulting instruction address (r15/pc): 0x45000000
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x2002c290 (idle 00)
Caught system error -- reason 0
...
```

lairdjm, 2021-11-26:
Without code coverage, the failures are:
`tests/kernel/sched/schedule_api/kernel.scheduler.multiq`:
```
===================================================================
START - test_slice_scheduling
A
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
B
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
C
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
D
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
E
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
F
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
G
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
H
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
I
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
FAIL - test_slice_scheduling in 0.421 seconds
```
`tests/kernel/sched/schedule_api/kernel.scheduler`:
```
===================================================================
START - test_slice_scheduling
A
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
B
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
C
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
D
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
E
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
F
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
G
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
H
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
I
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false
FAIL - test_slice_scheduling in 0.421 seconds
```
`tests/kernel/mem_protect/mem_protect/kernel.memory_protection`:
```
===================================================================
START - test_alloc_kobjects
E: ***** BUS FAULT *****
E:   Precise data bus error
E:   BFAR Address: 0x1d2000b2
E: r0/a1:  0x1d2000b2  r1/a2:  0x00001331  r2/a3:  0x20003028
E: r3/a4:  0x00000020 r12/ip:  0x20002e74 r14/lr:  0x0000df99
E:  xpsr:  0x21000000
E: Faulting instruction address (r15/pc): 0x00008904
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x20001150 (test_alloc_kobjects)
Caught system error -- reason 0 0
fatal error was unexpected, aborting
```
`tests/kernel/threads/thread_apis/kernel.threads.apis`:
```
===================================================================
START - test_k_busy_wait
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/threads/thread_apis/src/main.c:594: test_k_busy_wait: dt >= k_cyc_to_us_floor64(100) is false
FAIL - test_k_busy_wait in 0.15 seconds
```

PerMac, 2021-11-26:
@lairdjm Please, we need to keep some order in the issues. It becoming impossible to follow your report. Please check in advance with github search bar, if the same issue was not already reported. If it was, please add the comment that you are also observing it on a certain board. If there is none, create a new one. E.g error with `test_k_busy_work` is already reported https://github.com/zephyrproject-rtos/zephyr/issues/40373

cfriedt, 2021-11-30:
This could also be fixed by the user if there were a `--timeout` option for twister that overrides YAML / default. 

Covered also in #40624

nashif, 2021-12-14:
> This could also be fixed by the user if there were a `--timeout` option for twister that overrides YAML / default.

timeout is not the problem. 
Running coverage on hardware is not something that is well supported or supported at all. --coverage works fine qemu and native posix targets that have HAS_COVERAGE_SUPPORT.

Before reporting this as a twister bug, please make sure you get coverage data with just building and flashing on a device, see if you get the desired output, twister is just the final step, it does not help if the platform or the test are failing due to coverage being enabled.

lairdjm, 2021-12-15:
> Why is the gcov linker section resides in a CPP #ifdef? https://github.com/zephyrproject-rtos/zephyr/blob/main/include/linker/common-rom.ld#L85 If I try code coverage with twister and forcing CPP support, the applications no longer seem to lock up in the gcov section, they still don't work because of MPU faults, but that's a bit more progress:
> 
> However, it still seems that .gcda files are not being created so there is still no code coverage

@nashif this is relevant part of the bug

nashif, 2022-02-10:
closing this given that the platform was never tested or verified as one that supports coverage and this is not a bug in twister.

closed 2022-02-10 by nashif