zephyrproject-rtos/zephyr #40694: lib: libc: newlib: tests: sram overflows

PerMac, 2021-11-26:
**Describe the bug**
The tests 
`tests/lib/newlib/thread_safety/libraries.libc.newlib.thread_safety.userspace.stress`
`tests/lib/newlib/thread_safety/libraries.libc.newlib_nano.thread_safety.userspace.stress`
causes SRAM Overflow on nrf52dk_nrf52832 and nrf9160dk_nrf9160_ns
Is there a way to optimize these tests? It is a bit problematic for us internally as we want to keep track of what gets skipped on our platforms and use "--overflow-as-errors" in our CI. 
If it is impossible, can the testcase.yaml be updated with the minimum sram requirements?  
 
**To Reproduce**
Steps to reproduce the behavior:
1. run 'scripts/twister -T tests/lib/newlib/thread_safety/ -p nrf52dk_nrf52832 -v'
2. see overflows

**Expected behavior**
Tests are optimized and no overflows are generated on the reported platforms or testcase.yaml is updated for min requirements
**Impact**
Not clear

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain Zephyr-sdk 0.13.1
 - Commit SHA or Version used zephyr-v2.7.0

**Additional context**
@stephanosio I saw that you were doing some optimization/fixes for these tests. Could you please provide some context for this issue?

stephanosio, 2021-11-26:
> If it is impossible, can the testcase.yaml be updated with the minimum sram requirements?

This is a bit tricky because that tends to vary quite significantly across different boards.

> Is there a way to optimize these tests?

https://github.com/zephyrproject-rtos/zephyr/blob/5aedcabf510846b5cc6020728a67dc9e8da3d39c/tests/lib/newlib/thread_safety/src/stress.c#L22

You can try decreasing the `THREAD_COUNT` in the stress test (e.g. to 32) to reduce the SRAM requirement.

The problem with decreasing the `THREAD_COUNT`, however, is that we may not catch the thread-safety issues within the given execution time (`TEST_INTERVAL`, which is currently set to 30 seconds); so, it may be necessary to increase the `TEST_INTERVAL` to ensure that the thread-safety issues are caught with a smaller number of threads.

This is really a tradeoff between the SRAM usage vs. test execution time; so, if this is a blocker for many, I can look into decreasing the `THREAD_COUNT` and the `TEST_INTERVAL` (and of course, the test timeout, subsequently).

https://github.com/zephyrproject-rtos/zephyr/blob/5aedcabf510846b5cc6020728a67dc9e8da3d39c/tests/lib/newlib/thread_safety/testcase.yaml#L38-L39

This is marked as a "slow" test anyway; so, making this test take over 10 minutes would not be completely unreasonable.

stephanosio, 2021-11-26:
Marking this as an enhancement since this is not a "bug" per se.