zephyrproject-rtos/zephyr #40717: twister: failure in parsing code coverage file

lairdjm, 2021-11-29:
**Describe the bug**
Whilst attempting to run code coverage on a subset of kernel tests for the nRF5340, an error was thrown in the twister python code when processing the coverage files. Support for coverage for nRF5340 was added, a twister board file was generated with the following contents:
```
- connected: true
  id: 000960105481
  platform: nrf5340dk_nrf5340_cpuapp
  product: J-Link
  runner: nrfjprog
  serial: /dev/ttyACM2
```
And the test was ran:
```
/scripts/twister --coverage --device-testing --hardware-map nrf53.yml -p nrf5340dk_nrf5340_cpuapp -T tests/kernel/sched/schedule_api/
ZEPHYR_BASE unset, using "/tmp/zephyr/zephyr"
INFO    - Zephyr version: v2.7.99-1930-g8b55275d4814
INFO    - JOBS: 16
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 6 test scenarios (6 configurations) selected, 1 configurations discarded due to filters.

Device testing on:

| Platform                 |           ID | Serial device   |
|--------------------------|--------------|-----------------|
| nrf5340dk_nrf5340_cpuapp | 000960105481 | /dev/ttyACM2    |

INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    5/   5  100%  skipped:    1, failed:    0
INFO    - 5 of 5 test configurations passed (100.00%), 0 failed, 1 skipped with 0 warnings in 278.72 seconds
INFO    - In total 128 test cases were executed, 40 skipped on 1 out of total 407 platforms (0.25%)
INFO    - 5 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Generating coverage files...
Traceback (most recent call last):
  File "/tmp/zephyr/zephyr/./scripts/twister", line 1328, in <module>
    main()
  File "/tmp/zephyr/zephyr/./scripts/twister", line 1296, in main
    coverage_tool.generate(options.outdir)
  File "/tmp/zephyr/zephyr/scripts/pylib/twister/twisterlib.py", line 3934, in generate
    self.__class__.create_gcda_files(extracted_coverage_info)
  File "/tmp/zephyr/zephyr/scripts/pylib/twister/twisterlib.py", line 3926, in create_gcda_files
    fp.write(bytes.fromhex(hexdump_val))
ValueError: non-hexadecimal number found in fromhex() arg at position 5244
```
Which gave the included error at the end.

**Expected behavior**
Twister to not fail

**Impact**
Testing with code coverage isn't working in this specific environment

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA or Version used: 8b55275d48146d34dae2cb39f2d0be07cfe00a09

lairdjm, 2021-11-29:
The code coverage file has ` a1 01 02 00 00 00 00 00 00 00 00 00 00 00No Mem available to continue dump` on the final line so this seems to be the cause

nashif, 2022-02-10:
coming back to this, you need to increase the memory to get results out of the board, it seems like you are getting error messages interfering with  the coverage info going over UART.

closed 2022-02-10 by nashif