zephyrproject-rtos/zephyr #40726: Timer as counter

sadapenore, 2021-11-29:
This is continuation with earlier ticket supporting timer on STM32.
https://github.com/zephyrproject-rtos/zephyr/issues/40247

I am refering the test_counter.c file.
static const char * const devices[] = {
.
.

#ifdef CONFIG_COUNTER_TIMER_STM32
#define STM32_COUNTER_LABEL(idx) \
	DT_LABEL(DT_INST(idx, st_stm32_counter)),
#define DT_DRV_COMPAT st_stm32_counter
	DT_INST_FOREACH_STATUS_OKAY(STM32_COUNTER_LABEL)
#undef DT_DRV_COMPAT
#undef STM32_COUNTER_LABEL
#endif

.
.
.
}

I dont see reference to st_stm32_counter.

The resultant array value is null for this location of devices[].

It will be good that you provide link of a complete branch which is having all the code.
I also have update the overlay file for the sequence of location of st,prescaler.

erwango, 2021-11-29:
It would be good that you look at https://docs.zephyrproject.org/latest/guides/dts/api-usage.html#node-identifiers
`st_stm32_counter` is `st,stm32-counter` used through DT API.

erwango, 2021-11-29:
I'm closing this issue. Please reach out to us in discord or gh discussion if you have other questions.

closed 2021-11-29 by erwango