zephyrproject-rtos/zephyr #40740: tests: logging: test case logging/log_msg2/logging.log_msg2_64b_timestamp failed on qemu_cortex_a9

enjiamai, 2021-11-29:
**Describe the bug**
The logging test:  tests/subsys/logging/log_msg2/logging.log_msg2_64b_timestamp failed after this commit:
99b7e9d8603f42f5ed8a8691711ba77416a3238b  boards: arm: qemu_cortex_a9: change board's timing parameters

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? 
   qemu_cortex_a9
 - What have you tried to diagnose or workaround this issue?
   without CONFIG_LOG_TIMESTAMP_64BIT=y, the test case will pass.

**To Reproduce**
Steps to reproduce the behavior:
1. source zephyr-env.sh
2. twister -T tests/subsys/logging/log_msg2/ -p qemu_cortex_a9
    or
    west build tests/subsys/logging/log_msg2/ -b qemu_cortex_a9 -p always -t run -DCONFIG_LOG_TIMESTAMP_64BIT=y
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
twister output:
```
mm@mm-NUC8i7BEH:~/zephyrproject/zephyr$ twister -T tests/subsys/logging/log_msg2/ -p qemu_cortex_a9
Renaming output directory to /home/mm/zephyrproject/zephyr/twister-out.13
INFO    - Zephyr version: v2.7.99-1413-g99b7e9d8603f
INFO    - JOBS: 12
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testcase list...
INFO    - 5 test scenarios (5 configurations) selected, 0 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:    3/   5  60%  skipped:    0, failed:    0
ERROR   - qemu_cortex_a9            tests/subsys/logging/log_msg2/logging.log_msg2_64b_timestamp FAILED: Failed
ERROR   - see: /home/mm/zephyrproject/zephyr/twister-out/qemu_cortex_a9/tests/subsys/logging/log_msg2/logging.log_msg2_64b_timestamp/handler.log
INFO    - Total complete:    4/   5  80%  skipped:    0, failed:    1
ERROR   - qemu_cortex_a9            tests/subsys/logging/log_msg2/logging.log_msg2_fp_64b_timestamp FAILED: Failed
ERROR   - see: /home/mm/zephyrproject/zephyr/twister-out/qemu_cortex_a9/tests/subsys/logging/log_msg2/logging.log_msg2_fp_64b_timestamp/handler.log
INFO    - Total complete:    5/   5  100%  skipped:    0, failed:    2
INFO    - 3 of 5 test configurations passed (60.00%), 2 failed, 0 skipped with 0 warnings in 7.37 seconds
INFO    - In total 50 test cases were executed, 0 skipped on 1 out of total 406 platforms (0.25%)
INFO    - 5 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing xunit report /home/mm/zephyrproject/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/mm/zephyrproject/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```
and the content of the handler.log:
```
Running test suite test_log_msg2
===================================================================
START - test_log_msg2_0_args_msg
.
.
.
===================================================================
START - test_mode_size_plain_str_data
 PASS - test_mode_size_plain_str_data in 0.1 seconds
===================================================================
START - test_mode_size_str_with_2strings

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/logging/log_msg2/src/main.c:343: get_msg_validate_length: (len not equal to exp_len)
Unexpected message length 16 (exp:8)
 FAIL - test_mode_size_str_with_2strings in 0.1 seconds
===================================================================
START - test_saturate
 PASS - test_saturate in 0.1 seconds
===================================================================
Test suite test_log_msg2 failed.
===================================================================
PROJECT EXECUTION FAILED

```
**Environment (please complete the following information):**
 - OS: Linux,
 - Toolchain Zephyr SDK, 0.13.1
 - Commit SHA or Version used: 99b7e9d8603f42f5ed8a8691711ba77416a3238b

**Additional context**

nordic-krch, 2021-11-30:
It's related (or deplicate) to #39978

enjiamai, 2021-11-30:
@nordic-krch got it, close it, thanks!

closed 2021-11-30 by enjiamai