zephyrproject-rtos/zephyr #40759: Bluetooth: host: Improper restore of CCC values and handling Service Change indication when bonded peer reconnects

MarekPieta, 2021-11-30:
**Describe the bug**
Service Changed indication and CCC update on GATT server should be called after BLE security is reestablished (so that we know that we reconnected with peer that was previously bonded). Currently it's called right after the connection is reestablished which is not proper behavior.

**Expected behavior**
CCC update and Service changed indication should be delayed until the connection is secured.

**Impact**
The issues could lead to using improper values of CCC by the peripheral before the connection is secured. The Service Changed indication may be sent to a peer that only impersonates the bonded GATT Client. In that case it would not be properly delivered to actually bonded Central which could lead to inconsistency of cached GATT discovery data.

ilhanates, 2022-02-08:
This fix removes `update_ccc` call for good reason but it is called only when there is change in encryption status on l2cap chan anymore. This casuses Gatt service changed indication to be not sent after reconnect (bond), if Gatt service changes while bonded peer is disconnected. So service changed is saved properly but not restored in this case. This makes GATT/SR/GAS/BV-01-C test to fail always in my local, even though it is passing in host conformance CI on second try.

ilhanates, 2022-02-10:
Removing `update_ccc ` left `data.sec` is initialized to  `BT_SECURITY_L1 _(Level 1: No encryption and no authentication.)`_ and later steps, it is used to decide to set security automatically to higher level with option `CONFIG_BT_GATT_AUTO_SEC_REQ` which is selected as default. This prevents to set security in need.

seko-nordic, 2022-02-10:
I think this change is incomplete. If we require encryption we need to automatically request it. After this change following check is always false, so encryption is never requested any more https://github.com/zephyrproject-rtos/zephyr/blob/0a1f553dc23ca9c2b42092dbba44ab0b3204990b/subsys/bluetooth/host/gatt.c#L4922 (refer to data.sec set to L1 [here](https://github.com/zephyrproject-rtos/zephyr/blob/0a1f553dc23ca9c2b42092dbba44ab0b3204990b/subsys/bluetooth/host/gatt.c#L4882), before it was updated in the line that was removed by this change).

This change leads to PTS failures in GATT/SR/GAS/BV-01-C and GATT/SR/GAS/BV-07-C. This is blocker since we cannot pass conformance any more.

carlescufi, 2022-02-10:
@seko-nordic thanks for the comment. Is this issue a good description of what the bug actually is in the code though? If not, please open a new issue instead of reopening this one.

seko-nordic, 2022-02-10:
I think we need to revert this change and think more how to fix it properly. That's why I reopened it.

carlescufi, 2022-02-10:
> I think we need to revert this change and think more how to fix it properly. That's why I reopened it.

OK, in that case let's leave this open, thanks.

PR #40767 (2021-12-02): Bluetooth: host: Update CCC after connection is encrypted
The CCC update should be done after the connection is encrypted. Before the connection is encrypted, the GATT Server does not know if the reconnecting peer was actually bonded.

Fixes: #40759
commit 0a1f553d (2021-12-02): Bluetooth: host: Remove CCC update from GATT connected callback

The CCC update should be done after the connection is encrypted.
Before the connection is encrypted, the GATT Server does not know
if the reconnecting peer was actually bonded.

Fixes: #40759
PR #42694 (2022-02-11): bluetooth: gatt: Fix Service Changed indication sending
Revert fix, because if caused issues with tests. This needs further investigation until the change will be introduced to ensure that tests would not fail.

Fixes #40759
closed 2022-02-11 by carlescufi