zephyrproject-rtos/zephyr #40763: gpio_stm32_port_get_raw()   reads only Input Port (Input Data Reg), it should also read Output Port (Output Data Reg)

sumit404-work, 2021-11-30:
**Is your enhancement proposal related to a problem? Please describe.**
The driver function[ gpio_stm32_port_get_raw() at zephyrproject\zephyr\drivers\gpio\gpio_stm32.c](https://github.com/zephyrproject-rtos/zephyr/blob/8d40928b2a1a1fa4c445f4b18dbd7252a7048a93/drivers/gpio/gpio_stm32.c#L369) 
 ```
static int gpio_stm32_port_get_raw(const struct device *dev, uint32_t *value)
{
	const struct gpio_stm32_config *cfg = dev->config;
	GPIO_TypeDef *gpio = (GPIO_TypeDef *)cfg->base;

	*value = LL_GPIO_ReadInputPort(gpio);

	return 0;
}
```

 at zephyrproject\zephyr\drivers\gpio\gpio_stm32.c reads only 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

erwango, 2021-11-30:
@sumit404-work It seems that what you are requesting is a new function to GPIO API, is that correct ?

sumit404-work, 2021-11-30:
> @sumit404-work It seems that what you are requesting is a new function to GPIO API, is that correct ?

Yes

erwango, 2021-11-30:
ok, then if you want your request to be considered, please rework is to address the generic GPIO API.

sumit404-work, 2021-11-30:
> ok, then if you want your request to be considered, please rework it to address the generic GPIO API.

Hi @erwango I have updated the name and description to generic GPIO API.

mnkp, 2021-11-30:
The requested functionality is rarely asked for and needed by the real life applications. The application that does require to know the state of the output ports can keep track of the state locally. This takes little resources and is faster than going through the full driver API call.

I believe we should extend GPIO API only if this issue gets more traction / there are more users that need the feature. If that's not the case I propose to reject the feature request.

erwango, 2021-11-30:
Thanks for the feedback @mnkp. I propose to let the PR open for a time and see if more people concur.

Also, I would like to add that if the change is not made in the gpio API, it can't be implemented in drivers, but applications still have the possibility to directly access GPIO resources and get the expected output ports states (for instance in STM32, using LL API)

santakrishna, 2022-10-26:
i tried initializing like this
 gpio_pin_configure_dt(&LED5_spec, GPIO_OUTPUT_ACTIVE | GPIO_INPUT)

besmarsh, 2023-01-13:
I think this would be a useful feature, and one that doesn't seem too disruptive to implement. While reading output ports is obviously not as common as reading input ports, there are valid use-cases and tracking the state through an application can be messy. The alternative of directly accessing GPIO resources with a low-level driver API is not portable.

mlp6, 2023-03-01:
This would be a useful feature.  I use Zephyr in classes that I teach, and having the ability to explicitly get a pin status, regardless of being input/output would be great, especially for outputs that are toggled, and you might not be aware of things that previously changed state.

If this isn't considered as an API change, then it would be good for this function to return an error when trying to get the state of an output pin.  Right now, this will return `0`, regardless of the actual output pin state, which is confusing because you may think that the output state is actually LOW.

besmarsh, 2023-03-07:
@mnkp @erwango Any updated thoughts on this since it's had a couple more requests?