zephyrproject-rtos/zephyr #40775: stm32: mutli-threading broken after #40173

erwango, 2021-11-30:
**Describe the bug**
Following merge of #40173, some multi-threading use cases are not working anymore.
This change aimed at introducing a lock on uart tx in order to avoid conflicts between competitive calls to uart_poll and fifo_fill.
While it fixed the targeted issue, it introduced a regression in multi-thread use cases.

Regression can be observed with the following samples:
 samples/kernel/metairq_dispatch/sample.kernel.metairq_dispatch
samples/philosophers/sample.kernel.philosopher
samples/philosophers/sample.kernel.philosopher.fifos
samples/philosophers/sample.kernel.philosopher.lifos
samples/philosophers/sample.kernel.philosopher.same_prio
samples/philosophers/sample.kernel.philosopher.semaphores
samples/philosophers/sample.kernel.philosopher.stacks
samples/philosophers/sample.kernel.philosopher.static
samples/subsys/portability/cmsis_rtos_v1/philosophers/sample.portability.cmsis_rtos_v1.philosopher
and so on ...

**Logs and console output**
Info thread on `sample.kernel.philosopher`  reports the following:
```
0x08005100 in atomic_cas (new_value=1, old_value=0, target=0x20000020 <uart_stm32_data_0+8>) at /local/mcu/zephyrproject/zephyr/include/sys/atomic_builtin.h:40
40		return __atomic_compare_exchange_n(target, &old_value, new_value,
(gdb) info threads
  Id   Target Id                                                  Frame 
* 1    Thread 536871920 (Name: Philosopher 5, prio:-2,useropts:4) 0x08005100 in atomic_cas (new_value=1, old_value=0, target=0x20000020 <uart_stm32_data_0+8>)
    at /local/mcu/zephyrproject/zephyr/include/sys/atomic_builtin.h:40
Info : Getting thread 536871744 reg list
  2    Thread 536871744 (Name: Philosopher 4, prio:-1,useropts:4) 0x080013c0 in arch_irq_unlock (key=0)
    at /local/mcu/zephyrproject/zephyr/include/arch/arm/aarch32/asm_inline_gcc.h:95
Info : Getting thread 536871568 reg list
  3    Thread 536871568 (Name: Philosopher 3, prio:0,useropts:4)  0x080013c0 in arch_irq_unlock (key=0)
    at /local/mcu/zephyrproject/zephyr/include/arch/arm/aarch32/asm_inline_gcc.h:95
Info : Getting thread 536871392 reg list
  4    Thread 536871392 (Name: Philosopher 2, prio:1,useropts:4)  0x080013c0 in arch_irq_unlock (key=0)
    at /local/mcu/zephyrproject/zephyr/include/arch/arm/aarch32/asm_inline_gcc.h:95
Info : Getting thread 536871216 reg list
  5    Thread 536871216 (Name: Philosopher 1, prio:2,useropts:4)  0x080013c0 in arch_irq_unlock (key=0)
    at /local/mcu/zephyrproject/zephyr/include/arch/arm/aarch32/asm_inline_gcc.h:95
Info : Getting thread 536871040 reg list
  6    Thread 536871040 (Name: Philosopher 0, prio:3,useropts:4)  0x08005118 in LL_USART_IsActiveFlag_TXE (USARTx=0x40013800)
    at /local/mcu/zephyrproject/modules/hal/stm32/stm32cube/stm32l4xx/drivers/include/stm32l4xx_ll_usart.h:3161
Info : Getting thread 536872096 reg list
  7    Thread 536872096 (Name: idle 00, prio:40,useropts:1)       arch_cpu_idle () at /local/mcu/zephyrproject/zephyr/arch/arm/core/aarch32/cpu_idle.S:110
```

Philosopher 0 (prio:3) is preempted in `uart_stm32_poll_out` during TX (while waiting for flag TXE: `LL_USART_IsActiveFlag_TXE`), tx_lock is granted
Philosopher 5 (prio:-2) is running but stuck in `uart_stm32_poll_out`, waiting for `tx_lock` to be released, which won't happen as it is taken by lower prio thread.
-> Deadlock

**To Reproduce**
Run one of the above on any STM32 board

**Expected behavior**
These samples should be passed

**Impact**
Dead-lock situations in multi-threaded contexts

**Environment (please complete the following information):**
 - Toolchain: Zephyr SDK 0.13.1
 - Commit SHA: Starting d42cef17b06465393462210a224fdc5cc58b1490

jdascenzio, 2021-11-30:
Recently, I made a patch (#40782) which solve, I think, the problem of lock you find.
I wanted to wait a bit before making a pull request to ensure that I don't found some other problems...

erwango, 2021-11-30:
I had a try with #40782 on `samples/philosophers`, but unfortunately, this doesn't fix the issue.
I didn't investigate, but I guess this is because the issue reported here doesn't involve IRQ API but only several threads doing poll_out calls.

dcpleung, 2021-11-30:
I think it is low-level enough that a generic, API-level solution would not work on all drivers, and might require some drivers some significant gymnastics to work correctly.

Since you have mentioned the NRF driver, one significant difference is that NRF driver does `k_msleep()` if it cannot hold the TX lock. Maybe it's something to be implemented in the STM32 driver?

erwango, 2021-12-02:
> I think it is low-level enough that a generic, API-level solution would not work on all drivers, and might require some drivers some significant gymnastics to work correctly.

@dcpleung, ok, right let's fix this at driver level. Then, what about adding a dedicated test under `test/drivers/uart/` ?
It would be useful for non regression purpose as well as providing some light on this issue that is likely to impact all drivers.

DariusBabrauskas, 2021-12-02:
It seems to me, that  k_msleep() maybe can't work in interrupt context or it danger.  NRF driver used  !k_is_in_isr()  and in interrupt context doing force lock-unlock, but it is not good idea. Better remember last state.  
How about nested interrupts?

erwango, 2021-12-02:
> It seems to me, that k_msleep() maybe can't work in interrupt context or it danger. NRF driver used !k_is_in_isr() and in interrupt context doing force lock-unlock, but it is not good idea. Better remember last state. How about nested interrupts?

@DariusBabrauskas, can you check #40782, where @jdascenzio seems to solve the issue in an alternate way ?

DariusBabrauskas, 2021-12-02:
> @DariusBabrauskas, can you check #40782, where @jdascenzio seems to solve the issue in an alternate way ?

@erwango I reviewed only code.  It looks fine in poll function.  But I not familiar with CONFIG_PM part. 
> I had a try with #40782 on samples/philosophers, but unfortunately, this doesn't fix the issue.

Is was  test  "samples/philosophers" good before  merge of #40173 ?

I use zephyr 2.5, I tested similar variant. But  @jdascenzio have good idea to use double  check if(LL_USART_IsActiveFlag_TXE(UartInstance)) to avoid cpu lock/unlock storm.

```
static void uart_stm32_poll_out(const struct device *dev, unsigned char c)
{
	USART_TypeDef *UartInstance = UART_STRUCT(dev);
	int key;

	while (1) {
		key = irq_lock();//prevent race condition if nested interrupts  or thread switch 
		if(LL_USART_IsActiveFlag_TXE(UartInstance)){

			LL_USART_ClearFlag_TC(UartInstance);
			LL_USART_TransmitData8(UartInstance, (uint8_t)c);

			irq_unlock(key);
			break;
		}
		irq_unlock(key);
		if (!k_is_in_isr() && key == 0){
			//k_msleep(1);//we in not interrupt context //but got output latency, so disable it
			k_yield();//yield execution to another thread of the same or higher priority. Its better for software, but not output pretty (some symbols  from shell some from kprint)
		}
	}
}
```

erwango, 2021-12-02:
@DariusBabrauskas, thanks. Would you mind to formerly review #40782 ?

DariusBabrauskas, 2021-12-02:
> @DariusBabrauskas, thanks. Would you mind to formerly review #40782 ?
If I understood ok.
 I reviewed   #40782 . It looks fine in poll function. But I not familiar with CONFIG_PM part.
It must work  like before pull   https://github.com/zephyrproject-rtos/zephyr/pull/40173    without dead lock and without lost tx bytes.

erwango, 2021-12-03:
> > @DariusBabrauskas, thanks. Would you mind to formerly review #40782 ?
> > If I understood ok.

I actually meant review **in** #40782. But now that we've got one more review that's ok.
Reviewing PR is open to everyone, and that's encouraged as this helps maintainers' duty.

erwango, 2021-12-10:
@jdascenzio I'm sorry, but I have to reopen this as we're getting regressions on other tests:
- tests/kernel/sched/metairq/kernel.scheduler.metairq
- tests/kernel/profiling/profiling_api/kernel.common.profiling
- tests/kernel/sched/schedule_api/kernel.scheduler
- tests/kernel/sched/schedule_api/kernel.scheduler.multiq
- tests/kernel/tickless/tickless_concept/kernel.tickless.concept
- tests/kernel/profiling/profiling_api/kernel.common.profiling
- tests/kernel/workq/work_queue/kernel.workqueue

I'm starting analysis, but would you mind having a look on your side ?

jdascenzio, 2021-12-10:
The problems seem to come from calling k_yield in the poll_out function. The tests don't expect (and surely do not want) that the current thread change when writing a message with printk (which uses poll_out). So, it's maybe a bad idea to use k_yield function in poll_out: k_yield is only used in optimization to force a thread change if the usart send stack is full.

DariusBabrauskas, 2021-12-10:
I am testing key == 0 in case:
irq_lock();
uart_stm32_poll_out();
irq_unlock() ;

My full code.  __DSB  I inserted into  irq_lock(), irq_unlock() 
Note: possible  uart_stm32_poll_out can't be used before kernel init.
```
static void uart_stm32_poll_out(const struct device *dev,
					unsigned char c)
{
	USART_TypeDef *UartInstance = UART_STRUCT(dev);
	int key;

	while (1) {
		if(LL_USART_IsActiveFlag_TXE(UartInstance))
		{
			key = irq_lock();
			//__DSB();
			if(LL_USART_IsActiveFlag_TXE(UartInstance)){

				LL_USART_ClearFlag_TC(UartInstance);
				LL_USART_TransmitData8(UartInstance, (uint8_t)c);

				irq_unlock(key);
				break;
			}
			irq_unlock(key);
			//__DSB();
		}
		if (!k_is_in_isr() && key == 0){
			//k_msleep(1);//we in not interrupt context //but got output latency, so disable it
			k_yield();//yield execution to another thread of the same or higher priority. Its better for software, but not output pretty (some symbols  from shell some from kprint)
		}
	}
}
```

DariusBabrauskas, 2021-12-10:
> The problems seem to come from calling k_yield in the poll_out function. The tests don't expect (and surely do not want) that the current thread change when writing a message with printk (which uses [...]

Yes, this tests was not writed to switch tasks. 
It seem to me that, need to do with  config define, which defaul must be disabled: Exmp:

```
#if CONFIG_USE_YIELD_IN_UART_POLL==1
		if (!k_is_in_isr() && key == 0){
			//k_msleep(1);//we in not interrupt context //but got output latency, so disable it
			k_yield();//yield execution to another thread of the same or higher priority. Its better for software, but not output pretty (some symbols  from shell some from kprint)
		}
#endif
```

erwango, 2021-12-10:
> Yes, this tests was not writed to switch tasks.
It seem to me that, need to do with config define, which default must be disabled: Exmp:

I'd prefer we first fix the default behavior and make sure current code doesn't bring any obvious regressions.
Then, please submit an enhancement proposal with clean rationale.

DariusBabrauskas, 2021-12-10:
Yes @erwango   I agree.

DariusBabrauskas, 2021-12-10:
One note:
If defined CONFIG_PRINTK_SYNC=y then prink internally  use k_spin_lock and printing all message without interrupts. 
It seems to me that this tests must be use this config define, because uart_nrfx_uart.c  use  **k_msleep(1);** too, so task switched too.

```
void vprintk(const char *fmt, va_list ap)
{
	struct out_context ctx = { 0 };
#ifdef CONFIG_PRINTK_SYNC
	k_spinlock_key_t key = k_spin_lock(&lock);
#endif

	cbvprintf(char_out, &ctx, fmt, ap);

#ifdef CONFIG_PRINTK_SYNC
	k_spin_unlock(&lock, key);
#endif
}
```

erwango, 2021-12-10:
> One note:
If defined CONFIG_PRINTK_SYNC=y then prink internally use k_spin_lock and printing all message without interrupts.
It seems to me that this tests must be use this config define, because uart_nrfx_uart.c use k_msleep(1); too, so task switched too.

Let's fix the regression in the driver first.
Then we can suggest tests evolution and then get back to driver optimizations.

erwango, 2021-12-13:
> Yes, this tests was not writed to switch tasks.

@DariusBabrauskas, I had some thinking about this and I wonder if switching task on poll_out would make any sense.
When multiple threads are using same console output, switching tasks on poll_out would always have the consequence of mixing the chars at output, and hence application output would be garbled.
I can't think of one application where mixing threads messages (hence useful information) to a garbled output would be useful, even with a performance gain.

jdascenzio, 2021-12-13:
> @DariusBabrauskas, I had some thinking about this and I wonder if switching task on poll_out would make any sense. When multiple threads are using same console output, switching tasks on poll_out [...]

The new thread that run after a k_yield is not necessarily a thread that writes a message to the console.

DariusBabrauskas, 2021-12-14:
Poll output can be switched by preempted task with bigger priority. () 
But  k_yield  is better, because it release cpu time for other task with same priority. (Possible this task must handle received data, if not handle data can be lost).

PR #40782 (2021-12-08): uart_stm32: fix a dead lock on poll_out if it's called in interrupt
If the function poll_out is called in an interrupt with a higher
priority than the usart priority, the function could lock forever. In
fact, the lock variable is never unset if the usart interrupt is no
longer called. So now, we don't use the lock variable if poll_out is
called in interrupt.

Fixes #40775
PR #41116 (2021-12-17): drivers/uart: stm32: don't call k_yield on poll_out
Some tests like:
    tests/kernel/sched/metairq/kernel.scheduler.metairq
    tests/kernel/profiling/profiling_api/kernel.common.profiling
    tests/kernel/sched/schedule_api/kernel.scheduler
    tests/kernel/sched/schedule_api/kernel.scheduler.multiq
    tests/kernel/profiling/profiling_api/kernel.common.profiling
    tests/kernel/workq/work_queue/kernel.workqueue

don't support that the current thread change when writing a message with
printk (which uses poll_out). So, we remove the call to k_yield which is
useful only for optimizing cpu usa [... cut]
closed 2021-12-17 by cfriedt