zephyrproject-rtos/zephyr #40776: HCI_USB with nRF52840 dongle disconnect after 30 s

Vicpastaga, 2021-11-30:
Hi,

**Describe the bug**

I try to connect simultaneous 12 BLE devices to a Linux PC with Bluez.
Commercial controllers cannot go beyond 5 connections. I am using an nRF52840 dongle.

I use the hci_usb sample flashed by dfu.

On the PC, the 12 connections are made without problems with bluetoothctl.
Only after about 30s, a device disconnects.
It is not necessarily the same and it is possible to reconnect afterwards but this disconnection is a problem for my application.

It is not at the level of the devices, nor of the bluetoothctl commands since everything works with a commercial BLE controller (only for 5 devices).

Before the disconnection, the bluetoothctl "info" command line indicates that the device is connected and displays the list of all BLE services. 

I also tested with the nRF52832 DK boardand the hci_uart example for the same result.

Is there a way to solve this problem? 

**Logs and console output**

With btmon, I get the disconnection reason : Remote User Terminated Connection
HCI Command: Disconnect (0x01|0x0006) plen 3                                            
        Handle: 0
        Reason: Remote User Terminated Connection (0x13)

**Environment:**

OS: Linux Mint, bluez 
Toolchain: Zephyr SDK: Zephyr from master or v2.7.99.

**Additional context**

None

jfischer-no, 2021-11-30:
I can confirm it, but did not had time to investigate it, it is somehow related to USB device support. zephyr-v2.7.0 does not seem to be affected.

cvinayak, 2021-11-30:
As it is Remote User Terminated Connection, I suspect a ATT request that was not delivered to the host BT stack or a response from the host BT stack did not reach the remote peer, which then  initiated the disconnect on the 30s ATT transaction timeout.

Vicpastaga, 2021-12-01:
Thank for your replies.

I try with zephyr-v2.7.0 version and it is exactly the same behavior with a disconnection on 12 after 30s.
For hci_usb and hci_uart.

I also tried a multitude of different configuration for prj.conj but without success.

jfischer-no, 2021-12-01:
> I try with zephyr-v2.7.0 version and it is exactly the same behavior with a disconnection on 12 after 30s.

Please recheck, zephyr-v2.7.0 works fine.

git bisect shows 02c32d44746be8c2f45bcea375dfe37693ee5ae3 as cause. Host periodically requests Set Feature DEVICE_REMOTE_WAKEUP, suspends, resumes, requests Clear Feature; corresponding suspend/resume logic in _subsys/usb/class/bluetooth.c_ seems to have a bug.

Vicpastaga, 2021-12-01:
I recheck, I got the same behaviour again with v2.7.0 even with a different Linux PC.
One in 12 devices is disconnected 30 s after the first connection with the reason : Remote User Terminated Connection (0x13).
This device is connected a few seconds later and then there is no disconnection ever again.

Vicpastaga, 2021-12-01:
You are right, that's work with v2.7.0. 
My problem is due to an overflow on BLE stack.
When I wait a successful connection before launching another, I no longer have a disconnection.
Thank you for your time and contribution.

PR #40835 (2021-12-14): usb: bluetooth: do not start transfers when USB bus is suspended 
Do not start transfers when USB bus is suspended.
If the bus is suspended, initiate remote-wakeup and
wait until the bus is operational.

Allow to disable option USB_DEVICE_REMOTE_WAKEUP.

Fixes: #40776
closed 2021-12-14 by cfriedt