zephyrproject-rtos/zephyr #40795: Timer signal thread execution loop break SMP on ARM64

fkan-ec, 2021-12-01:
See code at ([https://github.com/fkan-ec/zephyr.git](https://github.com/fkan-ec/zephyr.git)) pulled from Zephyr latest main, the problem was replicated at multiple release points including v2.7.

The samples/old directory contains main function which spins up multiple threads that takes signals from multiple timer routines. The timer routine regularly signal the threads to  run a simple multiplication routine. The program would run a bit and one of the thread would crash from stack overflow. Adjusting stack size does not seem to remedy the problem. Enable CONFIG_THREAD_ANALYZER seems to deter the problem from happening.

Primarily uses the qemu_cortex_a53_smp platform to test this code, but also tried this on actual ARMv8 hardware.

Steps to reproduce the behavior: (0.13.2 SDK)
1. west build -p -b qemu_cortex_a53_smp samples/old
2. west -b build -t run

**Expected behavior**
One of the thread would stack overflow
uart:~$ E: ELR_ELn: 0x0000000040016670
E: ESR_ELn: 0x0000000056000002
E:   EC:  0x15 (Unknown)
E:   IL:  0x1
E:   ISS: 0x2
E: TPIDRRO: 0x020000004002b030
E: x0:  0x0000000040083d78  x1:  0x0000000000000000
E: x2:  0x0000000000000000  x3:  0x0000000000000000
E: x4:  0x00000000400284f0  x5:  0x0000000000000001
E: x6:  0x0000000000000000  x7:  0x0000000000000001
E: x8:  0x0000000000000002  x9:  0x00000000400284f0
E: x10: 0x0000000000000000  x11: 0x0000000040075a50
E: x12: 0x000000004001a2c8  x13: 0xaaaaaaaaaaaaaaaa
E: x14: 0xaaaaaaaaaaaaaaaa  x15: 0x0000000000000000
E: x16: 0x0000000000000000  x17: 0x0000000000000000
E: x18: 0xaaaaaaaaaaaaaaaa  x30: 0x0000000040014b64
E: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 3
E: Current thread: 0x40027040 (THREAD_A)
E: Halting system

**Environment (please complete the following information):**
 - Ubuntu 20.04 x86 server 
 - Centos 8 armv8 server

fkan-ec, 2021-12-01:
Changing the qemu a53 smp configuration to 4 cores made the issue much easier to reproduce.

venkatesh-eq, 2021-12-01:
Just to add to what feng has provided. It isnt exactly a stack overflow. We have enabled stack canary checks that detecting some sort of corruption.

venkatesh-eq, 2021-12-06:
Just reminder to see if someone in the community can help review the code posted at: https://github.com/fkan-ec/zephyr.git 
With aarch64 and with SMP this seems to happen pretty easily. Seems like a zephyr scheduler issue. Wondering if no-one runs zephyr on aarch64 with SMP?

carlocaione, 2021-12-07:
The code is very (very) far from a small reproducer. Please, try to come up with something more manageable that we can easily take a look without having to study hundreds of lines of custom code.

fkan-ec, 2021-12-12:
I pushed in a few small changes that clean up the code a bit. Most of the code is replication of threads and timers. The main code consist of 5 main threads and 10 timers (all identical). Each thread has two timers which fires off a signal which the thread catches with k_poll and process. Each timer uses 2ms duration to fire off signal to its respective thread using (k_poll_signal_raise). The thread processing loop is just a simple loop with dumb multiplications. The thread priority varies between negative to positive number. Setting the prio to the same value seems to make the problem go away.

When I tried to clean up the code to a single thread and two timers, the problem does not seem to appears. The problem starts to appear with two threads and four timers, but the reproduce duration varied and not consistent. 

It would be good if you guys have some insight or suggestions that could help me isolate this problem further.

fkan-ec, 2021-12-28:
I manage to reduce the code to five threads and five timers. It seems that with cooperative threads, the stack overflow does not happen. If I change a single thread to preemptive. The preemptive thread always throw stack overflow.

carlocaione, 2021-12-28:
> I manage to reduce the code to five threads and five timers.

Do you have the code somewhere?

carlocaione, 2021-12-28:
@fkan-ec again, if you want me to look at this you have to provide a minimal reproducer, otherwise I can give you my hourly rate as consultant and then I can debug your code as well. But, if you want a non-paid maintainer to look at an issue, you must be sure to not waste the non-paid maintainer time on unrelated code.

Looking at your code that means:
- Cleanup the code and have it properly formatted
- Delete all the unnecessary and dead code (i.e. `APP_TIMER_ONE_SHOT_START_US` is not used anywhere, delete it)
- Keep the custom glue code at minimum (i.e. why all the unnecessary enums and macros?)
- When possible use the Zephyr provided helpers instead of custom ones

and most importantly:
- Be sure that only the code related to the bug is present in the reproducer (i.e. the content of `thread_timeout_handler` is not important and the bug can be reproduced without that using an empty function, so why is that code still in there?)

fkan-ec, 2021-12-30:
I did some more cleanup. At this point, the timer is removed. The main thread would continuously send signals to the three threads (all cooperative). At this point, it seems that k_poll has some strangeness that causes corruption.

https://github.com/fkan-ec/zephyr

venkatesh-eq, 2022-01-06:
Is Zephyr commercially deployed with ARM64 + SMP? Is it part of the supported platform for Zephyr? There seems to be a fundamental problem with pre-emption that we were able to reproduce pretty easily. surprising to me that no one has complained so far.

carlocaione, 2022-01-06:
> Is Zephyr commercially deployed with ARM64 + SMP? Is it part of the supported platform for Zephyr? 

It is supported by Zephyr and not sure if there is any commercially deployed platform.

> There seems to be a fundamental problem with pre-emption that we were able to reproduce pretty easily. surprising to me that no one has complained so far.

I would not say that this is pretty easily reproducible given that all the Zephyr unit tests are all successful, including the ones testing SMP. This bug is hitting some kind of corner case that it is not straightforward to fix. I tried to look into this but my free time is very limited.

carlocaione, 2022-01-06:
I pushed here https://github.com/carlocaione/zephyr/commit/c6cecbeee6c23b3183b5319e06afbafdc33de739 a smaller and better reproducer for the issue in case anyone else wants to work on this.

venkatesh-eq, 2022-01-11:
Can we bump up the priority of the issue, while we are debugging this? if SMP, premptive scheduler and ARM64 are officia support for features for Zephyr, that seems to be broken.

carlocaione, 2022-01-11:
The problem seems related to thread migration from one CPU to another. You can workaround this issue by either selecting `CONFIG_SCHED_CPU_MASK_PIN_ONLY` or disabling IPI. I'm still not sure what is the root cause of this issue.

fkan-ec, 2022-01-12:
[image]
This is a trace of crash on the thread 2. Seems the poll was entered multiple times

Crash log:
E: ELR_ELn: 0x000000004001b010
E: ESR_ELn: 0x000000008600000f
E:   EC:  0x21 (Instruction Abort taken without a change in Exception level.)
E:   IL:  0x1
E:   ISS: 0xf
E: FAR_ELn: 0x000000004001b010
E: TPIDRRO: 0x010000004001b1c0
E: x0:  0x0000000000000000  x1:  0x00000000400033e0
E: x2:  0x0000000060000045  x3:  0x00000000400033e0
E: x4:  0x0000000041067ab0  x5:  0x000000004000be68
E: x6:  0x0000000000000001  x7:  0x000000004001b010
E: x8:  0x0000000041067ab0  x9:  0x400197004000bf84
E: x10: 0x0000000000000003  x11: 0x0032646165726874
E: x12: 0x0000000000000000  x13: 0x3000000000000000
E: x14: 0x0032646165726874  x15: 0x0000000000000000
E: x16: 0x0197003000000000  x17: 0x0000000000000340
E: x18: 0x0000000041067ab0  x30: 0x000000004000bd84
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 2
E: Current thread: 0x40019700 (thread2)
E: Halting system
qemu-system-aarch64: terminating on signal 2

fkan-ec, 2022-01-12:
Continue the previous comment, it seems in one case the thread 2 enter poll from one CPU but exit on another CPU.
[image]

carlocaione, 2022-01-12:
> Continue the previous comment, it seems in one case the thread 2 enter poll from one CPU but exit on another CPU.

This should be ok. It is possible that the task was migrated from one core to another. The multiple enter instead looks suspicious indeed.

andyross, 2022-01-16:
OK, I found it, though I'm going to have to leave it to you guys to fix it as I don't have ARM assembly in my fingers and would need to spend too long reading docs.  It's just a misordering, fundamentally.  Bear with me:

The arm64 interrupt entry is optimized the way a bunch of architectures are: it only saves off ABI caller-save registers immediately, because it's (probably) going to enter a C handler which can be relied on to preserve the remaining registers.  That way if you (as is likely) return into the same context that was interrupted you don't ned to do anything.

And that's great.  But look at the code in isr_wrapper.S: https://github.com/zephyrproject-rtos/zephyr/blob/main/arch/arm64/core/isr_wrapper.S#L105

What happens there is clear enough: the code calls z_get_next_switch_handle() (via a wrapper, but that's not important) to retrieve the next thread to run, and then calls into the same context switch code used by the SVC handler to finish the job.

But remember in SMP there are other CPUs racing against us.  So it's possible that z_arch_get_next_swtich_handle(), which is passed an "interrupted" context pointer, dutifully stores that into the interrupted thread struct, and that another CPU comes along instantaneously and sees it and uses it right away.

**But the context save isn't finished yet!  There's a race between here and a few dozen instructions later where the thread struct has incomplete data and will corrupt itself and crash if run.**

Basically: z_arch_get_next_switch_handle() needs to be passed a switch handle for a completely pickled thread.  z_arm64_context_switch needs to make that call itself somehow (which will require some surgery, as it's also used in cooperative/SVC contexts).

andyross, 2022-01-16:
Oh, and this explains why the proposed patch, which skipped the assignment of the interrupted thread pointer, "fixed" the reported bug:  it only goes non-NULL later on (inside the repurposed arch_switch() code) once things are correctly set up.

FWIW: I suspect, but didn't try, that you could hack your way around this without modifying the scheduler or the context switch code by passing a NULL "interrupted" argument to get_next_switch_handle() and relying on the z_arm64_context_switch code to do the assignment for you.  That would be ugly, but then it's not like context switch code is ever attractive.

carlocaione, 2022-01-16:
> But remember in SMP there are other CPUs racing against us. So it's possible that z_arch_get_next_swtich_handle(), which is passed an "interrupted" context pointer, dutifully stores that into the [...]

Yes, this is my understanding as well and this is what I tried to explain in the description of #41840 (sorry if I was not clear enough in the first place)

> Oh, and this explains why the proposed patch, which skipped the assignment of the interrupted thread pointer, "fixed" the reported bug: it only goes non-NULL later on (inside the repurposed arch_switch() code) once things are correctly set up.

Yes, and this is 100% the intention of that patch: postponing the non-NULL assignment later at a time when everything is correctly set-up instead of doing that too soon in `z_get_next_switch_handle()`.

> FWIW: I suspect, but didn't try, that you could hack your way around this without modifying the scheduler or the context switch code by passing a NULL "interrupted" argument to [...]

I think that would break when the scheduler decides to NOT switch (`old_thread == new_thread`) because that would keep the switch handle to NULL (since it is set to `!= NULL` by the context switch code that is not taken in that case).

I can probably work around this by passing a `NULL` interrupted argument, but setting the `switch_handle` to a `!= NULL` value when returning from `z_get_next_switch_handle()` also when the context switch is not taken.

@andyross I think I can fix this in an ARM64 specific way, but the question on my mind is then: why the PR is wrong? Isn't the fact that the `swtch_handle` is set too soon in the `z_get_next_switch_handle()` an actual fact? Are other architectures (`X86` mainly) working in a different way that it are able to fully save the context before the `switch_handle` is set in `z_get_next_switch_handle()`? I'm a bit confused about this TBH.

carlocaione, 2022-01-16:
Opened #41846 to address this only on the ARM64 code.

andyross, 2022-01-16:
>  Isn't the fact that the swtch_handle is set too soon in the z_get_next_switch_handle() an actual fact?

No, that's the architectural location where it **must** be set.  The meaning of that function is "I have finished an interrupt of this thread, here is its state, please arrange the scheduler state accordingly and tell me to which context to return".

The arch layer doesn't own the switch handle storage location.  That's a portable part of the kernel thread struct and the kernel owns it, which is why you get an opaque storage location as the second argument to arch_switch().  It happens that a lot of architectures (sort of unfortunately IMHO, but whatever) decided to make assumptions about it in order to port existing context switch code.   The original intention was that context would involve variable-sized blocks pushed to e.g. a stack, and that the switch handle would be a real data structure.  If you look at the xtensa architecture, that's how it works.

The bug is that arm64 is calling z_get_next_switch_handle() too soon, before the context is fully saved.  And the reason this works on other architectures is that they don't call the function until it's ready.

carlocaione, 2022-01-16:
> The bug is that arm64 is calling z_get_next_switch_handle() too soon, before the context is fully saved.

Ok, but still one tiny bit that I'm unclear about. If `z_get_next_switch_handle()` must be called after the context is fully saved, that means that we are saving the whole context also when we are not actually going to switch, so wasting a lot of cycles for each interrupt. Is this expected when using `USE_SWITCH`?

andyross, 2022-01-16:
On SMP where the race is present, yes, that's the requirement.  FWIW: total cycles in an interrupt aren't normally the thing you optimize for, the value of the incomplete save is that it reduces entry latency.

But longer term, it wouldn't be unreasonable to specify an API to "complete" the switch once the save was done (essentially just assigning to switch handle -- effectively we could bless this optimization as an official API, maybe with an extra function call just for aesthetics).

PR #41846 (2022-01-18): kernel: Reset the switch_handler only in the arch code
This PR follows the discussion at #40795 and is an ARM64-specific alternative to #41840

For details refer to that ticket and issue, but bottom line is that for ARM64 we do not set the `switch_handler` when exiting from `z_get_next_switch_handle()` but only in the context switching code when the context has been already fully saved to avoid races on multiple cores.

Fixes: #40795
closed 2022-01-18 by nashif