zephyrproject-rtos/zephyr #40812: Please mention the value of compile time constants (#defines) at the API docs.

FluffyBunniesTasteTheBest, 2021-12-01:
**Is your enhancement proposal related to a problem? Please describe.**
When reading a device tree, you'll often see magic numbers that represent flags. For example:
```
	button0: button_0 {
			gpios = < &gpio0 0x8 0x11 >;
			label = "Push button switch 0";
	};
```
**0x11** represent the flags. Figuring out what 0x11 actually means is getting a little challenging:
[The docs state](https://docs.zephyrproject.org/2.6.0/samples/basic/button/README.html#devicetree-details) _"FLAGS should be a logical OR of [GPIO configuration flags](https://docs.zephyrproject.org/2.6.0/reference/peripherals/gpio.html#gpio-api) meant to apply to the button, such as..."_
Problem is that the docs don't mention the flag-values, so a look into the sources is all that's left. 

**Describe the solution you'd like**
It would be useful if the API documentation would contain the values of compile time constants - in both, decimal and hexadecimal numbers.

**Describe alternatives you've considered**
grep your way through the sources...

gmarull, 2021-12-02:
Sounds like an issue, `0x11` should be replaced with proper flag definitions. Can you raise an issue with a list of affected boards?

gmarull, 2024-02-08:
Closing as these magic numbers do not appear in boards DT files, and GPIO flags are well documented.

closed 2024-02-08 by gmarull