zephyrproject-rtos/zephyr #40817: Create a HAL repository for Raspberry Pi Pico

yonsch, 2021-12-01:
## Origin

The HAL for the Raspberry Pi Pico is provided by the `raspberrypi/pico-sdk` repostitory: https://github.com/raspberrypi/pico-sdk .
Only the `src` directory is needed, and does not need any modification.

## Purpose

Second stage bootloader (XIP setup), register definitions and peripheral drivers.

## Mode of integration

The module should be integrated as an external HAL module.

The proposed name is `hal_raspberrypi_pico`. As discussed in the original PR, the `pico` differentiates between RPi's MCU and SBC lines (if Zephyr will support the SBCs in the future). `Pico` is the name of both the board and the SDK, so it's assumed that it's also the name of the series.

The proposed tree layout is as follows:
```
hal/raspberrypi_pico/
├── CMakeLists.txt
├── include
│   └── dt-bindings
├── pico-sdk
│   ├── CMakeLists.txt
│   ├── pico
│   └── src
├── README.md
└── zephyr
    └── module.yml
```
Where `src` is the `src` directory from `pico-sdk`.

## Pull Request

A demo of the desired repository can be found here: https://github.com/yonsch/hal_rp_demo .

A pull request currently using the demo repository (until the external module is ready) can be found here: https://github.com/zephyrproject-rtos/zephyr/pull/34835

## Description

`pico-sdk` is a C/C++ software development kit provided by Raspberry Pi for their line of MCUs (currently consisting of only RP2040). `pico-sdk` is stand alone, and provides the device headers, device drivers and build system for writing programs for the RP2040.

As Zephyr provides much of the functionality that is provided by `pico-sdk`, only the `src` directory, containing headers, drivers and boot code is needed.

`pico-sdk` generates a configuration file `config_autogen.h` that is included in most/every source file. This file can be used to configure the SDK, but also to work around some compatibility issues between `pico-sdk` and Zephyr.

# Dependencies

None

## Revision

1.3.0

## License

``BSD-3-Clause``
https://github.com/raspberrypi/pico-sdk/blob/master/LICENSE.TXT

gmarull, 2021-12-02:
Suggestion: since pico-sdk seems to be developed on Github, you can just create a fork of the repo, add a `zephyr/module.yml` with:
```yaml
name: pico-sdk
build:
  cmake-ext: True
  kconfig-ext: True
```

and then place all CMake/Kconfig/glue code in Zephyr. This method should make pico-sdk upgrades much easier (no manual tweaking). You can find an example of this approach in https://github.com/zephyrproject-rtos/zephyr/tree/main/modules/hal_gigadevice, https://github.com/zephyrproject-rtos/hal_gigadevice

carlescufi, 2021-12-02:
Thanks for this request. I will request this to be added in the agenda of the next TSC meeting.

yonsch, 2021-12-03:
@gmarull My additions to the original SDK are `pico-sdk/pico` (files that are expected to be auto generated) and the dts bindings in `include`. As long as these could be added to a forked version, I have no preference to one strategy or another.

carlescufi, 2021-12-03:
> @gmarull My additions to the original SDK are `pico-sdk/pico` (files that are expected to be auto generated) and the dts bindings in `include`. As long as these could be added to a forked version, I have no preference to one strategy or another.

Please add the dts bindings in the Zephyr tree, not in the module repo. Then please follow @gmarull's suggestion since I agree that it will simplify things in the future.

yonsch, 2022-01-17:
@carlescufi are there any news on this?

nashif, 2022-01-19:
@yonsch per TSC minutes, this was approved on Dec 8th, 2021. Was a repo created for the HAL?

yonsch, 2022-01-19:
@nashif I couldn't find one, so not yet

yonsch, 2022-01-19:
I also suggest calling the new repo `hal_rpi_pico` (not `hal_raspberrypi`), as discussed in the PR.

carlescufi, 2022-01-19:
Given that there might be additional members of the family not named pico, I would suggest keeping the HAL repo itself `hal_raspberrypi`, as per the vendor and not the series (like hal_nordic or hal_nxp).

carlescufi, 2022-01-19:
> @carlescufi are there any news on this?

Sorry, I was supposed to create this a while back and I did not. I will create it.

voxpelli, 2022-01-19:
> I also suggest calling the new repo `hal_rpi_pico` (not `hal_raspberrypi`), as discussed in the PR.

The chip itself is called RP2040, whereas the board that Raspberry Pi chips that uses it (+ the SDK) is called Pico. Is it the chip name or the board name that's essential?

carlescufi, 2022-01-19:
@yonsch I think that given that this is actually a GitHub repo we might as well fork it with GitHub and then rename it. You can then add the `zephyr/` folder at the root level.

@voxpelli the original repo is called pico-sdk, so that's the name we're going with.

carlescufi, 2022-01-19:
https://github.com/zephyrproject-rtos/hal_rpi_pico

closed 2022-01-19 by carlescufi