zephyrproject-rtos/zephyr #40827: Tensorflow example not working in zephyr v2.6

kclauah, 2021-12-02:
**Preface**
I have created a ticket on Nordic devzone and their engineer suggest me to create an issue here.
https://devzone.nordicsemi.com/f/nordic-q-a/82273/tensorflow-example-not-working-on-nrf-connect-sdk

**Describe the bug**
Building any tensorflow lite example using VS Code would show an error in prj.conf : 
`CONFIG_TENSORFLOW_LITE_MICRO was assigned the value y, but got the value n. Missing dependencies: 0`

**To Reproduce**
Steps to reproduce the behavior:
- Build any example under sample\modules\tensorflow 

**Expected behavior**
fatal error: tensorflow/lite/micro/all_ops_resolver.h: No such file or directory

**Impact**
Showstopper. 

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: nRF Connect SDK v1.7.0
 - Zephyr SDK v2.6 14f09a3b00

kclauah, 2021-12-06:
@laurenmurphyx64 I have followed your Youtube tutorial about integrating tflm into Zephyr. (https://www.youtube.com/watch?v=vMDxmEwu51M)

I have integrated the tensorflow module which is downloaded from https://github.com/zephyrproject-rtos/tensorflow and extracted to <root directory>/module/lib/tensorflow folder.
[image]
But "fatal error: tensorflow/lite/micro/all_ops_resolver.h: No such file or directory" error still shows up when I building the hello_world tf example.

And I think the reason that tflm files are missing is because the message from prj.conf. CONFIG_TENSORFLOW_LITE_MICRO was assigned the value y, but got the value n. Missing dependencies: 0

Would it be too much trouble for you to take a look at this?

laurenmurphyx64, 2021-12-07:
@kclauah, it wouldn't be any trouble! Thanks for filing an issue and tagging me. I read the advice from Simon in the devzone thread and took a look at the nRF Connect SDK's [west.yml](https://github.com/nrfconnect/sdk-nrf/blob/v1.8.0-rc1/west.yml). It looks like their west.yml pulls in the [Zephyr 2.6.0 west.yml](https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.6.0/west.yml#L154), but from [the comment at the top](https://github.com/nrfconnect/sdk-nrf/blob/v1.8.0-rc1/west.yml#L59) I suspect that their west.yml might be preventing west from importing tensorflow, even though you downloaded it.

Could you try the following for me and tell me what output you get?

1. Add tensorflow to the `name-allowlist` in `sdk-nrf/west.yml`:

```
name-allowlist:
          - tensorflow
          - TraceRecorderSource
          - canopennode
          - civetweb
```

2. Run `west update` to make sure TensorFlow is at the right commit and (try to) let west know it exists

3. Build again (`west build samples/modules/tensorflow/hello_world/ -b nrf9160dk_nrf9160_ns` or similar)

kclauah, 2021-12-07:
Thank you. Thank you so much for your assistance.

The error in prj.conf is gone and building the hello_world tf example works after following your suggestions.

I really appreciate your help in resolving the problem. 

> Could you try the following for me and tell me what output you get?
> 2. Run west update to make sure tensorflow is at the right commit and (try to) let west know it exists

Here is the log when running west update with verbose.
[west update log with tensorflow allowed.txt](https://github.com/zephyrproject-rtos/zephyr/files/7665525/west.update.log.with.tensorflow.allowed.txt)

> 3. Build again

Here is the log when building hello_world tf example.
[tf_hello_world build log.txt](https://github.com/zephyrproject-rtos/zephyr/files/7665555/tf_hello_world.build.log.txt)

Sorry for creating an issue here but this bug is actually related to nrf-sdk.

laurenmurphyx64, 2021-12-07:
No worries! I'm glad I could resolve your problem. 😄

closed 2021-12-07 by laurenmurphyx64

chandu07, 2022-07-14:
Hey there, I've had a similar issue and sorted it out now. For the future users, one modification to the solution proposed by @laurenmurphyx64 is to use `tflite-micro` rather than `tensorflow `in `name-allowlist`.

Cheers!

tripathidivya, 2022-11-16:
@laurenmurphyx64 Hello Lauren, as you mentioned in the above comment I have tried to follow the same steps but while updating west I have faced the issue like this for civetweb and tflite-micro
[image]

and in the end it generated the empty folders for civetweb and for tflite-micro.

While updating it is asking for the github credentials:
[image]

Also when I check on Zephyr-rtos repositry:
https://github.com/zephyrproject-rtos/zephyr/tree/main/modules

I have noticed there is no description for civetweb and for tflite-micro it has only only these two files inside:
[image]

I have tried to copy this tflite folder into the nrf-sdk folder but the issue is still same.

Can you please help me to resolve this issue?

Regards,
Divya

videvelop, 2023-10-09:
@tripathidivya  - did you mange to run zephyr/samples/modules/tflite-micro/hello_world ?

Dimfred, 2023-10-18:
For anyone looking, for me the issue was that one has to enable optional modules to be used.

```sh
west config manifest.group-filter -- +optional
west update
```

should do the trick.
The tflite example in the docs also lacks this, it is unclear that the module has to be enabled / how it can be enabled.