zephyrproject-rtos/zephyr #40856: PPP: gsm_modem: failure to establish LCP connection

marinbudimir, 2021-12-02:
**Describe the bug**

- I am trying sample [gsm_modem](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/gsm_modem). 
- I am using modem Quectel BG95.
- My modem is connected to NB-IoT network.
- I did not change CONFIG_MODEM_GSM_APN so it is using "internet" as default. I was trying to put APN that was returned by AT+COPS? command but AT+CGDCONT command fails if APN has spaces in it (APN returned by COPS has spaces).
- When I was using CONFIG_NET_PPP_LOG_LEVEL_DBG=y logs were huge and there was a lot of `<dbg> net_ppp.ppp_consume_ringbuf: Ringbuf 0x20001f54 is empty!` messages. 

This is my prj.conf
`
# UART support
CONFIG_SERIAL=y

# GSM modem support
CONFIG_MODEM=y
CONFIG_MODEM_GSM_PPP=y

# PPP networking support
CONFIG_NET_DRIVERS=y
CONFIG_NET_PPP=y
CONFIG_NET_L2_PPP=y
CONFIG_NET_NATIVE=y
CONFIG_NETWORKING=y

CONFIG_NET_L2_PPP_TIMEOUT=25000

# IPv4 enables PPP IPCP support
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n

# Network management events
CONFIG_NET_CONNECTION_MANAGER=y

# Log buffers, modem and PPP
CONFIG_LOG=y
CONFIG_NET_LOG=y
CONFIG_LOG_BUFFER_SIZE=16384
CONFIG_LOG_STRDUP_BUF_COUNT=200
CONFIG_MODEM_LOG_LEVEL_DBG=y
#CONFIG_NET_PPP_LOG_LEVEL_DBG=y
CONFIG_NET_L2_PPP_LOG_LEVEL_DBG=y
CONFIG_NET_MGMT_EVENT_LOG_LEVEL_DBG=y
CONFIG_NET_CONNECTION_MANAGER_LOG_LEVEL_DBG=y

CONFIG_NET_SHELL=y
CONFIG_MODEM_SHELL=y

CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y
`
This is logs and console output:
`
*** Booting Zephyr OS build zephyr-v2.6.0-5357-g81b1e7fdacf0  ***

[00:00:00.000,000] <dbg> modem_gsm.gsm_init: Generic GSM modem (0x200008b0)
[00:00:00.000,000] <dbg> modem_gsm.gsm_init: iface->read 0x80139cb iface->write 0x8013a01
[00:00:00.000,000] <dbg> modem_gsm.gsm_rx: starting
[00:00:00.000,000] <dbg> modem_gsm.gsm_configure: Starting modem 0x200008b0 configuration
[00:00:00.000,000] <dbg> net_mgmt.net_mgmt_event_init: (main): Net MGMT initialized: queue of 2 entries, stack size of 768
[00:00:00.000,000] <dbg> net_l2_ppp.net_ppp_init: (main): Initializing PPP L2 0x20001cf0 for iface 0x200003a0
[00:00:00.001,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:00.001,000] <dbg> modem_gsm.gsm_cmd_ok: ok
[00:00:00.006,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [+COPS:] (len:21)
[00:00:00.006,000] <dbg> modem_cmd_handler.cmd_handler_process_rx_buf: match cmd [OK] (len:2)
[00:00:00.006,000] <dbg> modem_gsm.gsm_cmd_ok: ok
[00:00:00.007,000] <dbg> net_mgmt.net_mgmt_add_event_callback: (conn_mgr): Adding event callback 0x20002b38
[00:00:00.007,000] <dbg> net_mgmt.net_mgmt_add_event_callback: (conn_mgr): Adding event callback 0x20002b4c
[00:00:00.007,000] <dbg> conn_mgr.conn_mgr_handler: (conn_mgr): Connection Manager started
[00:00:00.007,000] <dbg> net_l2_ppp.tx_handler: (tx_handler_thread): PPP TX started
[00:00:00.007,000] <inf> sample_gsm_ppp: Board 'stm32f4_disco' APN 'internet' UART 'UART_2' device 0x8014d94 (gsm_ppp)
[... 223 lines cut ...]
[00:03:22.496,000] <dbg> net_l2_ppp: 0x2000d014
                                     c0 21 01 01 00 04                                |.!....           
[00:03:47.496,000] <dbg> net_l2_ppp.ppp_fsm_timeout: (sysworkq): [LCP/0x20001d38] Current state REQUEST_SENT (6)
[00:03:47.496,000] <dbg> net_l2_ppp.fsm_send_configure_req: (sysworkq): [LCP/0x20001d38] Sending Configure-Req (1) id 1 to peer while in REQUEST_SENT (6)
[00:03:47.496,000] <dbg> net_l2_ppp.ppp_send_pkt: (sysworkq): [LCP/0x20001d38] Sending 6 bytes pkt 0x2000d014 (options len 0)
[00:03:47.496,000] <dbg> net_l2_ppp.net_pkt_hexdump: send L2
[00:03:47.496,000] <dbg> net_l2_ppp: 0x2000d014
                                     c0 21 01 01 00 04                                |.!....  
[00:04:12.496,000] <dbg> net_l2_ppp.ppp_fsm_timeout: (sysworkq): [LCP/0x20001d38] Current state REQUEST_SENT (6)
[00:04:12.496,000] <dbg> net_l2_ppp.ppp_fsm_timeout: (sysworkq): [LCP/0x20001d38] Configure-Req retransmit limit 0 reached
[00:04:12.496,000] <dbg> net_l2_ppp.ppp_change_state_debug: (sysworkq): [LCP/0x20001d38] state REQUEST_SENT (6) => STOPPED (3) (ppp_fsm_timeout():113)
[00:04:12.496,000] <dbg> net_l2_ppp.ppp_change_phase_debug: (sysworkq): [0x20001cf0] phase ESTABLISH (1) => DEAD (0) (ppp_link_terminated():123)
[00:04:12.496,000] <dbg> net_l2_ppp.ppp_link_terminated: (sysworkq): [0x20001cf0] Link terminated

`

Environment:
zephyr sdk 0.13.2
VERSION
`VERSION_MAJOR = 2
VERSION_MINOR = 7
PATCHLEVEL = 99
VERSION_TWEAK = 0
EXTRAVERSION =
`

austinphillips2, 2022-02-06:
I have previously observed a similar issue when connecting PPP from a Linux box.  The issue was addressed by setting `CONFIG_NET_CONFIG_AUTO_INIT=n`

With `CONFIG_NET_CONFIG_AUTO_INIT=y`, net autoconfiguration autostarts on boot prior to starting static threads.  However, because ppp requires static threads to run over uart, enabling this setting prevents all uart transmissions and the ppp connection fails.  If the net autoconfig isn't started using `CONFIG_NET_CONFIG_AUTO_INIT=y`, the application must call `net_config_init_app()` from main during startup.

From memory, I also had to set `CONFIG_NET_L2_PPP_OPTION_MRU=y` in my specific case to prevent ppp stack from dropping packets due to a MRU setting of 0.

mmurty, 2022-03-11:
I am also facing a similar issue where I get too many `net_ppp.ppp_consume_ringbuf: Ringbuf 0x2020xxxx is empty` messages. I have tried increasing the `CONFIG_NET_PPP_UART_BUF_LEN=1500` and `CONFIG_NET_PPP_RINGBUF_SIZE=1500` values but still I do not see any improvements. 
Please suggest if anyone has any inputs.

marinbudimir, 2022-03-12:
> I am also facing a similar issue where I get too many `net_ppp.ppp_consume_ringbuf: Ringbuf 0x2020xxxx is empty` messages. I have tried increasing the `CONFIG_NET_PPP_UART_BUF_LEN=1500` and [...]

Try setting CONFIG_NET_L2_PPP_OPTION_MRU=y.

mmurty, 2022-03-14:
I tried this but it doesn't help.

closed 2022-05-29 by github-actions[bot]