zephyrproject-rtos/zephyr #40866: Undefined behavior in lib/os/cbprintf_packaged.c: subtraction involving definitely null pointer

RobertoBagnara, 2021-12-02:
Subtracting buf0, which is null in line 507 of lib/os/cbprintf_packaged.c, is undefined behavior:
	/*
	 * If all we wanted was to count required buffer size
	 * then we have it now.
	 */
	if (!buf0) {
		return len + buf - buf0;  // Undefined behavior in pointer subtraction.
	}

npitre, 2021-12-03:
Fix in #40899.

cfriedt, 2021-12-07:
@RobertoBagnara - was this a flag raised by a particular tool? If so, could you also provide some additional detail here?

RobertoBagnara, 2021-12-08:
On the topic of undefined behavior, I recommend https://blog.regehr.org/archives/213

Concerning the functional safety and security of software written in C, for static analysis and code reviews/inspections there really are only two possibilities:

1) Verification at the assembly code level (however the assembly code is produced).
2) Verification of the source code and use of a qualified translation toolchain.

Any argument of the form "I don't care about undefined or critical unspecified behavior because I believe/know/suspect my compiler does the right thing with the compilation options I believe/know/suspect I use" belongs to possibility (1).

Any instance of undefined or critical unspecified behavior makes the program semantics undefined, and thus makes (2) impossible.

npitre, 2021-12-08:
> On the topic of undefined behavior, I recommend
> https://blog.regehr.org/archives/213

In theory, practice and theory are the same, but in practice they are not.

I'm repeating myself: please give me an example of a real compiler
implementation where this particular undefined behavior does matter.

Because not only this code does subtract NULL in this particular
instance, but the whole function does add to an initial NULL pointer
in some cases.

This is done that way because Zephyr is meant for small embedded systems
where code size matters, and this particular function is already too big
for its own sake.

So, unless there is proof that this would cause actual problem i.e. not
just a theoretical "undefined by the standard" thing that happens to be
perfectly fine in practice then , I'd suggest that this PR and issue
simply be discarded.

> Verification at the assembly code level (however the assembly code
> is produced).

Sure. Let's have a look:

```
int ptr_diff(char *p, char *q) { return p - q; }
```

X86-64 result (both gcc and clang):

```
        movq    %rdi, %rax
        subq    %rsi, %rax
        ret
```

ARM32 result (both gcc and clang):

```
        sub     r0, r0, r1
        bx      lr
```

This will always work, even if one or both pointers are NULL.

Now for an example that reflects the case in question here:

```
#include <stddef.h>
int test(void *ptr)
{
        char *buf = ptr, *buf0 = buf;

        buf += 42;
        if (buf0 == NULL) {
                return buf - buf0;
        }
        return -1;
}
```

On ARM we get:

```
        cmp     r0, #0
        mvnne   r0, #0
        moveq   r0, #42
        bx      lr
```

The compiler did figure out that if ptr is NULL then no subtraction is
needed.

On X86 this is a bit more obscur but still correct:

```
        cmpq    $1, %rdi
        sbbl    %eax, %eax
        andl    $43, %eax
        subl    $1, %eax
        ret
```

In this case, not only did the compiler pull a nice optimization, but
it did so while fully accounting for the "undefined" behavior we're
counting on.

And finally, this function and the undefined behavior it relies on is
thoroughly exercized by the test suite already. So this won't catch
us off guard if ever some alien compiler is used on this code.

closed 2021-12-08 by npitre