zephyrproject-rtos/zephyr #40870: tests: syscall: failed to build on fvp_baser_aemv8r_smp

LixinGuoX, 2021-12-03:
**Describe the bug**
Failed to build tests/kernel/mem_protect/syscalls on fvp_baser_aemv8r_smp.
It seems like that the `CONFIG_MAX_THREAD_BYTES` by default is too small,
maybe we can define it in test case's `prj.conf` file manually.

**To Reproduce**
Steps to reproduce the behavior:
west build -b fvp_baser_aemv8r_smp tests/kernel/mem_protect/syscalls/

**Expected behavior**
CI should be running smoothly.

**Impact**
Showstopper

**Logs and console output**
```
Including boilerplate (Zephyr base): /__w/zephyr/zephyr/cmake/app/boilerplate.cmake
-- Application: /__w/zephyr/zephyr/tests/kernel/mem_protect/syscalls
-- Zephyr version: 2.7.99 (/__w/zephyr/zephyr), build: v2.7.99-2049-gb15c8fda1f63
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: fvp_baser_aemv8r_smp
-- Cache files will be written to: /github/home/.cache/zephyr
-- Using toolchain: zephyr 0.13.2 (/opt/toolchains/zephyr-sdk-0.13.2)
-- Found dtc: /opt/toolchains/zephyr-sdk-0.13.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /__w/zephyr/zephyr/boards/arm64/fvp_baser_aemv8r/fvp_baser_aemv8r_smp.dts
-- Generated zephyr.dts: /__w/zephyr/zephyr/twister-out/fvp_baser_aemv8r_smp/tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /__w/zephyr/zephyr/twister-out/fvp_baser_aemv8r_smp/tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /__w/zephyr/zephyr/twister-out/fvp_baser_aemv8r_smp/tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /__w/zephyr/zephyr/twister-out/fvp_baser_aemv8r_smp/tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls/zephyr/dts.cmake
Parsing /__w/zephyr/zephyr/Kconfig
Loaded configuration '/__w/zephyr/zephyr/boards/arm64/fvp_baser_aemv8r/fvp_baser_aemv8r_smp_defconfig'
Merged configuration '/__w/zephyr/zephyr/tests/kernel/mem_protect/syscalls/prj.conf'
Configuration saved to '/__w/zephyr/zephyr/twister-out/fvp_baser_aemv8r_smp/tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls/zephyr/.config'
Kconfig header saved to '/__w/zephyr/zephyr/twister-out/fvp_baser_aemv8r_smp/tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/toolchains/zephyr-sdk-0.13.2/aarch64-zephyr-elf/bin/aarch64-zephyr-elf-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /__w/zephyr/zephyr/twister-out/fvp_baser_aemv8r_smp/tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls
[1/154] Preparing syscall dependency handling
[2/154] Generating misc/generated/syscalls_subdirs.trigger
[3/154] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
...
[132/154] Linking C executable zephyr/zephyr_pre0.elf
[133/154] Generating app_smem_aligned linker section
[134/154] Generating kobject_prebuilt_hash.gperf
FAILED: zephyr/kobject_prebuilt_hash.gperf 
cd /__w/zephyr/zephyr/twister-out/fvp_baser_aemv8r_smp/tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls/zephyr && /usr/bin/python3.8 /__w/zephyr/zephyr/scripts/gen_kobject_list.py --kernel /__w/zephyr/zephyr/twister-out/fvp_baser_aemv8r_smp/tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls/zephyr/zephyr_pre0.elf --gperf-output kobject_prebuilt_hash.gperf --include /__w/zephyr/zephyr/twister-out/fvp_baser_aemv8r_smp/tests/kernel/mem_protect/syscalls/kernel.memory_protection.syscalls/zephyr/misc/generated/struct_tags.json
Too many thread objects (22)
Increase CONFIG_MAX_THREAD_BYTES to 3
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain zephyr-sdk-0.13.2

cfriedt, 2021-12-07:
@LixinGuoX - do your PRs fix the issue? Thanks for reporting the issue.
cc @dcpleung

LixinGuoX, 2021-12-08:
> @LixinGuoX - do your PRs fix the issue? Thanks for reporting the issue. cc @dcpleung

yes, #40871 will fix this issue.

PR #40871 (2021-12-10): tests: syscall: fix low value of CONFIG_MAX_THREAD_BYTES
Failed to build tests/kernel/mem_protect/syscalls on fvp_baser_aemv8r_smp
because the default value of CONFIG_MAX_THREAD_BYTES is too low.
So manually set the value to 5 in test case's prj.conf file.

Fixes #40870
closed 2021-12-10 by cfriedt