zephyrproject-rtos/zephyr #40873: qemu_cortex_r5: fail to handle user_string_alloc_copy() with null parameter

LixinGuoX, 2021-12-03:
**Describe the bug**
When I push this PR #40830 to validate string_alloc_copy() API with null parameter,
CI got blocked.
It works fine on other platforms, and it seems like the implementation of arch_user_string_nlen()
has some problem with processing null parameter.

**To Reproduce**
see #40830

**Expected behavior**
CI should be running smoothly.

**Impact**
Showstopper

**Logs and console output**
```
*** Booting Zephyr OS build v2.7.99-2075-g75ac3364c2e9  ***
Running test suite syscalls
===================================================================
START - test_string_nlen
 PASS - test_string_nlen in 0.1 seconds
===================================================================
START - test_string_nlen
E: syscall z_vrfy_string_nlen failed check: Memory region 0x20880 (size 24) read access denied
 PASS - test_string_nlen in 0.4 seconds
===================================================================
START - test_to_copy
E: syscall user_copy failed check: Memory region 0x20860 (size 32) write access denied
 PASS - test_to_copy in 0.4 seconds
===================================================================
START - test_user_string_copy
E: string too long 0x109e5 (32)
E: syscall user_copy failed check: Memory region 0x20880 (size 24) read access denied
 PASS - test_user_string_copy in 0.5 seconds
===================================================================
START - test_user_string_alloc_copy
E: string too long 0x109e5 (32)
E: syscall z_user_alloc_from_copy failed check: Memory region 0x20880 (size 24) read access denied

    Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/mem_protect/syscalls/src/main.c:259: test_user_string_alloc_copy: (ret not equal to -1)
got -2
 FAIL - test_user_string_alloc_copy in 0.9 seconds
===================================================================
START - test_arg64
 PASS - test_arg64 in 0.2 seconds
===================================================================
START - test_more_args
 PASS - test_more_args in 0.2 seconds
===================================================================
START - test_syscall_torture
Running syscall torture test with 4 threads on 1 cpu(s)

 PASS - test_syscall_torture in 15.126 seconds
===================================================================
START - test_syscall_context
 PASS - test_syscall_context in 0.2 seconds
===================================================================
Test suite syscalls failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain zephyr-sdk-0.13.2
 - Commit ce99227ad4fbfd08e5d2df4200684f818bc426bf

cfriedt, 2021-12-07:
@dcpleung - please raise priority if necessary

LixinGuoX, 2021-12-31:
In zephyr/arch/arm/core/aarch32/userspace.S/arch_user_string_nlen(const char *s, size_t maxsize, int *err_arg), dealing with the address of NULL will not trigger a fault, will be treated as an empty string in my opinion. So this may not be a bug, and close this issue.

closed 2021-12-31 by LixinGuoX