zephyrproject-rtos/zephyr #40874: mps2_an521_ns: fail to handle user_string_alloc_copy() with null parameter

LixinGuoX, 2021-12-03:
**Describe the bug**
When I push this PR #40830 to validate string_alloc_copy() API with null parameter,
CI got blocked.
It works fine on other platforms.

**To Reproduce**
see #40830

**Expected behavior**
CI should be running smoothly.

**Impact**
Showstopper

**Logs and console output**
```
[1;34m[Sec Thread] Secure image initializing![0m
[1;34mBooting TFM v1.4.1[0m
[1;34m[Crypto] Dummy Entropy NV Seed is not suitable for production![0m
*** Booting Zephyr OS build v2.7.99-2075-g75ac3364c2e9  ***
Running test suite syscalls
===================================================================
START - test_string_nlen
 PASS - test_string_nlen in 0.1 seconds
===================================================================
START - test_string_nlen
E: syscall z_vrfy_string_nlen failed check: Memory region 0x281008f0 (size 24) read access denied
 PASS - test_string_nlen in 0.2 seconds
===================================================================
START - test_to_copy
E: syscall user_copy failed check: Memory region 0x281008d0 (size 32) write access denied
 PASS - test_to_copy in 0.2 seconds
===================================================================
START - test_user_string_copy
E: string too long 0x10e1e7 (32)
E: syscall user_copy failed check: Memory region 0x281008f0 (size 24) read access denied
 PASS - test_user_string_copy in 0.3 seconds
===================================================================
START - test_user_string_alloc_copy
E: string too long 0x10e1e7 (32)
E: syscall z_user_alloc_from_copy failed check: Memory region 0x281008f0 (size 24) read access denied
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain zephyr-sdk-0.13.2
 - Commit ce99227ad4fbfd08e5d2df4200684f818bc426bf

theotherjimmy, 2022-01-24:
Reproduced while connected with GDB. I'll post what I find.

theotherjimmy, 2022-01-24:
It looks to me like this test case is generating a secure falt, which is handled by tf-m, instead of the normal zephyr fault handling on the invalid address `0xffff_ffff`. Since the tf-m secure fault handler just spins, we get stuck and timeout the test.

closed 2022-04-20 by github-actions[bot]