zephyrproject-rtos/zephyr #40881: Bluetooth: shell: fatal error because ctx_shell is NULL

anrudolf, 2021-12-03:
# Description
With `CONFIG_BT_SHELL=y`, when an external device connects to Zephyr via Bluetooth, there is a fatal error. 

The first "bad" commit is [8e1682d1](https://github.com/zephyrproject-rtos/zephyr/commit/8e1682d1ea5865fd1a38f22546d2eeb7cb026f53).

The underlying problem is that `shell_print(ctx_shell, ...)` is invoked in the Bluetooth shell module in multiple callbacks (`.connected`, `.disconnected`, `.le_param_req`, etc.) and `ctx_shell` is undefined at this stage. See [here](https://github.com/zephyrproject-rtos/zephyr/blob/3f826560aaf81a444018293bd6acce3c339fe150/subsys/bluetooth/shell/bt.c#L252).

It used to be that these Bluetooth callbacks were registered dynamically with `bt_conn_cb_register(&conn_callbacks)` through the `bt init` shell command, which also sets `ctx_shell` to the current shell beforehand. See [here](https://github.com/zephyrproject-rtos/zephyr/blob/79a6c07536bc14583198f8e3555df6134d8822cf/subsys/bluetooth/shell/bt.c#L580) and [here](https://github.com/zephyrproject-rtos/zephyr/blob/79a6c07536bc14583198f8e3555df6134d8822cf/subsys/bluetooth/shell/bt.c#L593).

Now, however, the callbacks are registered statically with `BT_CONN_CB_DEFINE(...)`, and are called when a Bluetooth connection is made for instance. At this stage `ctx_shell` is still undefined, and `shell_print(ctx_shell, ...)` is invoked, which results in the mentioned crash.

**Note**: If shell command `bt init` is invoked before the connection is made, then `ctx_shell` is set to the current shell, and there is no error.

# To Reproduce

We are using `nrf52840dk_nrf52840` target board for testing.

## Steps to reproduce the behavior with mcumgr SMP server sample:
```
# build and flash mcuboot bootloader
$ west build --pristine -d build-mcuboot --board nrf52840dk_nrf52840 bootloader/mcuboot/boot/zephyr/ && west flash -d build-mcuboot

# build SMP server sample with CONFIG_BT_SHELL=y
$ west build --pristine --board nrf52840dk_nrf52840 zephyr/samples/subsys/mgmt/mcumgr/smp_svr/ -- -DOVERLAY_CONFIG=overlay-bt.conf -DCONFIG_BT_SHELL=y

# sign and flash SMP server sample
$ west sign -t imgtool -- --key bootloader/mcuboot/root-rsa-2048.pem && west flash --hex-file build/zephyr/zephyr.signed.hex

# open console
$ picocom -b 115200 /dev/ttyACM0

# On host: try to list images on target board through mcumgr
$ mcumgr --conntype ble --connstring 'peer_name=Zephyr' image list

# observe fatal error in console
[00:00:07.122,711] <inf> smp_bt_sample: Connected
[00:00:07.122,863] <err> os: ***** USAGE FAULT *****
[00:00:07.122,863] <err> os:   Unaligned memory access
[00:00:07.122,863] <err> os: r0/a1:  0x00005bf1  r1/a2:  0x00020000  r2/a3:  0x0003d481
[00:00:07.122,894] <err> os: r3/a4:  0x00005bf1 r12/ip:  0x0000000a r14/lr:  0x00030a93
[00:00:07.122,894] <err> os:  xpsr:  0x21000000
[00:00:07.122,894] <err> os: Faulting instruction address (r15/pc): 0x0003087c
[00:00:07.122,894] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:07.122,924] <err> os: Current thread: 0x20001700 (BT RX)
[00:00:07.495,422] <err> os: Halting system
```

## Steps to reproduce the behavior with Bluetooth peripheral sample:
```
# build and flash Bluetooth peripheral sample with CONFIG_BT_SHELL=y
west build --pristine --board nrf52840dk_nrf52840 zephyr/samples/bluetooth/peripheral -- -DCONFIG_BT_SHELL=y && west flash

# open console
$ picocom -b 115200 /dev/ttyACM0

# connect to the target board, for instance with the Nordic nRF Connect for Mobile app (the name will be "Zephyr Peripheral Sample Long Name")

# observer fatal error in console
Connected
[00:00:56.973,052] <err> os: ***** USAGE FAULT *****
[00:00:56.973,052] <err> os:   Unaligned memory access
[00:00:56.973,083] <err> os: r0/a1:  0x00028ea9  r1/a2:  0x00020000  r2/a3:  0x00033483
[00:00:56.973,083] <err> os: r3/a4:  0x00028ea9 r12/ip:  0x0000000a r14/lr:  0x000262ef
[00:00:56.973,083] <err> os:  xpsr:  0x21000000
[00:00:56.973,083] <err> os: Faulting instruction address (r15/pc): 0x000260d8
[00:00:56.973,083] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:56.973,114] <err> os: Current thread: 0x20001748 (BT RX)
[00:00:57.259,399] <err> os: Halting system
```

# Expected behavior
No fatal error when connecting to the target board :-)

# Environment
 - OS: Linux
 - Toolchain: Zephyr SDK 0.13.2
 - Zephyr v2.7.0

# Additional context
We would like to have an mcumgr SMP server running to allow firmware upgrades over Bluetooth. At the same time, we would also like to have shell commands such as `bt name` from `CONFIG_BT_SHELL=y` available to be used for automated testing et al.

# Possible workarounds
1. Revert back to dynamically registering the Bluetooth callbacks in the `cmd_init & bt_ready` functions.
2. Check if `ctx_shell` is defined before invoking `shell_print(ctx_shell, ...)` from a Bluetooth callback.

All in all, I am not sure if writing messages from Bluetooth callbacks to a shell is the right way to do it. For shell commands such as `bt name`, writing the name back to the shell is of course great. But for Bluetooth callbacks such as `.connected` and similar which could be invoked at any time, may be it would be better to use `LOG_XXX` instead?

I am happy to create a PR, but I am not sure which of the above mentioned fixes is preferred. If a project member could give me any pointers that would be great.

jhedberg, 2021-12-07:
I'm a bit confused: how can a device connect to us if `bt init` hasn't been issued? You need to initialize the stack before requesting it to be connectable (perform connectable advertising), so it seems like `bt init` is a pre-requisite for any connections and related callbacks to happen.

anrudolf, 2021-12-07:
Hi @jhedberg and thanks for replying

Yes, if Bluetooth shell is the only entity interacting with the Bluetooth stack, then one needs to issue `bt init`, otherwise the device is not connectable, and no Bluetooth callbacks could ever happen.

However, as in the example above - for instance with `samples/subsys/mgmt/mcumgr/smp_svr/` or `samples/bluetooth/peripheral` **and** `CONFIG_BT_SHELL=y` - these applications also call `bt_enable` at some point during initialization (see [here](https://github.com/zephyrproject-rtos/zephyr/blob/3f826560aaf81a444018293bd6acce3c339fe150/samples/subsys/mgmt/mcumgr/smp_svr/src/bluetooth.c#L78) and [here](https://github.com/zephyrproject-rtos/zephyr/blob/3f826560aaf81a444018293bd6acce3c339fe150/samples/bluetooth/peripheral/src/main.c#L333)). Thus, the device is connectable, and when a connection is made, the Bluetooth shell callbacks are also invoked. These Bluetooth shell callbacks then try to `shell_print(ctx_shell, ...)`, but `ctx_shell` may very well be undefined at this stage. As mentioned, if one issues `bt init` before any connection is made, then ctx_shell is set (see [here](https://github.com/zephyrproject-rtos/zephyr/blob/3f826560aaf81a444018293bd6acce3c339fe150/subsys/bluetooth/shell/bt.c#L588)), and no fatal errors occur. (`bt init` will fail however with `-EALREADY`).

Maybe a little more background. We would like to run a Bluetooth application (including the ability to do Bluetooth firmware upgrades) - so the Bluetooth stack will be initialized by the application - and at the same time we would also like to enable the Bluetooth shell (`CONFIG_BT_SHELL=y`) in order to use shell commands such as `bt name` and similar. I assume this is a common use case for Bluetooth shell, however, please let me know if this use is not intended. In the current state of subsys Bluetooth shell, as soon as a Bluetooth connection is made - without explicitly issuing `bt init` beforehands - there is a fatal error in the Bluetooth shell `connected` callback because `ctx_shell` is still undefined. It used to be that Bluetooth shell registered its callbacks dynamically only after `bt init` has been issued. Now the callbacks are registered statically, and may be invoked with an undefined `ctx_shell`.

LingaoM, 2021-12-08:
Thank you very much for your careful analysis of this problem,
There are several solutions:

- Change to dynamic registration
- Add a null pointer to judge

@anrudolf Can you submit PR for this?

anrudolf, 2021-12-08:
Hi @LingaoM 

Pull request [41007](https://github.com/zephyrproject-rtos/zephyr/pull/41007) is open :-)

commit 92fbd7da (2021-12-09): Bluetooth: shell: Register callbacks dynamically

Bluetooth callbacks are now registered dynamically again. When Bluetooth
callbacks were registered statically, they could be invoked before the
Bluetooth shell (ctx_shell) is initialized. The callbacks try to
shell_print(ctx_shell, ...), which results in a fatal error if ctx_shell
is undefined.

Fixes #40881.