zephyrproject-rtos/zephyr #40883: Limitation on logging module

Isaia-F, 2021-12-03:
Compilation fails after a certain number (16 in my case though the size of the individual argument might matter) of arguments 

Using SEGGER Embedded Studio - Nordic Edition and nRF Connect SDK

**To Reproduce**
Steps to reproduce the behavior:
1. In SES - Nordic Edition, open the nRF Connect SDK Template sample project for Zigbee
2. Edit `zcl_general_commands.c`  to include the following:
#include <logging/log.h>
LOG_MODULE_DECLARE(app)
3. Still in `zcl_general_commands.c`, edit `zb_zcl_configure_reporting_handler()` to include the following LOG_INF()

> LOG_INF("Status of variables at this point"
>         "param=%d \n"
>         "cmd_info\n"
>         "|----addr_data.common_data \n"
>         "|    |----dst_addr=0x%04x \n"
>         "|    |----dst_endpoint=0x%02x \n"
>         "|    |----fc=0x%02x \n"
>         "|    |----source \n"
>         "|    |    |----addr_type=0x%02x \n"
>         "|    |    |----u \n"
>         "|    |         |----ieee_addr=0x%02x%02x%02x%02x%02x%02x%02x%02x \n"
>         "|    |         |----short_addr=0x%04x \n"
>         "|    |         |----src_id=0x%08x \n"
>         "|    |\n"
>         ,param,
>         cmd_info->addr_data.common_data.dst_addr, cmd_info->addr_data.common_data.dst_endpoint,
>         cmd_info->addr_data.common_data.fc, cmd_info->addr_data.common_data.source.addr_type,
>         cmd_info->addr_data.common_data.source.u.ieee_addr[0], cmd_info->addr_data.common_data.source.u.ieee_addr[1],
>         cmd_info->addr_data.common_data.source.u.ieee_addr[2], cmd_info->addr_data.common_data.source.u.ieee_addr[3],
>         cmd_info->addr_data.common_data.source.u.ieee_addr[4], cmd_info->addr_data.common_data.source.u.ieee_addr[5],
>         cmd_info->addr_data.common_data.source.u.ieee_addr[6], cmd_info->addr_data.common_data.source.u.ieee_addr[7],
>         cmd_info->addr_data.common_data.source.u.short_addr,cmd_info->addr_data.common_data.source.u.src_id
> );

3. Build project, note the absence of compilation errors.
4. Add `"|    |----src_endpoint=0x%02x \n"` and corresponding `, cmd_info->addr_data.common_data.src_endpoint` 
5. Build project, SEGGER will output the following error

> Compiling 'zcl_general_commands.c'
>     util_macro.h
>     util.h
>     log_msg.h
>     log_core.h
>     log.h
>     zcl_general_commands.c
>     In function 'zb_zcl_configure_reporting_handler':
>     '_LOG_INTERNAL_cmd_info' undeclared (first use in this function)
>     in definition of macro 'UTIL_PRIMITIVE_CAT'
>     in expansion of macro 'UTIL_CAT'
>     in expansion of macro 'Z_LOG_INTERNAL_X'
>     in expansion of macro 'Z_LOG_INTERNAL2'
>     in expansion of macro 'Z_LOG_INTERNAL'
>     in expansion of macro 'Z_LOG2'
>     in expansion of macro 'Z_LOG'
>     in expansion of macro 'LOG_INF'
>     each undeclared identifier is reported only once for each function it appears in
>     in definition of macro 'UTIL_PRIMITIVE_CAT'
>     in expansion of macro 'UTIL_CAT'
>     in expansion of macro 'Z_LOG_INTERNAL_X'
>     in expansion of macro 'Z_LOG_INTERNAL2'
>     in expansion of macro 'Z_LOG_INTERNAL'
>     in expansion of macro 'Z_LOG2'
>     in expansion of macro 'Z_LOG'
>     in expansion of macro 'LOG_INF'
> Build failed `

**Expected behavior**
No compilation issues

**Impact**
Annoyance

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: same as nRF Connect SDK

carlescufi, 2021-12-03:
@Isaia-F thanks for reporting. This is the Zephyr Project and not Nordic's support site, so could you please first reproduce this with upstream, vanilla Zephyr and describe the steps here?

nordic-krch, 2021-12-03:
@Isaia-F this is the hard limitation in log v1 (up to 15 arguments) but logging v2 does not have this limitation. However, since macro is using compiler switch both version are compiled and using more than 15 arguments ends up in compilation failure.

I have fixed that in #40886 where i used preprocessor switch instead. If you want to use more than 15 arguments you need to swtich to version 2. This is done by setting `CONFIG_LOG2_MODE_DEFERRED=y` or `CONFIG_LOG2_MODE_IMMEDIATE=y`.

Isaia-F, 2021-12-03:
@nordic-krch i have attempted to set CONFIG_LOG2_MODE_DEFERRED via guiconfig but the problem appears to persist

In the .config file i see 
\# CONFIG_JWT is not set
CONFIG_LOG=y
\# CONFIG_LOG_MODE_DEFERRED is not set
CONFIG_LOG2_MODE_DEFERRED=y
\# CONFIG_LOG2_MODE_IMMEDIATE is not set
\# CONFIG_LOG_MODE_IMMEDIATE is not set
\# CONFIG_LOG_MODE_MINIMAL is not set
\# CONFIG_LOG_FRONTEND is not set
CONFIG_LOG2=y
CONFIG_LOG_MINIMAL=y

but when i build the solution i still get the same error

PR #40886 (2021-12-04): logging: Extend support for compilation with up to 31 arguments
Logging v2 supports 255 arguments but since logging v1 was supporting
only 15, and it is still compiled, more than 15 could not be used.
Extending logging v1 macro to accept up to 31 arguments. It will fail
during processing in case of v1 but will work with v2 enabled. It is
a temporary solution since v1 will be removed at some point.

Fixes #40883.
closed 2021-12-04 by carlescufi