zephyrproject-rtos/zephyr #40895: mgmt/mcumgr/lib: Continue partial application upload

de-nordic, 2021-12-03:
**Is your feature request related to a problem? Please describe.**
An application image upload context is only kept in RAM, so in case if reset occurs during upload the upload needs to restart with all the costs of erasing flash and so on. 

**Describe the solution you'd like**
An upload may exchange with an application control sums of subsequent flash pages to decide where to re-start upload process, saving on upload time and reducing need for erasing flash when part of an image has already been successfully written to flash.

**Describe alternatives you've considered**
Re-uploading whole image as it happens now.

**Additional context**
Control sums should be calculated per page size as visible on the device side, as pages may have variable size and that page size dictates how much data would have to be erased and re-uploaded.

Laczen, 2021-12-04:
@de-nordic, it would be good if such a feature is added. The stream flash api has support for something like that. Wouldn't this feature require a complete mcumgr rework where the device request for blocks ?

Laczen, 2021-12-04:
@de-nordic, BTW you are an example of what maintainers should be doing.

de-nordic, 2021-12-06:
> @de-nordic, it would be good if such a feature is added. The stream flash api has support for something like that. Wouldn't this feature require a complete mcumgr rework where the device requests for blocks ?

Currently a mcumgr enabled device has the maximum size of the "chunk" of upload data hard-coded  at compile time (https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/mcumgr/Kconfig#L158. but rather than using that number, the client side should send sum with offset and size it used to calculate it on, and the device should use that data. Reason for above is that your client side application may be keeping track on chunks that have already been sent to a device so it may choose its own strategy how to handle checks; for example it may as to check sum of everything from offset 0 to last chunk or starting at chunk -2 from the last that has been confirmed.

The image upload protocol allows device to ask for previous chunk by sending lower offset in response, so it is basically able to ask to restart transfer at some specific offset/

As you know the erase needs to happen at page boundaries, so in case when sum fails anywhere within page boundary, the request offset would have to point at the page offset.

> BTW you are an example of what maintainers should be doing.
Thanks