zephyrproject-rtos/zephyr #40897: west build/west sign should issue warning that built image may not be swappable by mcuboot

de-nordic, 2021-12-03:
**Is your enhancement proposal related to a problem? Please describe.**
The mcuboot needs some bytes at the end of image slot to store progress of swap and some flags, which is not reserved during build; this means that when image grows beyond certain point the mcuboot will no longer be able to swap images, making DFU updates not possible.

**Describe the solution you'd like**
When BOOTLOADER_MCUBOOT is enabled and image grows beyond the point where addition of mcuboot data at run-time is possible, warning should be issued.

**Describe alternatives you've considered**
Leaving it to user tests of application: this situation will possibly be detected by users at application tests, but it often results in "always pending" state of secondary image and spawns reports of broken mcuboot/Zephyr.

Laczen, 2021-12-06:
@de-nordic, probably this is something that needs to be addressed by the image signing application. During the image signing there is extra data added to the image (the tlv's) that might leed to this situation.

de-nordic, 2021-12-06:
> @de-nordic, probably this is something that needs to be addressed by the image signing application. During the image signing there is extra data added to the image (the tlv's) that might leed to this situation.

@Laczen True, this should probably be done with imgtool but should also be fixed in "west sign" to at least notify user.

Except for the TLV, you mention, there is additional information that is added as part of swap process, for example to track how many pages of flash have been swapped, in case when reset happens while mcuboot does its stuff.
Additionally  the "swap-move" process requires one additional page, to be able to shift the image first, which means that you have at least one page less of primary slot to use for your application.
Maybe we should consider reserving the space for both during application build?

Laczen, 2021-12-06:
> @Laczen True, this should probably be done with imgtool but should also be fixed in "west sign" to at least notify user.
>
> Except for the TLV, you mention, there is additional information that is [...]

I know about the tracking of the swap process, I forgot to add the word also (... that might also leed to this situation).

Regarding the swap move, I don't know how robust mcuboot is against being interrupted during the first part of the swap move (and how this is stored in the image area). It would be better not to include the area for the swap move in the image but to consider it as a scratch area. And the same is valid for the swap state, it should not be inside the image.

de-nordic, 2021-12-06:
> 
> Regarding the swap move, I don't know how robust mcuboot is against being interrupted during the first part of the swap move (and how this is stored in the image area).

Not so many problems reported here, so I guess it mostly survives.

> It would be better not to include the area for the swap move in the image but to consider it as a scratch area. And the same is valid for the swap state, it should not be inside the image.

This actually happens in mcuboot and the mcugmr has nothing to say (at least now), the problem is that developers should be notified that update will not be possible before they even attempt it.

Laczen, 2021-12-06:
> This actually happens in mcuboot and the mcugmr has nothing to say (at least now), the problem is that developers should be notified that update will not be possible before they even attempt it.

OK, but since this is mcuboot specific, the best way to handle this would be inside `imgtool` not producing a signed image. BTW is there an easy way to calculate the space for image swapping state without knowing how mcuboot has been compiled (is the write-block-size not used to determine the area) ?

utzig, 2021-12-06:
> The mcuboot needs some bytes at the end of image slot to store progress of swap and some flags, which is not reserved during build; this means that when image grows beyond certain point the mcuboot will no longer be able to swap images, making DFU updates not possible.

This was already implemented a long time ago, which is why `-S` was added for slot size. It was used together with `--align` to determine the exact free space based on TLVs, trailer, etc. The caveat is that it only works correctly for `swap-with-scratch`, since that allows the image to almost touch the trailer while swapping. `swap-without-scratch` moves sectors, so it's sector aligned on size, which would probably need a new option for providing the sector size used. Also it's possible that since this was not updated in a long time, that it might be missing some information to do the calculation correctly!

nordicjm, 2024-02-09:
Fixed in 3.6

closed 2024-02-09 by nordicjm