zephyrproject-rtos/zephyr #40903: documentation generation fails on function typedefs

npitre, 2021-12-03:
**Describe the bug**

The source code parsing tool is having issues with function typedefs.
And here I insist on "function" typedefs as opposed to "function pointer"
typedefs. Something like:

```
typedef bool (npf_test_fn_t)(struct npf_test *test, struct net_pkt *pkt);
```

The result is:

```
Warning, treated as error:
/home/runner/work/zephyr/zephyr/doc/_build/src/reference/networking/net_pkt_filter.rst:92:Type
must be either just a name or a typedef-like declaration.
If just a name:
  Error in declarator or parameters
  Invalid C declaration: Expected identifier in nested name, got keyword: bool [error at 4]
    bool() npf_test_fn_t (struct npf_test *test, struct net_pkt *pkt)
    ----^
If typedef-like declaration:
  Error in declarator
  If declarator-id with parameters:
    Invalid C declaration: Expected identifier in nested name. [error at 4]
      bool() npf_test_fn_t (struct npf_test *test, struct net_pkt *pkt)
      ----^
  If parenthesis in noptr-declarator:
    Error in declarator or parameters
    Invalid C declaration: Expected identifier in nested name. [error at 5]
      bool() npf_test_fn_t (struct npf_test *test, struct net_pkt *pkt)
      -----^
```

As can be seen above, parenthesis are moved to the bool keyword as if that was
a function which is not what the original code does at all.

Moving the typedef to a function pointer does not exhibit such issue:

```
typedef bool (*npf_test_fn_t)(struct npf_test *test, struct net_pkt *pkt);
```

but that is an unacceptable compromise as it makes the code much less
straightforward and enjoyable to work with.

**To Reproduce**

Check out PR #39936 and build the documentation.

Or look at the corresponding CI failure report.

**Expected behavior**

The CI documentation check should pass.

**Impact**

This is preventing PR #39936 from reaching a mergeable state.

gmarull, 2021-12-06:
This is really a Sphinx C domain limitation, we can't fix it. Meanwhile, I'd suggest moving typedef to a function pointer and declare external functions without any typedef.

npitre, 2021-12-12:
Someone more familiar with the Sphinx tool could file a bug report with 
that project.

What I did is to hide that typedef from view. This is an implementation 
detail that users of the code don't need to know about anyway.

gmarull, 2021-12-13:
> Someone more familiar with the Sphinx tool could file a bug report with that project. What I did is to hide that typedef from view. This is an implementation detail that users of the code don't need to know about anyway.

You are welcome to open an issue on the [Sphinx repository](https://github.com/sphinx-doc/sphinx) with an example to reproduce the problem. I'll close this issue, since it's not really a Zephyr bug.

closed 2021-12-13 by gmarull