zephyrproject-rtos/zephyr #40916: Assertion in nordic's BLE controller lll.c:352

ipshiv, 2021-12-06:
**Describe the bug**
Time after time I catch next assert (after periodic public configuration and next once an hour)
```
00> ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:352
00> E: r0/a1: 0x00000003 r1/a2: 0x20009cf0 r2/a3: 0x00000001
00> E: r3/a4: 0x20009d46 r12/ip: 0xa0000000 r14/lr: 0x0002ecbb
00> E: xpsr: 0x61000011
00>...Code
```

**To Reproduce**
Assert occurs most often when I use 3+ mesh sensors in one sensor server (from NCS) with periodic publication configured.

**Expected behavior**
Expected periodic publishing without resets

**Impact**
Pre-release stopper

**Logs and console output**
RTT Log
```
00> I: 2 Sectors of 4096 bytes
00> I: alloc wra: 0, fb8
00> I: data wra: 0, 34
00> I: HW Platform: Nordic Semiconductor (0x0002)
00> I: HW Variant: nRF52x (0x0002)
00> I: Firmware: Standard Bluetooth controller (0x00) Version 2.6 Build 99
00> I: No ID address. App must call settings_load()
00> I: Bluetooth ready
00> D: Sensor 0x0052
00> D: Sensor 0x0054
00> D: Sensor 0x0057
00> D: Sensor 0x0059
00> D: Sensor 0x00a3
00> I: Identity: C9:C8:96:33:17:06 (random)
00> I: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x01db
00> I: LMP: version 5.2 (0x0b) subver 0xffff
00> I: Device UUID: 00000000-0000-0080-b149-e8a3dc72e362
00> I: Bluetooth mesh initialized
00> D: ctrl1=0x47 @tick=352898
00> I: Primary Element: 0x0008
00> D: net_idx 0x0000 flags 0x00 iv_index 0x0000
00> D: addr 0x0008 elem_count 1
00> D: addr 0x0008 mod_count 10 vnd_mod_count 0
00> D: 
00> D: Sensor 0x0052
00> D: Sensor 0x0054
00> D: Sensor 0x0057
00> D: Sensor 0x0059
00> D: Sensor 0x00a3
00> I: Identity: C9:C8:96:33:17:06 (random)
00> I: HCI: version 5.2 (0x0b) revision 0x0000, manufacturer 0x01db
00> I: LMP: version 5.2 (0x0b) subver 0xffff
00> I: Device UUID: 00000000-0000-0080-b149-e8a3dc72e362
00> I: Bluetooth mesh initialized
00> D: ctrl1=0x47 @tick=352898
00> I: Primary Element: 0x0008
00> D: net_idx 0x0000 flags 0x00 iv_index 0x0000
00> D: addr 0x0008 elem_count 1
00> D: addr 0x0008 mod_count 10 vnd_mod_count 0
00> D: 
00> D: app_idx 0xfffe src 0x0001 dst 0x0008
00> D: len 3: 8008ff
00> D: OpCode 0x00008008
00> D: net_idx 0x0000 app_idx 0xfffe src 0x0001 len 1: ff
00> D: Composition page 255 not available
00> D: net_idx 0x0000 app_idx 0xfffe dst 0x0001
00> D: len 36: 0200db0110c1203e0a00070001000a00000002000c100010041006100710011000110111
00> W: No matching TX context for ack
00> D: app_idx 0xfffe src 0x0001 dst 0x0008
00> D: len 2: 800c
00> D: OpCode 0x0000800c
00> D: net_idx 0x0000 app_idx 0xfffe src 0x0001 len 0: 
00> D: net_idx 0x0000 app_idx 0xfffe dst 0x0001
00> D: len 3: 800e64
00> D: app_idx 0xfffe src 0x0001 dst 0x0008
00> D: len 20: 000000003fe6b1c56ae39fff7f3d5062c9addef5
00> D: OpCode 0x00000000
00> D: AppIdx 0x0000 NetIdx 0x0000
00> D: net_idx 0x0000 app_idx 0xfffe dst 0x0001
00> D: len 6: 800300000000
00> D: app_idx 0xfffe src 0x0001 dst 0x0008
00> D: len 3: 80240a
00> D: OpCode 0x00008024
00> D: net_idx 0x0000 app_idx 0xfffe src 0x0001 len 1: 0a
00> D: Transmit 0x0a (count 2 interval 20ms)
00> D: net_idx 0x0000 app_idx 0xfffe dst 0x0001
00> D: len 3: 80250a
00> D: app_idx 0xfffe src 0x0001 dst 0x0008
00> D: len 8: 803d080000000011
00> D: OpCode 0x0000803d
00> D: ID 0x1100 addr 0x0008
00> D: model 0x20000c18 key_idx 0x000
00> D: status 0x00
00> D: net_idx 0x0000 app_idx 0xfffe dst 0x0001
00> D: len 9: 803e00080000000011
00> D: model key 0x0000
00> D: Stored bt/mesh/s/8/bind value
00> D: app_idx 0xfffe src 0x0001 dst 0x0008
00> D: len 12: 03080000c00000ff4a090011
00> D: OpCode 0x00000003
00> D: elem_addr 0x0008 pub_addr 0xc000 cred_flag 0
00> D: pub_app_idx 0x000, pub_ttl 255 pub_period 0x4a
00> D: retransmit 0x09 (count 1 interval 100ms)
00> D: ID 0x1100 addr 0x0008
00> D: period 10000 ms
00> D: net_idx 0x0000 app_idx 0xfffe dst 0x0001
00> D: len 14: 801900080000c00000ff4a090011
00> W: No matching TX context for ack
00> D: Stored bt/mesh/s/8/pub value
00> D: 
00> D: #1 Period: 10000 ms Divisor: 0 (normal)
00> W: Device INA219 is not ready
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: sample: 114
00> D: Temperature:28,500000
00> D: Temp sensor value: 28,500000
00> D: Delta: 28 (28 - 0) thrsh: 0
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: Delta: 4 (4 - 0) thrsh: 0
00> D: Accel value #0: 0,-38304
00> D: Accel value #1: -1,-264032
00> D: Accel value #2: 9,499392
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: err 0
00> D: Publishing next time in 100ms
00> D: 
00> D: err 0
00> D: Publishing took 3205ms
00> D: Publishing next time in 6795ms
00> D: 
00> D: #2 Period: 10000 ms Divisor: 0 (normal)
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: sample: 114
00> D: Temperature:28,500000
00> D: Temp sensor value: 28,500000
00> D: Delta: 0 (28 - 28) thrsh: 0
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: Delta: 0 (4 - 4) thrsh: 0
00> D: Accel value #0: 0,-114912
00> D: Accel value #1: -1,-225728
00> D: Accel value #2: 9,537696
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: err 0
00> D: Publishing next time in 100ms
00> D: 
00> D: err 0
00> D: Publishing took 3204ms
00> D: Publishing next time in 6796ms
00> D: 
00> D: #3 Period: 10000 ms Divisor: 0 (normal)
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: sample: 114
00> D: Temperature:28,500000
00> D: Temp sensor value: 28,500000
00> D: Delta: 0 (28 - 28) thrsh: 0
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: Delta: 0 (4 - 4) thrsh: 0
00> D: Accel value #0: 0,-153216
00> D: Accel value #1: -1,-225728
00> D: Accel value #2: 9,499392
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: err 0
00> D: Publishing next time in 100ms
00> D: 
00> D: err 0
00> D: Publishing took 3204ms
00> D: Publishing next time in 6796ms
00> D: 
00> D: #4 Period: 10000 ms Divisor: 0 (normal)
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: sample: 114
00> D: Temperature:28,500000
00> D: Temp sensor value: 28,500000
00> D: Delta: 0 (28 - 28) thrsh: 0
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: Delta: 0 (4 - 4) thrsh: 0
00> D: Accel value #0: 0,-76608
00> D: Accel value #1: -1,-187424
00> D: Accel value #2: 9,576000
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: err 0
00> D: Publishing next time in 100ms
00> D: 
00> D: err 0
00> D: Publishing took 3204ms
00> D: Publishing next time in 6796ms
00> D: 
00> D: #5 Period: 10000 ms Divisor: 0 (normal)
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: sample: 114
00> D: Temperature:28,500000
00> D: Temp sensor value: 28,500000
00> D: Delta: 0 (28 - 28) thrsh: 0
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: Delta: 0 (4 - 4) thrsh: 0
00> D: Accel value #0: 0,-38304
00> D: Accel value #1: -1,-302336
00> D: Accel value #2: 9,461088
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: err 0
00> D: Publishing next time in 100ms
00> D: 
00> D: err 0
00> D: Publishing took 3204ms
00> D: Publishing next time in 6796ms
00> D: 
00> D: #6 Period: 10000 ms Divisor: 0 (normal)
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: sample: 113
00> D: Temperature:28,250000
00> D: Temp sensor value: 28,250000
00> D: Delta: 0 (28 - 28) thrsh: 0
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: Delta: 0 (4 - 4) thrsh: 0
00> D: Accel value #0: 0,-38304
00> D: Accel value #1: -1,-225728
00> D: Accel value #2: 9,537696
00> D: Delta: 0 (0 - 0) thrsh: 0
00> D: err 0
00> D: Publishing next time in 100ms
00> D: 
00> ASSERTION FAIL @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:352
00> E: r0/a1:  0x00000003  r1/a2:  0x20009be0  r2/a3:  0x00000001
00> E: r3/a4:  0x20009c3e r12/ip:  0x00000000 r14/lr:  0x0002fd57
00> E:  xpsr:  0x61000011
00> E: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00000000  s[ 3]:  0x00000000
00> E: s[ 4]:  0x00000000  s[ 5]:  0x00000000  s[ 6]:  0x00000000  s[ 7]:  0x00000000
00> E: s[ 8]:  0x00000000  s[ 9]:  0x00000000  s[10]:  0x00000000  s[11]:  0x00000000
00> E: s[12]:  0x00000000  s[13]:  0x00000000  s[14]:  0x00000000  s[15]:  0x00000000
00> E: fpscr:  0x00031f06
00> E: Faulting instruction address (r15/pc): 0x0002fd62
00> E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
00> E: Fault during interrupt handling
00> 
00> E: Current thread: 0x20002bc8 (unknown)
00> E: Resetting system
00> 
00> 

```

**Environment (please complete the following information):**
* Board: Custom board based on nRF52840
* Zephyr: build: v2.6.99-ncs1-1
* NCS: 1.7.1
* Toolchain: gcc-arm-none-eabi-9-2019-q4-major \ gcc-arm-none-eabi-10-2020-q4-major
* CMAKE_VERSION 3.20.5
* West: version "0.10.1"

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

cvinayak, 2021-12-08:
@ipshiv Please cherry-pick this commit: https://github.com/zephyrproject-rtos/zephyr/commit/ed7e9028139e4c3d2a7deb98ccc5400cd6fc41c4

If you are still facing the assertion, I will need a branch with an application that I can use on nRF5x DK boards, preferably only build (so I can add some debug code) the application in the branch, flash and reproduce the assertion.

ipshiv, 2021-12-08:
Thanks for response! Unfortunately assert is still there. I'll prepare sample for DK in couple days.

ipshiv, 2021-12-17:
@cvinayak does The Nordic Thingy:52™ (nRF6936)  ok?

cvinayak, 2021-12-17:
@ipshiv I dont have Thingy:52, but you could make the sample and I can use as nRF52832 DK, if sensors are not required to reproduce the assertions.

carlescufi, 2021-12-22:
@ipshiv can you please provide a sample that we can run either on 52832 or 52840 DK?

carlescufi, 2022-02-10:
Closing due to inactivity. Please reopen if this is still an issue.

closed 2022-02-10 by carlescufi