zephyrproject-rtos/zephyr #40917: twister --export-tests export all cases even this case can not run on given platform

hakehuang, 2021-12-06:
**Describe the bug**
`scripts/twister -p frdm_k64f -T samples -T tests  --enable-slow --export-tests frdm_k64f.csv`
and I get exported cases which is a full set, not dedicated to given platform

However, looking through the attached .CSV file generated, I see many applications listed that would not apply to this FRDM-K64F board.  I picked out a few examples below that are for different boards or architectures.  Is there a way to strip out the paths for the projects that Twister would skip for the selected platform?  This file includes 1526 projects, but many don't apply to this board.

samples/boards/intel_s1000_crb/audio,sample.board.intel_s1000_crb.audio
samples/bluetooth/scan_adv,sample.bluetooth.scan_adv
samples/drivers/lora/send,sample.driver.lora.send
tests/arch/x86/nmi,arch.interrupt.nmi
tests/arch/arm64/arm64_high_addresses,arch.arm64.high_addr.high_sram_low_vm
tests/drivers/can/stm32,drivers.can.stm32
tests/drivers/build_all/ieee802154,ieee802154.build.kw41z

**Expected behavior**
the exported cases shall be filterred by platform

**Impact**
export cases

hakehuang, 2021-12-06:
`scripts/twister -p frdm_k64f -T samples -T tests --enable-slow --save-tests frdm_k64f.csv`

does not work well either, I still get 1440 cases, which has many unsupported cases in it.

nashif, 2021-12-06:
> does not work well either, I still get 1440 cases, which has many unsupported cases in it.

look at the output, it does have:

`samples/boards/intel_s1000_crb/dmic/sample.board.intel_s1000_crb.dmic,arm,frdm_k64f,skipped,,na,,0,0`

it is marked as `skipped`, which is correct.

closed 2021-12-07 by hakehuang