zephyrproject-rtos/zephyr #40920: Bluetooth audio: client/servet naming scheme

koffes, 2021-12-06:
As an example, we now use `CONFIG_BT_VCS_CLIENT` and `CONFIG_BT_VCS`. This structure applies to many of the services.
IMO, this is confusing for the user. Suggest using `CLIENT` and `SERVER` explicitly, or figure out some alternative naming scheme.

@Thalley

Thalley, 2021-12-06:
The suggestion is noted. 

Generally, the server/client naming scheme may not be usable at all, and this may be a much larger issue. 

For VCS (or rather the VCP), there's a typically a client (called `volume controller`), and a server (called `volume renderer`). VCP _requires_ but does not define VCS, so VCS could technically be used by other profiles, and in fact AICS is a good example of that, as that is used by both VCP and MICP. The point of this, is that there is not a 1:1 between a service and a server, but rather a n:n as AICS can be used by multiple servers (though with their own respective instances of AICS) and likewise a single server may include multiple AICS instances. 

AICS and VOCS are currently designed to be profile agnostic for both sides (i.e. both server and client agnostic), and as such do not really conform to the suggest _SERVER and _CLIENT naming. Given that we need the services to be individually address and server agnostic, then perhaps the issue is on the other side (currently name client). https://en.wikipedia.org/wiki/Client%E2%80%93server_model suggests `service requester` as the entity that interact with a service (and not a server). 

The naming may also be impacted by how we structure the code. The services are typically defined separately from the profiles (makes them reusable by other profiles), and if we want to offer that same flexibility, then we also need a separation between the service instances and the servers/profile roles. 

Right now, the server of the VCP (i.e. the `volume renderer`) is implemented simply as VCS, but perhaps we need to restructure the code such that we have a `bt_vcp_volume_renderer` and a `bt_vcp_volume_controller` (as defined by the profile), which _uses_ (but does not implement) the VCS, AICS and VOCS as well as the service requesters for these. This would expose an API to the services (e.g. the `bt_vcs_*`) and an API to the profile role (e.g. `bt_vcp_volume_renderer_*`), where the service API could potentially be internal-only. 
Alternatively, the profile implementations _implement_ the services themselves. This was the original design for VCP and MICP, but since the AICS and AICS requester (client) were identical for VCP and MICP, it was moved to a separate implementation that could by used by any implementation. 

In summary:

1. There isn't a 1:1 between services and servers, but rather a n:n
2. Client may not be the proper term for an entity that accesses a service (but perhaps service requester is)
3. There isn't always only a client and server and a profile, but may be other roles as well (e.g. BAP has 6 roles). 
4. Service implementations may be either implemented by a role or used by a role via an API, the former resulting in almost duplicated code and additional memory, and the latter of which requires additional APIs (possibly internal-only). 
5. Given that most profiles only have a server and a client role defined, we have simply used the client/server naming, but perhaps we should move to a general design of using the profile names instead.

Vudentz, 2021-12-17:
First Server and Client are GATT roles and there can be used simultaneously in a connection. Typically each profile has a service specification which defines the GATT attributes to be implemented by the Server and a Profile implementation that specifies the client procedures to access the GATT attributes of the Service, now there are some profiles that only have a service specification so they either define the client procedures directly or in some very special cases there could be no client procedures at all.

With this in mind I would recommend not mixing profiles roles with GATT roles, in fact even the GATT API don't use roles just the procedure names.

Thalley, 2021-12-20:
> Typically each profile has a service specification which defines the GATT attributes to be implemented by the Server and a Profile implementation that specifies the client procedures to access the [...]

I'm not sure that's strictly correct. We have profiles that specify roles without the use of any services (e.g. broadcast source and sink), and some roles the use multiple services (e.g. unicast server). 

I think it is important to keep in mind that service != server, and profile != client - In fact some of the profiles roles are not even clients or servers at all!

> GATT API don't use roles just the procedure names.

I like this idea, and is what I have attempted to do with the VCS API (VCS should in that regard be changed to VCP), as a procedure is defined in a specific way and very often (if not always) works in one way, and one way only, so that it can be applied to any scenario (locally executed or remotely executed). 

What I am starting to think is that we need to use the profile roles and procedures for the naming, rather than server/client. Furthermore, we could split the service API from the server API, like what has been done for e.g. AICS and VOCS, which do not have a server implementation, but are used by the VCP server (volume renderer), or similar to how the media player uses the MCS. 
We can then consider whether we also want to expose the service APIs as public include headers, or keep them internally.

Thalley, 2022-03-08:
This issue relates to https://github.com/zephyrproject-rtos/zephyr/discussions/33592

Thalley, 2022-04-05:
@Thalley To provide a few suggestions for a future naming scheme in the upcoming F2F meeting in Oslo.

Thalley, 2022-04-08:
**Suggestions for naming schemes**

The current scheme we have is that we use `bt_<service>_<action>` for the server API as well as common functionality, and `bt_<service>_<client>_<action>` for the client.
Similarly is the server functionality enabled by `CONFIG_BT_<SERVICE>` and the client functionality enabled by `CONFIG_BT_<SERVICE>_CLIENT`.
Examples for this is `CONFIG_BT_VCS` and `CONFIG_BT_VCS_CLIENT` that uses functions like
 - `bt_vcs_register` (service/server only)
 - `bt_vcs_client_conn_get` (client only)
 - `bt_vcs_vol_get` (common)

Some of the challenges that we face is that
1) Some profiles have multiple roles (BAP has 6 roles, CAP has 3, HAP has 3), so there is not ever only a client-server relationsship
2) Some roles are not clients nor servers at all, e.g. the BAP broadcast source and BAP broadcast sink roles
3) Some roles use multiple services, e.g. the BAP unicast server which has both PACS and ASCS
4) Some functionality is support by multiple roles, e.g. `bt_audio_stream_send` which works for the BAP unicast client, BAP unicast server and BAP broadcast source

**Suggestion 1.A: Explicit server infix**
Add `SERVER`/`server` for the server related API and Kconfigs. In the example of the Volume Control Profile (VCP) and service (VCS), this would add the following changes
- `CONFIG_BT_VCS` => `CONFIG_BT_VCS_SERVER`
- `bt_vcs_register` => `bt_vcs_server_register`
and leave the current `client` infixes and keep the common code with just the `_vcs_` infix.

**Suggestion 1.B: Explicit server infix**
Add `SERVER`/`server` for the server related API and Kconfigs AND change to profile name. In the example of the Volume Control Profile (VCP) and service (VCS), this would add the following changes
- `CONFIG_BT_VCS` => `CONFIG_BT_VCP_SERVER`
- `bt_vcs_register` => `bt_vcp_server_register`
and leave the current `client` infixes and keep the common code with just the `_vcp_` infix.

**Suggestion 2: Use profile names and roles**
This suggestion is to remove the client/server naming (unless part of the profile) as use the spec defined roles for the API.
In the example of the Volume Control Profile (VCP) and service (VCS), this would add the following changes
- `CONFIG_BT_VCS` => `CONFIG_BT_VOLUME_RENDERER` OR `CONFIG_BT_VCP_RENDERER` OR `CONFIG_BT_VCP_VOLUME_RENDERER`
- `CONFIG_BT_VCS_CLIENT` => `CONFIG_BT_VOLUME_CONTROLLER` OR `CONFIG_BT_VCP_CONTROLLER` OR `CONFIG_BT_VCP_VOLUME_CONTROLLER`
- `bt_vcs_register` => `bt_vcp_renderer_register`
- `bt_vcs_client_conn_get` => `bt_vcp_controller_conn_get`
- `bt_vcs_vol_get` => `bt_vcp_vol_get`

This suggestion is already partly used for e.g. BAP where we have `bt_audio_unicast_client`, `bt_audio_unicast_server`, `bt_audio_broadcast_source` and `bt_audio_broadcast_sink` precisely because we do not have a 1:1 between clients and services. 
With this suggestion these may also be renamed as
- `bt_audio_unicast_client` => `bt_bap_unicast_client`
- `bt_audio_unicast_server` => `bt_bap_unicast_server`
- `bt_audio_broadcast_source` => `bt_bap_broadcast_source`
- `bt_audio_broadcast_sink` =>`bt_bap_broadcast_sink`

**Other considerations**

1. We should also decide whether to use the `bt_audio` prefix or not. Some of the Services and Profiles are actually not directly involved with the audio streams, and some like Coordinated Set Identification Profile (CSIP) is not even audio-only. 
We can have `bt_<service>_<action>` or `bt_audio_<service>_<action>`. This also applies for Kconfig options, e.g. `BT_VCS` or `BT_AUDIO_VCS`. 
2. For some we may even also want to expose both a service API and a role API. E.g. we could have a `bt_vcp_volume_renderer_<action>` and a `bt_vcs_<action>`, but not such use case has been found yet.

Thalley, 2022-05-19:
Multiple smaller issues have been created, so closing this.

closed 2022-05-19 by Thalley