zephyrproject-rtos/zephyr #40925: mesh_badge not working reel_board_v2

boschkundendienst, 2021-12-06:
**Describe the bug**
I have a reel_board_v2 with the new display `HINK-E0213A22` on the display cable which si then a `GDEH0213B72` so the board designation is `reel_board_V2`. I can try all samples that do not use the display, e.g. `samples/bluetooth/beacon` or ones involving the different LEDs. After some tests I wanted to flash the original software `samples/boards/reel_board/mesh_badge` but it does not work.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
   Reel Board V2
 - What have you tried to diagnose or workaround this issue?
   Tried different IDEs
   Tried various parameters when flashing like delete flash etc.

**To Reproduce**
Steps to reproduce the behavior:
1. cd ~/zephyrproject/zephyr
2. source ~/zephyrproject/.venv/bin/activate
3. west build -p always -b reel_board_v2 samples/boards/reel_board/mesh_badge

The above command works without error and produces the following ouput:

```
-- west build: making build dir /home/pcfreak/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/pcfreak/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/pcfreak/zephyrproject/zephyr/samples/boards/reel_board/mesh_badge
-- Zephyr version: 2.7.99 (/home/pcfreak/zephyrproject/zephyr), build: v2.7.99-1980-g81b1e7fdacf0
-- Found Python3: /home/pcfreak/zephyrproject/.venv/bin/python3 (found suitable exact version "3.8.10") found components: Interpreter 
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: reel_board_v2
-- Cache files will be written to: /home/pcfreak/.cache/zephyr
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Using toolchain: zephyr 0.13.2 (/home/pcfreak/zephyr-sdk-0.13.2)
-- Found dtc: /home/pcfreak/zephyr-sdk-0.13.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/pcfreak/zephyrproject/zephyr/boards/arm/reel_board/reel_board_v2.dts
-- Generated zephyr.dts: /home/pcfreak/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/pcfreak/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/pcfreak/zephyrproject/zephyr/build/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/pcfreak/zephyrproject/zephyr/build/zephyr/dts.cmake
Parsing /home/pcfreak/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/pcfreak/zephyrproject/zephyr/boards/arm/reel_board/reel_board_v2_defconfig'
Merged configuration '/home/pcfreak/zephyrproject/zephyr/samples/boards/reel_board/mesh_badge/prj.conf'
Configuration saved to '/home/pcfreak/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/pcfreak/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/pcfreak/zephyr-sdk-0.13.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pcfreak/zephyrproject/zephyr/build
-- west build: building application
[1/289] Preparing syscall dependency handling

[279/289] Linking C executable zephyr/zephyr_pre0.elf

[283/289] Linking C executable zephyr/zephyr_pre1.elf

[289/289] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      269228 B         1 MB     25.68%
            SRAM:       59686 B       256 KB     22.77%
        IDT_LIST:          0 GB         2 KB      0.00%
```

4. Flashing with `west flash --skip-rebuild`

This produces the following output:

```
-- west flash: rebuilding
ninja: no work to do.
[... 92 lines cut ...]
(gdb) continue
Continuing.
```
**Activity LED is flashing continously**

6. stop with CTRL-C or pressing reset button

**Expected behavior**
Sample should flash without errors and work.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

I can not use the board as expected. It is a showstopper!

**Logs and console output**
If you need more logs just inform me.

**Environment (please complete the following information):**
 - OS: (Linuxt/Ubuntu 20.04.3 LTS)
 - Toolchain (Zephyr SDK)
 - ??

**Additional context**
I am not a developer but a sysadmin. I can give you all information you need but I can not debug C-Code!

jfischer-no, 2021-12-07:
Either https://github.com/zephyrproject-rtos/zephyr/pull/40765 or https://github.com/zephyrproject-rtos/zephyr/pull/39708 will fix it, whatever gets merged first.

boschkundendienst, 2021-12-08:
Thanks for your reply. Will look into this, when I receive my new board (sent the current one back to make sure it is not hardware-broken).

Question: What fork/tree/commit did they flash originally since when delivered a working mesh_badge sample was on the device?!

jfischer-no, 2021-12-08:
> Question: What fork/tree/commit did they flash originally since when delivered a working mesh_badge sample was on the device?!

I cannot say, but zephyr-v2.6.0 does not have this bug.

carlescufi, 2022-01-26:
This is now fixed

PR #41065 (2022-01-15): [Backport v2.7-branch] drivers: ssd16xx: fix driver initialization
backport 5e722c436842068c3ac1d75b440ea505035dfaf6

ssd16xx driver is not well designed and does not pass configuration
via device->config but via struct ssd16xx_data, this was not taken
into account in the commit 4d6d50e2bc60
("display: ssd16xx: convert to spi_dt_spec").

Fixes: #40925
closed 2022-01-26 by carlescufi