zephyrproject-rtos/zephyr #40931: mgmt/mcumgr/lib: RFC: Addion of new error codes

de-nordic, 2021-12-06:
**Is your feature request related to a problem? Please describe.**
The subset of mcumgr reported error coded is quite limited and makes it hard to precisely pin-point what has really went wrong with requested action. 

**Describe the solution you'd like**
My proposal is to add four new error codes of following meaning:
 1. Read error - this code would be issued when API call for read fails; it means "we could not get the data from device to RAM";
 2. write error - this code would be issued when write API call fails in any way; it means "we attempted the write but we got error and can not confirm any data got out there"
 3. write-read-verify error - this code would be issued when verification of attempted write occurs; the error is reported when last of sequence of immediate write-read-verify operations fail, for example the sequence leading to such error would look like this:
 
          Write of size S of data D0 via API A to location L has been successful (2. did not happen)
          Read of size S of data D1 via API A from location L has been successful (1. did not happen)
          Contents of D0 compared to contents of D1, directly or checksums, are not found to be equal

4. Checksum calculation failed  - as a result of direct request of checksum calculation when condition for 3. is not met and 1. does not apply.
  
**Describe alternatives you've considered**
Stay with what we have and suffer.

**Additional context**
This may be problematic for existing applications and mcumgr-cli, which currently support only predefined subset of error codes as defined here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/mgmt/mcumgr/lib/mgmt/include/mgmt/mgmt.h#L42-L55

lairdjm, 2022-03-23:
Additional ones I can think of from errno:
```
#define EIO 5           /**< I/O error */
#define E2BIG 7         /**< Arg list too long */
#define EACCES 13       /**< Permission denied */
#define ENOSPC 28       /**< No space left on device */
```

Thinking about it, if the plan is to replace mcumgr with a python version, would it make more sense to get rid of the mcumgr error codes entirely and replace them with errno directly?

nordicjm, 2023-05-16:
Completed by https://github.com/zephyrproject-rtos/zephyr/pull/54041

closed 2023-05-16 by nordicjm