zephyrproject-rtos/zephyr #40936: STM32 ADC gets stuck in Calibration

pdietl, 2021-12-06:
**Describe the bug**
Here https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/adc/adc_stm32.c#L595 we try to calibrate the ADC, but at least on STM32G4, the ADC is not permitted to be calibrated unless the ADC is disabled. 

[image]

I discovered this bug while trying to use the sensor here: https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/sensor/stm32_temp/stm32_temp.c#L116 which requests calibration. I saw the thread in question get stuck here: https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/adc/adc_stm32.c#L337

I need help confirming this ADC disabled requirement on other STM32 models.

FRASTM, 2021-12-07:
that's right, ADC calibration requires ADC disable (ADCEN =0) except for **stm32F1** mcus 
_"Before starting a calibration, the ADC must have been in power-on state (ADON bit = ‘1’) for
at least two ADC clock cycles."_
and stm32L1x (no cal.)

pdietl, 2021-12-07:
I think I may have split our conversation between the PR and issue :p

PR #40937 (2021-12-15): drivers: adc: stm32: Disable ADC before calibration
The STM32 docs state that the ADC may not be calibrated unless
the ADC is disabled (ADEN=0). This commit implements this constraint

Fixes #40936
commit bc1b2628 (2021-12-15): drivers: adc: stm32: Disable ADC before calibration

The STM32 docs state that the ADC may not be calibrated unless
the ADC is disabled (ADEN=0). This commit implements this constraint

Fixes #40936
commit 8369a0a3 (2022-03-02): drivers: adc: stm32: Disable ADC before calibration

The STM32 docs state that the ADC may not be calibrated unless
the ADC is disabled (ADEN=0). This commit implements this constraint

Fixes #40936
closed 2021-12-15 by MaureenHelm