zephyrproject-rtos/zephyr #40942: Xtensa debug bug

hongshui3000, 2021-12-07:
**Describe the bug**
code path: zephyr\arch\xtensa\core\xtensa-asm2-util.S
```
.global _restore_context
_restore_context:
	call0 xtensa_restore_high_regs

	l32i a0, a1, BSA_PC_OFF
	wsr a0, RCTX_EPC_REG
	l32i a0, a1, BSA_PS_OFF
	wsr a0, RCTX_EPS_REG

	l32i a0, a1, BSA_SAR_OFF
	wsr a0, SAR
#if XCHAL_HAVE_LOOPS
	l32i a0, a1, BSA_LBEG_OFF
	wsr a0, LBEG
	l32i a0, a1, BSA_LEND_OFF
	wsr a0, LEND
	l32i a0, a1, BSA_LCOUNT_OFF
	wsr a0, LCOUNT
#endif
#if XCHAL_HAVE_S32C1I
	l32i a0, a1, BSA_SCOMPARE1_OFF
	wsr a0, SCOMPARE1
#endif
#if XCHAL_HAVE_THREADPTR && defined(CONFIG_THREAD_LOCAL_STORAGE)
	l32i a0, a1, BSA_THREADPTR_OFF
	wur a0, THREADPTR
#endif
	rsync

	l32i a0, a1, BSA_A0_OFF
	l32i a2, a1, BSA_A2_OFF
	l32i a3, a1, BSA_A3_OFF
	addi a1, a1, BASE_SAVE_AREA_SIZE

	rfi XCHAL_NUM_INTLEVELS ------------------------------------>  debug not work
```
[image]

XCHAL_NUM_INTLEVELS == 6 is debug INT 

When the interrupt is added here, the address of EPC6 will become the current address, and rfi 6 will not be able to return

cfriedt, 2021-12-07:
Could be related to #40946 .

andyross, 2021-12-07:
Zephyr isn't handling the debug interrupts at all, though.  We simply don't support this feature (not that we shouldn't!  we just don't have the code written).  I'm guessing the bug report here is that you have an application which is adding its own handler for the debug interrupt, that it's using the EPC/EPS registers assigned to that interrupt level, and that this is being clobbered by Zephyr's own use of those registers?

To explain: what's happening here is that Zephyr simply doesn't care about which interrupt level we're handling.  Our nested interrupt architecture is software-based.  So in principle we could use any EPCx/EPSx pair and return to them with "RFI x".

As it happens, the code as written always chooses the highest interrupt level.  You can see this logic at the top of that file, where the RCTX_EP{C,S}_REG values are defined.  My memory is that this was to work around a bug in Qemu, which would misbehave if you tried to return into an interrupt level lower than the one that is currently executing.  Hardware seems not to care and per the ISA reference there's no dependency on any state.

So... I'm tempted to not-a-bug this, simply because the Zephyr interrupt architecture isn't designed to live in a single vector table with application-defined handlers that don't conform to Zephyr conventions.   The "Right Thing" for your application to be doing is to register your debug handler with IRQ_CONNECT() and enter and exit it on Zephyr's code paths.

But for debug handlers, I guess I can see an argument for wanting to keep this separate from the OS.  And in any case it's easy to fix: you could specify a separate way (kconfig, for example) to define the RCTX_* register choices and leave the current "max" value as the default.  Again, any[1] of the register pairs will work on hardware.

[1] Actually no.  IIRC level 1 is special owing to the older XEA1 semantics.  But any level 2 or higher works the same way.

hongshui3000, 2021-12-08:
Let me re-describe the phenomenon of the problem:
I use the candence Xplorer IDE debugger (ocd). When I add a breakpoint and the cpu is paused, the address displayed by the EPCS (the highest interrupt level) used by the debug interrupt and the address displayed by my PC are the same breakpoint address, then if At this time, it happened to be the EPCS (the highest interrupt level) used in zephyr, and the problem appeared, because the EPCS used in zephyr was actually changed by my debug interrupt.

1. DEBUG interrupt will change his related EPCS
2. This EPCS may be being used by zephyr
3. The RFI cpu instruction will not return the correct address

andyross, 2021-12-08:
OK, that sounds exactly like what I described then.   Is it really not possible to make your debug exception handler into a proper Zephyr interrupt?  I'll admit I'm not familiar with the protocol used by the Xplorer debugger or how it's supposed to be integrated.  Just in general, having mixed-convention interrupts in the same vector table is likely to be difficult.

If not, please try the suggested workaround, setting the Zephyr interrupt EPC/EPS registers to a different pair.  If that works, I'd be happy to +1 a patch that selects the "highest interrupt level that is not DEBUG" as the default.

hongshui3000, 2021-12-08:
My current solution is to change my EPCS (LEVLE5 MY NMI INT),Nmi int is not used in my system, so it works
```
#if XCHAL_NUM_INTLEVELS == 1
# define RCTX_EPC_REG		EPC1
# define RCTX_EPS_REG		EPS1
#elif XCHAL_NUM_INTLEVELS == 2
# define RCTX_EPC_REG		EPC2
# define RCTX_EPS_REG		EPS2
#elif XCHAL_NUM_INTLEVELS == 3
# define RCTX_EPC_REG		EPC3
# define RCTX_EPS_REG		EPS3
#elif XCHAL_NUM_INTLEVELS == 4
# define RCTX_EPC_REG		EPC5 ---------------------------------------------------------------->my
# define RCTX_EPS_REG		EPS5----------------------------------------------------------------->my
#elif XCHAL_NUM_INTLEVELS == 5
# define RCTX_EPC_REG		EPC5
# define RCTX_EPS_REG		EPS5
#elif XCHAL_NUM_INTLEVELS == 6
# define RCTX_EPC_REG		EPC6
# define RCTX_EPS_REG		EPS6
#elif XCHAL_NUM_INTLEVELS == 7
# define RCTX_EPC_REG		EPC7
# define RCTX_EPS_REG		EPS7
#else
/* Catch all */
# define RCTX_EPC_REG		_CONCAT(EPC, XCHAL_NUM_INTLEVELS)
# define RCTX_EPS_REG		_CONCAT(EPS, XCHAL_NUM_INTLEVELS)
#endif

.global _restore_context
_restore_context:
	call0 xtensa_restore_high_regs

	l32i a0, a1, BSA_PC_OFF
	wsr a0, RCTX_EPC_REG
	l32i a0, a1, BSA_PS_OFF
	wsr a0, RCTX_EPS_REG

	l32i a0, a1, BSA_SAR_OFF
	wsr a0, SAR
#if XCHAL_HAVE_LOOPS
	l32i a0, a1, BSA_LBEG_OFF
	wsr a0, LBEG
	l32i a0, a1, BSA_LEND_OFF
	wsr a0, LEND
	l32i a0, a1, BSA_LCOUNT_OFF
	wsr a0, LCOUNT
#endif
#if XCHAL_HAVE_S32C1I
	l32i a0, a1, BSA_SCOMPARE1_OFF
	wsr a0, SCOMPARE1
#endif
#if XCHAL_HAVE_THREADPTR && defined(CONFIG_THREAD_LOCAL_STORAGE)
	l32i a0, a1, BSA_THREADPTR_OFF
	wur a0, THREADPTR
#endif
	rsync

	l32i a0, a1, BSA_A0_OFF
	l32i a2, a1, BSA_A2_OFF
	l32i a3, a1, BSA_A3_OFF
	addi a1, a1, BASE_SAVE_AREA_SIZE

	rfi XCHAL_NUM_INTLEVELS+1 -------------------------------------------------------------->my
```

andyross, 2021-12-08:
You're welcome to contribute your solution. :)

hongshui3000, 2021-12-08:
MY FIX:
```
 /*
  * Here we use the EPS and EPC of the NMI interrupt, 
  * we need to make sure that the software using this code is 
  * not an NMI interrupt handler
  */
#if XCHAL_NUM_INTLEVELS == 1
# define RCTX_EPC_REG		EPC2
# define RCTX_EPS_REG		EPS2
#elif XCHAL_NUM_INTLEVELS == 2
# define RCTX_EPC_REG		EPC3
# define RCTX_EPS_REG		EPS3
#elif XCHAL_NUM_INTLEVELS == 3
# define RCTX_EPC_REG		EPC4
# define RCTX_EPS_REG		EPS4
#elif XCHAL_NUM_INTLEVELS == 4
# define RCTX_EPC_REG		EPC5
# define RCTX_EPS_REG		EPS5
#elif XCHAL_NUM_INTLEVELS == 5
# define RCTX_EPC_REG		EPC6
# define RCTX_EPS_REG		EPS6
#elif XCHAL_NUM_INTLEVELS == 6
# define RCTX_EPC_REG		EPC7
# define RCTX_EPS_REG		EPS7
#elif XCHAL_NUM_INTLEVELS == 7
# define RCTX_EPC_REG		EPC8
# define RCTX_EPS_REG		EPS8
#else
/* Catch all */
# define RCTX_EPC_REG		_CONCAT(EPC, XCHAL_NUM_INTLEVELS+1)
# define RCTX_EPS_REG		_CONCAT(EPS, XCHAL_NUM_INTLEVELS+1)
#endif

.global _restore_context
_restore_context:
	call0 xtensa_restore_high_regs

	l32i a0, a1, BSA_PC_OFF
	wsr a0, RCTX_EPC_REG
	l32i a0, a1, BSA_PS_OFF
	wsr a0, RCTX_EPS_REG

	l32i a0, a1, BSA_SAR_OFF
	wsr a0, SAR
#if XCHAL_HAVE_LOOPS
	l32i a0, a1, BSA_LBEG_OFF
	wsr a0, LBEG
	l32i a0, a1, BSA_LEND_OFF
	wsr a0, LEND
	l32i a0, a1, BSA_LCOUNT_OFF
	wsr a0, LCOUNT
#endif
#if XCHAL_HAVE_S32C1I
	l32i a0, a1, BSA_SCOMPARE1_OFF
	wsr a0, SCOMPARE1
#endif
#if XCHAL_HAVE_THREADPTR && defined(CONFIG_THREAD_LOCAL_STORAGE)
	l32i a0, a1, BSA_THREADPTR_OFF
	wur a0, THREADPTR
#endif
	rsync

	l32i a0, a1, BSA_A0_OFF
	l32i a2, a1, BSA_A2_OFF
	l32i a3, a1, BSA_A3_OFF
	addi a1, a1, BASE_SAVE_AREA_SIZE

	rfi XCHAL_NUM_INTLEVELS+1

```

hongshui3000, 2021-12-08:
> 

I don't feel relevant

hongshui3000, 2021-12-08:
> Could be related to #40946 .

I don't feel relevant

dcpleung, 2021-12-08:
Hm... that's not right. If `XCHAL_NUM_INTLEVELS == 7`, there are no `EPC8`/`EPS8` so it won't work.

hongshui3000, 2021-12-09:
> 

I don’t have a processor equal to 7, so I think it’s better to use the NMI interrupt level directly, but I’m not sure that all processors have NMI interrupts.

andyross, 2022-01-20:
This should be fixed now that the ZSR feature merged, actually.  It automatically detects the highest non-debug interrupt level to use for the OS EPCx/EPSx registers.  Can you verify and close?

hongshui3000, 2022-01-21:
Thanks ,
I will verify it

closed 2022-02-01 by andyross