zephyrproject-rtos/zephyr #40943: SPI performance with zephyr and the nRF52832 is currently relatively slow

dwalton65, 2021-12-07:
A 5 byte SPI transfer is taking about 17us when using the nordic spim drivers and about 56us when using the zephyr spim drivers.
This is with clock freq = 8M and using DMA.

Would this be considered normal when using the Zephyr drivers as apposed to calling the Nordic drivers?

More information an be seen [here](https://devzone.nordicsemi.com/f/nordic-q-a/81966/slow-spi-performance-with-zephyr).

SPI timing with the nordic drivers
[image]

SPI timing with the zephyr drivers
[image]

anangl, 2021-12-09:
@dwalton65 Please take a look at https://github.com/zephyrproject-rtos/zephyr/issues/33583#issuecomment-826854527.

carlescufi, 2021-12-09:
Closing since this is responded.

closed 2021-12-09 by carlescufi