zephyrproject-rtos/zephyr #40944: `BUILTIN_STACK_CHECKER` and `MPU_STACK_GUARD` with a threading using the FPU will fault the bulltin stack checker

XenuIsWatching, 2021-12-07:
With `BUILTIN_STACK_CHECKER=y` and `MPU_STACK_GUARD=y` and a thread that uses the FPU. The stack checker will always fault thinking there was an overflow, but when really there wasn't.

`BUILTIN_STACK_CHECKER` puts a magic word at the start of the stack when the thread is created. But then after the thread runs `z_arm_thread_stack_info_adjust` could move the pointer to the start of the stack after the magic word was already written for the built in stack checker. `z_check_stack_sentinel` when then check that pointer to the start of the stack and it won't find it's magic word there anymore, and will believe something overwrote it and then trigger a fault.

XenuIsWatching, 2021-12-07:
*Ideally, someone shouldn't need  `BUILTIN_STACK_CHECKER=y` and `MPU_STACK_GUARD=y` together, but it shouldn't at least fault if this happens. It should either rewrite the magic word if the kconfig is enabled, or it should build assert if both k configs are enabled.*

closed 2022-02-21 by github-actions[bot]

box-section, 2023-01-04:
I encountered this issue as well. My project has the following two config flags.

```
CONFIG_STACK_SENTINEL=y
CONFIG_MPU_STACK_GUARD=y
```

The stack and sentinel are set in `setup_thread_stack()`, but then the fields are modified in `z_arm_thread_stack_info_adjust()` (via `z_arm_configure_dynamic_mpu_regions()`) without writing the sentinel word to the new stack start location. This results in a false trigger of the stack check in `z_check_stack_sentinel()`. 

This simple patch within `z_arm_thread_stack_info_adjust()` seems to resolve my case:
```
#ifdef CONFIG_STACK_SENTINEL
				/* Update sentinel, if enabled */
				*((uint32_t *)thread->stack_info.start) = STACK_SENTINEL;
#endif /* CONFIG_STACK_SENTINEL */
```