zephyrproject-rtos/zephyr #40946: Xtensa Interrupt nesting issue

hongshui3000, 2021-12-07:
**Describe the bug**
Correct code behavior
[image]
zephyr  The remaining program of level1 is not executed
[image]

andyross, 2021-12-07:
Can we get some context?  Nested interrupts are exercised in the zephyr test suite; I'm reasonably certain that something more complicated has to be happening here.  Is this something you can make happen in qemu?

hongshui3000, 2021-12-08:
Let me correct what I see now
[image]

hongshui3000, 2021-12-09:
https://github.com/zephyrproject-rtos/zephyr/issues/41039

hongshui3000, 2022-01-21:
@andyross 
NEST processing I only see in the assembly code below, but what I see is that he can not enter the thread switch program when the interrupt nest returns

In any case he will call the function "xtensa_int##l##_c", and then call z_get_next_switch_handle

Am I understanding it wrong?

```
_switch_stacks_\@:
	addi a0, a0, 1
	s32i a0, a3, \NEST_OFF       >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>  count + 1
	l32i a3, a3, \INTSTACK_OFF 
_do_call_\@:
	CROSS_STACK_CALL

	/* Mask interrupts (which have been unmasked during the handler
	 * execution) while we muck with the windows and decrement the nested
	 * count.  The restore will unmask them correctly.
	 */
	rsil a0, XCHAL_NMILEVEL

	/* Decrement nest count */>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>   count - 1                            
	rsr.\SR a3
	l32i a0, a3, \NEST_OFF
	addi a0, a0, -1
	s32i a0, a3, \NEST_OFF
```

```
There is a nesting count kept on the CPU struct to use during interrupt exit and decide whether to call z_get_next_switch_handle() or just do a synchronous RFI.
```
Can you point me to where the code you're talking about is here?

andyross, 2022-02-15:
So... I finally got around to validating nested interrupts on xtensa.  And indeed, we **don't** have a test case in the current tree (I wrote an arch-specific one long ago which is no longer enabled, and the general one never got an xtensa version).  So I spent a little time coming up with what IMHO is a more portable framework for doing this in #42806

And... it works.  :)  I'm not saying that you can't possibly be having trouble with nested interrupts.  But I suspect it's more complicated than described here and may be related to the complicated high priority interrupt handling I know you're doing.  I'm going to close this for now.  Please feel free to reopen if you can point to a reproducing case I can test with existing platforms.

closed 2022-02-15 by andyross