zephyrproject-rtos/zephyr #40955: ADC: STM32: Add device tree config for clock source

pdietl, 2021-12-07:
Looking here https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/adc/adc_stm32.c#L897, you can see that ADC clock source is hard-coded to `LL_ADC_CLOCK_ASYNC_DIV4`. I'd like to see some config added to the device tree to allow one to specify this.

I really need some help on what would be a good device tree addition to represent this.

FRASTM, 2022-01-06:
I think that a new entry in the dts/bindings/adc/st,stm32-adc.yaml
```
    st,prescaler:
      type: int
      required: false
      description: |
        Clock prescaler at the input of the adc
        Could be in range [0 .. 0xF] for STM32 ADC prescaler
```

With a `clock_prescaler` field  in the `adc_stm32_cfg` structure directly defines the index of a `adc_clock_prescaler[]` table. So the stm32 adc driver init is passing this `adc_clock_prescaler[config->clock_prescaler]` instead of a fixed value,  as the CommonClock parameter of the `LL_ADC_SetCommonClock` function.

This `adc_clock_prescaler[]` table dimension depends on the STM32 serie, to fit the clock source (second) parameter LL_ADC_SetClock or LL_ADC_SetCommonClock.

erwango, 2022-03-21:
@pdietl Do you plan to push a PR for this ?

TaiL-Bmt, 2022-12-07:
Can anyone guide me on how to set the ADC Prescaler for STM32F105? The maximum value is 14MHz, but I cannot find any way to set this configuration on Zephyr.

FRASTM, 2023-01-13:
for the stm32F105, The ADC prescaler is bit [15:14] of the RCC configuration register CFGR  (see [RM008](https://www.st.com/content/ccc/resource/technical/document/reference_manual/59/b9/ba/7f/11/af/43/d5/CD00171190.pdf/files/CD00171190.pdf/jcr:content/translations/en.CD00171190.pdf))
```
Bits 14:14 ADCPRE[1:0]: ADC prescaler
Set and cleared by software to select the frequency of the clock to the ADCs.
00: PCLK2 divided by 2
01: PCLK2 divided by 4
10: PCLK2 divided by 6
11: PCLK2 divided by 8
```

erwango, 2023-01-13:
> Can anyone guide me on how to set the ADC Prescaler for STM32F105? The maximum value is 14MHz, but I cannot find any way to set this configuration on Zephyr. @FRASTM

If prescaler is within RCC, I suffest to adapt fro what was done for USB:
https://github.com/zephyrproject-rtos/zephyr/pull/53257

TaiL-Bmt, 2023-01-13:
I am using zephyr 2.6.0, and the problem is I cannot find the proper one in dts file to set ADCPRE bit field. Therefore, I created one for myself. Anw, thanks for your information.

FRASTM, 2023-01-13:
You can have a look at https://github.com/FRASTM/zephyr/tree/f1_adcpre

PR #59874 (2023-08-29): Add support for STM32 ADC clock and prescaler
This PR adds the support and configuration of the full clock capabilities of the STM32 ADCs.

On many series, the ADC can be configured to use as clock source either a synchronous clock or an asynchronous one, though only one of them may be available on some series. Furthermore, a prescaler is configurable to divide the chosen clock. Also, depending on the series, the asynchronous clock can be selected between different clocks (for example SYSCLK, PLL out, HSE...).

Functionalities vary between series, and this PR tries to address all cases, while  [... cut]
closed 2023-08-29 by fabiobaltieri