zephyrproject-rtos/zephyr #40963: mgmt/mcumgr/lib: RFC Black-box testing of tansports

de-nordic, 2021-12-07:
**Is your feature request related to a problem? Please describe.**
The mcumgr is missing black-box type tests of its features over supported transports (uart,bluetooth,udp).

**Describe the solution you'd like**
Automatic black-box tests of various configurations of mcumgr SMP protocol with use of smp_server as test subject.
Probably mcumgr-cli would have to be used as test client, as we currently do not have other tool that would allow us to perform black-box testing of SMP protocol.

**Describe alternatives you've considered**
Manual tests are done, but they are very time consuming and sometimes need to be rushed which leads to mistakes where defects are not detected until other users stumble upon them. The smp_svr is already built, so at least building with various flags is already covered.

**Additional context**
smp_svr sample build configurations (sanity checks of builds) https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/subsys/mgmt/mcumgr/smp_svr/sample.yaml

lairdjm, 2021-12-30:
I've been looking into this for testing our custom SMP commands, not so much for testing the UART/BLE transports and it's apparent how much of a pain it is to do this. The way I currently have it is having a virtual log output and log backend, with essentially a clone of the UART mcumgr files but instead of using the UART, putting the data in a buffer and then from the tests themselves, encapsulating SMP commands using cddl (very useful tool by the way), sending them through the 'virtual UART' transport, waiting for a response, then parsing through the data and decoding it using the cddl decoders.
This however, has created an absolute behemoth of a test application, weighing in at 130KB flash and 68KB RAM usage (nRF52840). My first takeaway from this would be it would probably be useful to follow the same lead as the shell module, and have a dummy SMP module that can be enabled by Kconfig, that way test applications can be created to check SMP functions without needing to dump a whole lot into each test, I haven't looked into the dummy shell module to see how it actually works or how to access the data from it.
The other test ability by using mcumgr from ztest is an interesting one, and I'm not sure how that can be made possible. E.g. with a real board connected, you can have a real serial port or network interface, but if using a virtualised qemu board, is there any way that mcumgr can communicate with that as-is through the serial transport mechanism? And how would that test run from twister itself, as far as I'm aware it doesn't support running external scripts or applications to interact with tests.

Lots of unanswered questions to think about.

de-nordic, 2021-12-30:
I believe that we should be able to setup native posix/qemu x86 smp_svr application that will allow us to at least test UART transport.