zephyrproject-rtos/zephyr #40965: Halt on receipt of Google Cloud IoT Core MQTT message sized 648+ bytes

JeffreyUrban, 2021-12-07:
**Describe the bug**

I adapted the [google_iot_mqtt sample](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/cloud/google_iot_mqtt) to the Nordic nRF9160-DK dev board. I found that when setting the Google Cloud IoT Core config payload to content of approximately 648 bytes or larger, the Zephyr application halts and becomes unresponsive. If shell is enabled, the shell halts and becomes unresponsive as well. It does not output any error on the console or reset. This behavior become more consistent with higher payload sizes, such as 768 bytes. Note that these payloads are smaller than the rx_buffer, which is left at the default 1024. See also [my issue on my repo](https://github.com/JeffreyUrban/nRF9160-google_iot_mqtt/issues/2).

In debugging, I found that this halt occurs before generating log output from entering the corresponding MQTT_EVT_PUBLISH event in protocol.c, so this appears to be happening in the Zephyr MQTT library. See `zephyr/subsys/net/lib/mqtt/mqtt_rx.c` `evt.type = MQTT_EVT_PUBLISH`.

Perhaps there is a difference (possible MQTT non-compliance) on the Google Cloud IoT Core side that is not accommodated by Zephyr's MQTT library.

I do not have a target that is compatible with the original example, so I cannot confirm if this issue affects the original sample. I could use some assistance or suggestions to troubleshoot further.

**To Reproduce**
Steps to reproduce the behavior:
1. `mkdir workspace`
2. `cd workspace`
3. `git clone git@github.com:JeffreyUrban/nRF9160-google_iot_mqtt.git` 
4. Create `nRF9160-google_iot_mqtt/west.yml`:
```
manifest:
  remotes:
    - name: ncs
      url-base: https://github.com/nrfconnect
  projects:
    - name: sdk-nrf
      remote: ncs
      path: nrf
      revision: v1.7.1
      import: true
      clone-depth: 1
  self:
    path: nRF9160-google_iot_mqtt
```
5. `west init -l nRF9160-google_iot_mqtt`
6. `west update`
7. (Set up GCP IoT Core, download certs and program to the nRF9160-DK, set GCP values in prj.conf... see [README.rst](https://github.com/JeffreyUrban/nRF9160-google_iot_mqtt) for full detail. Optionally, add `CONFIG_SHELL=y`)
8. `cd nRF9160-google_iot_mqtt`
9. `west build -b nrf9160dk_nrf9160_ns`
10. (plug in by USB and power on dev board)
11. `west flash --erase`
12. Boot and monitor the console via picocom while entering a config message in Google Cloud IoT Core console. Or, enter a config message in Google Cloud IoT Core console, and then boot and monitor the console via picocom.

**Expected behavior**

In experimenting with other similar samples using the Zephyr MQTT library, such as in the nRF Connect SDK mqtt_simple sample, and on different MQTT brokers (mqtt.eclipseprojects.io, test.mosquitto.org), larger packet sizes are received OK, or generate an error publish_get_payload failed: -122 when the payload exceeds the buffer size, and the application disconnects and reconnects to the MQTT broker. I expected similar behavior.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

This prevents use of Google Cloud IoT Core as an IoT backend.

**Logs and console output**

None. The console becomes completely unresponsive before any output corresponding to the message.

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain nRF v1.7.1 via west
 - Commit SHA or Version used: nRF v1.7.1

**Additional context**

Original and unchanged nRF9160-DK dev board.

rlubos, 2021-12-09:
Hi @JeffreyUrban 
I've managed to reproduce the behaviour with a different MQTT broker, by duplicating the PUBLISH processing logic from the Google Cloud sample. It seems that the sample has a bug, specifically here:
https://github.com/zephyrproject-rtos/zephyr/blob/main/samples/net/cloud/google_iot_mqtt/src/protocol.c#L169
The sample uses non-blocking `mqtt_read_publish_payload()`  function which may return `-EAGAIN`, but this fact is completely ignored (and the numeric value of `EAGAIN` is added to the remaining `len`. This is clearly wrong, as it leads to an infinite loop within the PUBLISH handler.

A simple fix for the issue is to use `mqtt_read_publish_payload_blocking()` instead - this function will only return if there is some actual data to read, or an error occurs.

JeffreyUrban, 2021-12-09:
@rlubos thanks so much for catching that. I have made this change and submitted a pull request: https://github.com/zephyrproject-rtos/zephyr/pull/41042

PR #41042 (2021-12-14): samples: net: google_iot_mqtt: Change read publish to blocking 
Avoid endless loop in `case MQTT_EVT_PUBLISH`

Fixes #40965
closed 2021-12-14 by carlescufi