zephyrproject-rtos/zephyr #40970: Upgrade qemu to fix breakage in mps3-an547

theotherjimmy, 2021-12-07:
**Describe the bug**
The test `tests/lib/sprintf/libraries.libc.sprintf`fails on the mps3 an547 platform.

**To Reproduce**
it's reproducable with zephyr sdk 0.13.2 and the following command:
```
python zephyr/scripts/twister -c -s tests/lib/sprintf/libraries.libc.sprintf -p mps3_an547
```

**Expected behavior**
The test should pass

**Impact**
This is problematic for both testing things on qemu with the mps3 an547 platform, and also
this blocks any PR that affects the mps3 an547 target.

**Logs and console output**
```
    ; head -n 10  /home/jimbri01/src/c/zephyr/twister-out/mps3_an547/tests/lib/sprintf/libraries.libc.sprintf/handler.log
*** Booting Zephyr OS build v2.7.99-2141-gf6fd4444b6be  ***
Running test suite test_sprintf
===================================================================
START - test_sprintf_misc
 PASS - test_sprintf_misc in 0.2 seconds
===================================================================
START - test_sprintf_double

    Assertion failed at WEST_TOPDIR/zephyr/tests/lib/sprintf/src/main.c:219: test_sprintf_double: ((strcmp(buffer, "1.000000") == 0) is false)
sprintf(1.0) - incorrect output '6.257955'
```
**Environment (please complete the following information):**
 - Host OS: Linux x86_64
 - Toolchain: Zephyr SDK 0.13.2, `sha256-D4CI4hgipnJwy9OPxurGA9PH0b7g0XhsygPrvlXOzFo=`
 - Commit SHA of Zephyr: `f6fd4444b6be157d0293406dbefdf884601b796b`

stephanosio, 2021-12-08:
This is a known issue and not necessarily a bug:
https://github.com/zephyrproject-rtos/zephyr/commit/96c7f6ab75a4055a292d5fcd52d413d3a8ba692f

MVE emulation support will be added in QEMU 6.2, which has not been released yet.

Note that it should work if you run this on the ARM FVP.

Please disable the affected tests for now.

theotherjimmy, 2021-12-08:
Ah, yeah should have mentioned that I knew it was fixed on main (master?). 
1) How long do you expect between QEMU 6.2, which is soon to be released, and updating the zephyr sdk with the new version? 
2) do you expect that any changes will be blocked by this? 

(it's my understanding that mps3 an547 tests are not run all that often). I ask because it may not be worth the effort if we can get a new SDK out in short order, as it will likely impact very few people.

stephanosio, 2021-12-08:
> How long do you expect between QEMU 6.2, which is soon to be released, and updating the zephyr sdk with the new version?

I will try to get a new SDK released as soon as QEMU 6.2 is released -- it will likely take 1-2 weeks after the QEMU release.

> do you expect that any changes will be blocked by this?

I am not aware of any changes that will be blocked by upgrading QEMU to 6.2, as of now.

> (it's my understanding that mps3 an547 tests are not run all that often). I ask because it may not be worth the effort if we can get a new SDK out in short order, as it will likely impact very few people.

As long as the PRs blocked by this can wait, I agree that we should leave this as is until the new SDK release.

theotherjimmy, 2021-12-08:
> I will try to get a new SDK released as soon as QEMU 6.2 is released -- it will likely take 1-2 weeks after the QEMU release.

That sounds quite quick, especially for December :grin: 

> I am not aware of any changes that will be blocked by upgrading QEMU to 6.2, as of now.

Ah, I meant Zephyr PRs. I think mine was the only one.

> As long as the PRs blocked by this can wait, I agree that we should leave this as is until the new SDK release.

I don't think there are any such PRs. So I think we should skip both the test-disable step and the test-enable step. I'll see what I can do about testing with the rc of qemu just to be sure there are no further bugs that need fixing.

stephanosio, 2021-12-30:
Related to https://github.com/zephyrproject-rtos/sdk-ng/issues/420.
To be fixed by https://github.com/zephyrproject-rtos/sdk-ng/pull/421.

PR #45287 (2022-05-05): boards: mps3_an547: Enable MVE on QEMU
This series re-enables the M-Profile Vector Extension (MVE) feature when QEMU emulation is used.

Note that the MVE was disabled for QEMU because earlier versions (< 6.2) did not support the emulation of the MVE instructions: see #40569 and #41018.

This series also includes some fixes for the CMSIS-DSP integration and tests so that they can build successfully with the MVE enabled.

Fixes #40970

~~Merge after #45301~~ merged
closed 2022-05-05 by stephanosio