zephyrproject-rtos/zephyr #40974: Xtensa High priority interrupts cannot be masked during initialization

hongshui3000, 2021-12-08:
**Describe the bug**
High priority interrupts cannot be masked during initialization

Once the high-priority interrupt is enabled later before the interrupt stack is initialized and the interrupt occurs again, there will be issues.
file :crt1.s
```
# ifdef __XTENSA_CALL0_ABI__
	/*
	 * PS.WOE = 0
	 * PS.UM = 1
	 * PS.EXCM = 0
	 * PS.INTLEVEL = XCHAL_EXCM_LEVEL
	 */
	movi	a3, PS_UM|PS_INTLEVEL(XCHAL_EXCM_LEVEL)
# else
	/*
	 * PS.WOE = 1
	 * PS.UM = 1
	 * PS.EXCM = 0
	 * PS.INTLEVEL =XCHAL_EXCM_LEVEL
	 */
	movi	a3, PS_UM|PS_WOE|PS_INTLEVEL(XCHAL_EXCM_LEVEL)
# endif

```
My fix

```
#if XCHAL_HAVE_EXCEPTIONS
# ifdef __XTENSA_CALL0_ABI__
	/*
	 * PS.WOE = 0
	 * PS.UM = 1
	 * PS.EXCM = 0
	 * PS.INTLEVEL = XCHAL_DEBUGLEVEL-1
	 */
	movi	a3, PS_UM|PS_INTLEVEL(XCHAL_DEBUGLEVEL-1)
# else
	/*
	 * PS.WOE = 1
	 * PS.UM = 1
	 * PS.EXCM = 0
	 * PS.INTLEVEL = XCHAL_DEBUGLEVEL-1
	 */
	movi	a3, PS_UM|PS_WOE|PS_INTLEVEL(XCHAL_DEBUGLEVEL-1)
# endif
```

zephyr V2.7

andyross, 2022-01-20:
This is the same comment from a different bug:  That's not a Zephyr interrupt.  It's not a Zephyr handler.  If you want to interrupt the OS at arbitrary times with an interrupt that it does not mask, you need to write a safe handler for that interrupt.

I don't know what you want the OS to do for you?

hongshui3000, 2022-01-21:
These are the xtensa arch code, it exists in zephyr, I think it should be reusable by most of the xtensa architecture, so I think this should be part of the zephyrrtos xtensa architecture. Of course I can change and PR

andyross, 2022-01-25:
I'm going to close this one, as it's effectively a duplicate of #41039 

Zephyr doesn't have a solution currently for applications registering interrupts at or beyond EXCM_LEVEL.  It would be a good feature if we had it, but we don't yet.

closed 2022-01-25 by andyross