zephyrproject-rtos/zephyr #40982: Build system: West: Add a warning when used repository does not match manifest

rakons, 2021-12-08:
**Is your feature request related to a problem? Please describe.**
I often forget to run west update before building. Sometimes it takes time to understand that reported build system errors are related to wrong repositories used.

**Describe the solution you'd like**
Add a warning at the very beginning that used repository does not match the manifest. It should be only a warning and should allow build continuation.

**Describe alternatives you've considered**
I do not see a real alternatives. We should not force the user to use the right repositories as it would make testing solutions before pushing it into a branch much harder.

henrikbrixandersen, 2022-11-02:
CC: @mbolivar-nordic

mbolivar-nordic, 2022-11-02:
Dupe of https://github.com/zephyrproject-rtos/zephyr/issues/23025

closed 2022-11-02 by mbolivar-nordic