zephyrproject-rtos/zephyr #40989: Moving Cpu power states to SoC dtsi

Michel-Haber, 2021-12-08:
**Is your enhancement proposal related to a problem? Please describe.**
Power states for SoCs (when defined) are currently in the board device trees. This means that each board needs to define its own states and assign them to the cpu-power-states. I'm not sure my reasoning is correct, but it seems to me that the power states as defined by the current power management model are in the sole purview of the SoC.

**Describe the solution you'd like**
I think it's a good idea to move the definitions of the power states to the SoC definitions, and give cpu-power-states a reasonable default (i.e normal and simple low power modes). This leaves the possibility for each specific board to push its power management further, and include (but not define) more power modes. I think this should allow board designers to not go into defining their own states, and thus remove a barrier of entry for power management. It also avoids mindless copy-pasting of power states between boards.

**Describe alternatives you've considered**
Another alternative would be to not even allow the boards to specify the state, but force them to rely on the applicative codes to set/clear constraints on certain states.

**Additional context**
We have implemented more than 1 board with need for low power modes, and this always seems like a weird concept, even if not necessarily very problematic, so I thought I'd get your thoughts on this :)

By the way, I do not discount the possibility of external power management of a board. But those do not seem to fit in the system power management model in the first place, and would probably go in the drivers for regulators, or even applicative code.

henrikbrixandersen, 2022-11-02:
Desired power states are really application-specific, not specific to the board nor the SoC.

Michel-Haber, 2022-11-03:
Agreed.
My idea is to split definition and usage:
1. The broadest definition of the possible power states is given by the SoC. It's the ability of the MCU itself to go into sleep, deep-sleep, snooze, etc...
2. The application level is the only one able to define which states it needs to use.

Last time I looked, it was already possible for the application to specify its desired states, and to say when to switch between them, but the state definitions themselves were defined at the board level, and this is what I was proposing to change.