zephyrproject-rtos/zephyr #40999: Unable to sign smp_svr sample image as documentation suggests

russelltg, 2021-12-08:
**Describe the bug**

I'm developing an nrf52832-based application and I'm trying to add DFU via MCUMGR over BLE transport. I'm following [the documentation](https://docs.zephyrproject.org/2.6.0/samples/subsys/mgmt/mcumgr/smp_svr/README.html), but the commands suggested for signing the binary don't work.

From my digging, this is related to `CONFIG_ROM_START_OFFSET`, which needs to be >32 for Mcuboot, but setting it makes mcuboot fail to find the image. 

What is this config option supposed to be, and how can I get it to work? Am I missing another config option?

**To Reproduce**

```bash
rm -rf build
west build -b nrf52dk_nrf52832 zephyr/samples/subsys/mgmt/mcumgr/smp_svr -- -DOVERLAY_CONFIG=overlay-bt.conf
west sign -t imgtool -- --key bootloader/mcuboot/root-rsa-2048.pem
```

The sign fails

**Expected behavior**

The image is able to sign, and DFU with the image works properly. 

**Impact**

Showstopper

**Logs and console output**

The build completes successfully ([log](https://gist.github.com/russelltg/c6e5d5c7df86321c2bc5c385c97a1440)), but signing fails with:

```log
west sign -t imgtool -- --key bootloader/mcuboot/root-rsa-2048.pem
=== image configuration:
partition offset: 49152 (0xc000)
partition size: 204800 (0x32000)
rom start offset: 0 (0x0)
=== signing binaries
unsigned bin: /home/russelltg/ncs/build/zephyr/zephyr.bin
signed bin:   /home/russelltg/ncs/build/zephyr/zephyr.signed.bin
Usage: imgtool sign [OPTIONS] INFILE OUTFILE

Error: Invalid value for '-H' / '--header-size': Minimum value for -H/--header-size is 32
FATAL ERROR: command exited with status 2: /home/russelltg/.local/bin/imgtool sign --version 0.0.0+0 --align 4 --header-size 0 --slot-size 204800 --key bootloader/mcuboot/root-rsa-2048.pem /home/russelltg/ncs/build/zephyr/zephyr.bin /home/russelltg/ncs/build/zephyr/zephyr.signed.bin
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gcc-arm-none-eabi-9-2019-q4-major
 - Commit SHA: `a652410da5`

heinwessels, 2021-12-09:
You could try `west flash --hex-file build/zephyr/zephyr.hex --skip-rebuild`. In my experience if you don't do this the application will quickly rebuild, thus overwriting the signed header, which means that MCUBoot will not recognise the image. It could also explain why it works when you first merge the hex file. That said, I usually use `west flash --build-dir build --skip-rebuild`, and don't specify the hex file explicitly.

russelltg, 2021-12-09:
Doesn't seem to produce any different results

heinwessels, 2021-12-09:
> Doesn't seem to produce any different results

That's unfortunate. It's also peculiar that `CONFIG_ROM_START_OFFSET` is set to zero. I've only had it not create that offset when `CONFIG_BOOTLOADER_MCUBOOT` is not `y`, but this should be set correctly since it's set in `prj.conf`. You could confirm that `CONFIG_BOOTLOADER_MCUBOOT` is set by checking `build/zephyr/.config`.

russelltg, 2021-12-09:
Certainally seems like the case. 

```bash
$ rg 'CONFIG_ROM_START_OFFSET|CONFIG_BOOTLOADER_MCUBOOT' build/zephyr/.config
18:CONFIG_ROM_START_OFFSET=0
2022:CONFIG_BOOTLOADER_MCUBOOT=y
````

Fwiw, this checks out after looking at [the docs for `CONFIG_ROM_START_OFFSET`](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_ROM_START_OFFSET.html)--none of the snippets that would set it to anything else apply to this board/soc.

utzig, 2021-12-09:
I just tried the steps given

```
west build -b nrf52dk_nrf52832 zephyr/samples/subsys/mgmt/mcumgr/smp_svr -- -DOVERLAY_CONFIG=overlay-bt.conf
west sign -t imgtool -- --key bootloader/mcuboot/root-rsa-2048.pem
```

And there is no error, also `CONFIG_ROM_START_OFFSET=0x200` is properly set. I have not tried to run. Zephyr was updated this morning, and `west update` and `West version: v0.12.0`.

utzig, 2021-12-09:
Btw, I would suggest always building in separate dirs which can be achieved with `west build -d <dir>`, to avoid ever mixing the bootloader and app builds...

russelltg, 2021-12-09:
Okay tried this again with a brand new zephyr workspace (that has latest zephyr, not using the zephyr tree supplied by Nordic, which is missing the line in Kconfig.zephyr which sets the CONFIG_ROM_START_OFFSEt), with different results. First off, the bootloader doesn't build:

```log
$ west build -b nrf52dk_nrf52832 -d build_mcuboot bootloader/mcuboot/boot/zephyr
-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): /home/russelltg/ncsmain/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/russelltg/ncsmain/bootloader/mcuboot/boot/zephyr
-- Zephyr version: 2.7.99 (/home/russelltg/ncsmain/zephyr), build: v2.7.99-2227-g92fbd7da14d1
-- Found west (found suitable version "0.12.0", minimum required is "0.7.1")
-- Board: nrf52dk_nrf52832
-- Cache files will be written to: /home/russelltg/.cache//zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.6.1", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (/home/russelltg/ncs/gcc-arm-none-eabi-9-2019-q4-major)
-- Found BOARD.dts: /home/russelltg/ncsmain/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832.dts
-- Found devicetree overlay: /home/russelltg/ncsmain/bootloader/mcuboot/boot/zephyr/dts.overlay
-- Found devicetree overlay: /home/russelltg/ncsmain/bootloader/mcuboot/boot/zephyr/dts.overlay
-- Found devicetree overlay: /home/russelltg/ncsmain/bootloader/mcuboot/boot/zephyr/dts.overlay
-- Found devicetree overlay: /home/russelltg/ncsmain/bootloader/mcuboot/boot/zephyr/dts.overlay
-- Found devicetree overlay: /home/russelltg/ncsmain/bootloader/mcuboot/boot/zephyr/dts.overlay
-- Found devicetree overlay: /home/russelltg/ncsmain/bootloader/mcuboot/boot/zephyr/dts.overlay
-- Generated zephyr.dts: /home/russelltg/ncsmain/build_mcuboot/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/russelltg/ncsmain/build_mcuboot/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/russelltg/ncsmain/build_mcuboot/zephyr/include/generated/device_extern.h
-- Including generated dts.cmake file: /home/russelltg/ncsmain/build_mcuboot/zephyr/dts.cmake

/home/russelltg/ncsmain/bootloader/mcuboot/boot/zephyr/prj.conf:3: warning: attempt to assign the value 'y' to the undefined symbol SYSTEM_CLOCK_DISABLE
Parsing /home/russelltg/ncsmain/bootloader/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/home/russelltg/ncsmain/zephyr/boards/arm/nrf52dk_nrf52832/nrf52dk_nrf52832_defconfig'
Merged configuration '/home/russelltg/ncsmain/bootloader/mcuboot/boot/zephyr/prj.conf'

error: Aborting due to Kconfig warnings

CMake Error at /home/russelltg/ncsmain/zephyr/cmake/kconfig.cmake:272 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/russelltg/ncsmain/zephyr/cmake/app/boilerplate.cmake:554 (include)
  /home/russelltg/ncsmain/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/russelltg/ncsmain/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:44 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/russelltg/ncsmain/build_mcuboot/CMakeFiles/CMakeOutput.log".
See also "/home/russelltg/ncsmain/build_mcuboot/CMakeFiles/CMakeError.log".
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python -B/home/russelltg/ncsmain/build_mcuboot -S/home/russelltg/ncsmain/bootloader/mcuboot/boot/zephyr -GNinja -DZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb '-DGNUARMEMB_TOOLCHAIN_PATH=~/ncs/gcc-arm-none-eabi-9-2019-q4-major/'

```

I just commented out that line that sets CONFIG_SYSTEM_CLOCK_DISABLE (idk if that's a problem)--and was able to get the flashing to work fine. 

DFU didn't work, giving an error "Not enough free space to run swap upgrade", but adding `-DCONFIG_BOOT_SWAP_USING_SCRATCH=y` to my commandline while building the bootloader got everything to work.

Thanks!

It seems like there's a bug with Nordic's zephyr version, and there may have been some regression in the bootloader build system, as well as maybe a bad default option for SWAP_USING_SCRATCH?

closed 2021-12-10 by russelltg