zephyrproject-rtos/zephyr #41004: [backport v2.7-branch] backport of #40127 failed

zephyrbot, 2021-12-08:
This issue was created automatically because the backport of #40127 to `v2.7-branch` failed.

The backport to `v2.7-branch` failed:
```
The process '/usr/bin/git' failed with exit code 1
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v2.7-branch v2.7-branch
# Navigate to the new working tree
cd .worktrees/backport-v2.7-branch
# Create a new branch
git switch --create backport-40127-to-v2.7-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick 2a232bdee3e087c3389abe253e9d7ed1df625d45~3..2a232bdee3e087c3389abe253e9d7ed1df625d45
# Push it to GitHub
git push --set-upstream origin backport-40127-to-v2.7-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v2.7-branch
```
Then, create a pull request where the `base` branch is `v2.7-branch` and the `compare`/`head` branch is `backport-40127-to-v2.7-branch`.

de-nordic, 2022-02-07:
Shouldn't this be closed with merge of #41008?

de-nordic, 2022-03-11:
@cfriedt  Can we close this?

cfriedt, 2022-03-11:
> Shouldn't this be closed with merge of #41008?

Issues do not close automatically unless the commit message has `Fixes #nnn`.

closed 2022-03-11 by cfriedt

de-nordic, 2022-03-11:
> > Shouldn't this be closed with merge of #41008?
> 
> Issues do not close automatically unless the commit message has `Fixes #nnn`.

Sometime I have thought that it is enough if the PR message contains the info.

PR #41008 (2022-01-15): [Backport v2.7-branch] Fix for: mcumgr over serial does not add CRC to length of packet len
Backport of https://github.com/zephyrproject-rtos/zephyr/pull/40127

Fixes #41004