zephyrproject-rtos/zephyr #41012: irq_enable() doesn’t support enabling IRQ number more than 127

jvasanth1, 2021-12-08:
**Describe the bug**
We are updating kscan driver for MEC172x microchip device. The kscan peripheral is routed to NVIC 135. In the driver init function we enable NVIC 135 by calling the function irq_enable() and passing the IRQ number from the device instance. 

The function irq_enable() casts the IRQ number input to IRQn_type enum (defined in cortex_m/[cmsis.h](https://github.com/MicrochipTech/zephyr/blob/main/include/arch/arm/aarch32/cortex_m/cmsis.h)) which results as a signed 8-bit number. The function will write the NVIC Set Enable register only if the signed 8-bit number is not negative. The NVIC irq 135 results in a negative number (see register R0, R3 in snapshot which shows the conversion). Hence the NVIC register is not written.

Attached screenshot of the debugger window inside function irq_enable()

The platform we are using is mec172xevb_assy6906

Note that if we add one last member to the IRQn_type structure (as the below line), then the issue gets resolved, as compiler no longer treats IRQn_type as 8-bit.

max_IRQn = 0x100,

We are unable to submit the MEC172x kscan driver without this fix in Zephyr core.
[image]

MaureenHelm, 2021-12-14:
I think you have an issue pulling in the correct `IRQn_Type` definition from the MEC HAL, and may be falling back onto some default generic CMSIS definition. There are several NXP and STM M7 SoCs in-tree that support a large number of IRQs, so I think this problem is specific to Microchip.

jvasanth1, 2021-12-14:
Previously it worked for us using MEC1501 device because we created separate typedef enum in MEC1501hsz.h under zephyr/modules/HAL/microchip/mec/mec1501/

But for MEC172x we are following zephyr guideline to avoid create new typedefs (such as IRQn_Type) in the main Zephyr code tree. Therefore enum IRQn {} comes from zephyr/include/arch/arm/aarch32/cortex_m/cmsis.h
We believe this is not as issue for other EC SoCs because their chips have less than 128 external interrupt inputs to the NVIC which fits in 8-bits.

For devices with more than 128 external interrupts, this is an issue as pointed out in this bug and hence needs to be addressed.

jvasanth1, 2021-12-22:
This pull request [#41279](https://github.com/zephyrproject-rtos/zephyr/pull/41279) suggests a possible fix for this issue.

bbolen, 2022-01-03:
I am not a Cortex-M expert, but I think Maureen was implying that there would be a HAL layer in zephyr/modules/hal/microchip which would define IRQn_Type in which you could create your custom irq type definition.  It looks like the 172x doesn't have that.  Can you point to the documentation that led you to the comment "we are following zephyr guideline to avoid create new typedefs "?

I see zephyr/soc/arm/bcm_vk/viper/soc.h defines an IRQn_Type.  This implies to me that defining IRQnType in zephyr/soc/arm/microchip_mec/mec172x/soc.h may be a valid thing to do.  Hopefully, one of the Cortex-M people will chime in and supply the real answer, but I think this path may be more valid.  I'm not sure adding an arbitrary max irqs define is the right thing to do since that number is different for various non-HAL based Cortex-M systems, at least from what I think I understand from the code.

jvasanth1, 2022-01-05:
Hi bbolen,

Thanks for your comments. I fixed this issue by defining IRQn_Type in zephyr/soc/arm/microchip_mec/mec172x/soc.h and tested with our keyscan driver (NVIC 135). The relevant pull request is here [#41279](https://github.com/zephyrproject-rtos/zephyr/pull/41279) 

Note: we would need a exception to the following warning
-:39: WARNING:NEW_TYPEDEFS: do not add new typedefs

Please help to get the pull request merged.

Thanks

PR #41279 (2022-01-20): arm cortex_m: cmsis.h - Fix IRQn_Type enum
Fixed issue [#41012](https://github.com/zephyrproject-rtos/zephyr/issues/41012) based on bbolen's comment. 
Defined custom IRQn_Type in zephyr/soc/arm/microchip_mec/mec172x/soc.h. This will ensure NVIC numbers
more than 127 will work correctly with irq_enable() API.

Fixes #41012
PR #51106 (2022-10-12): [Backport v2.7-branch] arm cortex_m: cmsis.h - Fix IRQn_Type enum
Backport 4495f43dca440de4e1aa810f7448625a351f61da from #41279

Fixes #41012
closed 2022-01-20 by nashif